Joshua-Riek/ubuntu-rockchip #577: Support for OV9281 camera

ahsan-kaukab, 2024-01-24:
Hello all, I am using Orange Pi 5 board and the OS I am currently running is OrangeOS but the problem is it don't support camera module OV9281 and v4l12. So I want to ask is their any default support for OV9281 in this ubuntu-rockchip or some how we can add support for v4l12 and can run OV9281  camera in this OS ?

Joshua-Riek, 2024-01-24:
I don't know if it works, but I would look at the kernel to see if the drivers exist. Then, you may need to create a device tree or overlay to support the hardware.

ahsan-kaukab, 2024-01-24:
can you write steps for creating device tree or overlay for our specific hardware ?

Joshua-Riek, 2024-01-24:
Sorry, this is not something I can assist with. There should be reference information available from the hardware vendor that can be used to create the device tree.

everskies, 2024-01-28:
Hey! I'm currently working on a similar thing for 4 OV9281 cameras. Managed to get I2C communication to work with a custom DTS:

```
&i2c3 {
    status = "okay";

    ov9282_1: ov9282-1@60 {
            compatible = "ovti,ov9281";
            status = "okay";
            reg = <0x60>;

            clocks = <&clk_cam_24m>;
            clock-names = "xvclk";

            avdd-supply = <&cam0_reg>;
            dovdd-supply = <&cam_dovdd>;
			dvdd-supply = <&cam_dvdd>;
            rockchip,camera-module-index = <0>;
            rockchip,camera-module-facing = "back";
            rockchip,camera-module-name = "OV9282";
            rockchip,camera-module-lens-name = "default";
            port {
                    ov9282_out1: endpoint {
                            remote-endpoint = <&mipi_in_ucam2>;
                            data-lanes = <1>;
                    };
            };
    };
...
```

That paired with https://github.com/Joshua-Riek/linux-rockchip/blob/6e8316cd73b25e9c0a06750034552ee1b77e852f/arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi

```
[    7.848054] ov9281 3-0060: driver version: 00.01.05
[    7.848224] ov9281 3-0060: Looking up avdd-supply from device tree
[    7.848308] ov9281 3-0060: Looking up dovdd-supply from device tree
[    7.848369] ov9281 3-0060: Looking up dvdd-supply from device tree
[    7.850821] ov9281 3-0060: Detected OV009281 sensor
[    7.850909] ov9281 3-0010: driver version: 00.01.05
[    7.850990] ov9281 3-0010: Looking up avdd-supply from device tree
[    7.851413] ov9281 3-0010: Looking up dovdd-supply from device tree
[    7.851536] ov9281 3-0010: Looking up dvdd-supply from device tree
[    7.854519] ov9281 3-0010: Detected OV009281 sensor
[    7.854679] ov9281 7-0060: Looking up avdd-supply from device tree
[    7.854878] ov9281 7-0060: Looking up dovdd-supply from device tree
[    7.854912] ov9281 7-0060: Looking up dvdd-supply from device tree
[    7.860212] ov9281 7-0060: Detected OV009281 sensor
[    7.860356] ov9281 7-0010: driver version: 00.01.05
[    7.860447] ov9281 7-0010: Looking up avdd-supply from device tree
[    7.860937] ov9281 7-0010: Looking up dovdd-supply from device tree
[    7.861151] ov9281 7-0010: Looking up dvdd-supply from device tree
[    7.868809] ov9281 7-0010: Detected OV009281 sensor
```

I can't get an image yet as I get spammed with rkcif logs:

```
[    7.943705] rkcif-mipi-lvds3: rkcif_update_sensor_info: stream[1] get remote terminal sensor failed!
[    7.943713] stream_cif_mipi_id1: update sensor info failed -19
[    7.943785] rkcif-mipi-lvds3: rkcif_update_sensor_info: stream[1] get remote terminal sensor failed!
[    7.943793] rkcif_tools_id1: update sensor info failed -19
[    7.944645] rkcif-mipi-lvds3: rkcif_update_sensor_info: stream[2] get remote terminal sensor failed!
[    7.944655] stream_cif_mipi_id2: update sensor info failed -19
[    7.944785] rkcif-mipi-lvds5: rkcif_update_sensor_info: stream[2] get remote terminal sensor failed!
[    7.944794] rkcif_scale_ch2: update sensor info failed -19
[    7.944995] rkcif-mipi-lvds3: rkcif_update_sensor_info: stream[0] get remote terminal sensor failed!
[    7.945002] stream_cif_mipi_id0: update sensor info failed -19
[    7.945333] rkcif-mipi-lvds3: rkcif_update_sensor_info: stream[0] get remote terminal sensor failed!
[    7.945340] rkcif_tools_id0: update sensor info failed -19
[    7.945919] rkcif-mipi-lvds5: rkcif_update_sensor_info: stream[3] get remote terminal sensor failed!
[    7.945930] rkcif_scale_ch3: update sensor info failed -19
[    7.946318] rkcif-mipi-lvds3: rkcif_update_sensor_info: stream[2] get remote terminal sensor failed!
[    7.946326] rkcif_tools_id2: update sensor info failed -19
[    7.947211] rkcif-mipi-lvds5: rkcif_update_sensor_info: stream[1] get remote terminal sensor failed!
[    7.947221] rkcif_tools_id1: update sensor info failed -19
[    7.947358] rkcif-mipi-lvds5: rkcif_update_sensor_info: stream[2] get remote terminal sensor failed!
[    7.947366] rkcif_tools_id2: update sensor info failed -19
[    7.951196] rkcif-mipi-lvds3: rkcif_update_sensor_info: stream[1] get remote terminal sensor failed!
[    7.951208] rkcif_scale_ch1: update sensor info failed -19
[    7.952175] rkcif-mipi-lvds3: rkcif_update_sensor_info: stream[0] get remote terminal sensor failed!
[    7.952186] rkcif_scale_ch0: update sensor info failed -19
[    7.955828] rkcif-mipi-lvds5: rkcif_update_sensor_info: stream[0] get remote terminal sensor failed!
[    7.955840] stream_cif_mipi_id0: update sensor info failed -19
[    7.957786] rkcif-mipi-lvds3: rkcif_update_sensor_info: stream[3] get remote terminal sensor failed!
[    7.957798] stream_cif_mipi_id3: update sensor info failed -19
[    7.958139] rkcif-mipi-lvds3: rkcif_update_sensor_info: stream[3] get remote terminal sensor failed!
[    7.958160] rkcif_scale_ch3: update sensor info failed -19
```

But hopefully this will get you started :)

everskies, 2024-01-29:
Okay, got it to work! Had to make a change in rockchip_linux_defconfig:

`CONFIG_VIDEO_OV9281=m`
to
`CONFIG_VIDEO_OV9281=y`

Joshua-Riek, 2024-01-29:
Maybe you needed to `sudo modprobe ov9281`?

everskies, 2024-01-29:
@Joshua-Riek That didn't work either unfortunately, I'm sure it would if I added it to `/etc/modules` so it's available at early boot

ahsan-kaukab, 2024-01-31:
Thanks for help @everskies  

As I got , following are steps to enable OV9281 camera feature for rk3588 :

-> I have to first clean build directory using
`./build.sh  -c -board=orangepi-5 `

->  I have to change to ` CONFIG_VIDEO_OV9281=y ` in 
`ubuntu-rockchip/build/linux-rockchip/arch/arm64/configs/rockchip_linux_defconfig`

-> I have to add device tree in
`ubuntu-rockchip/build/linux-rockchip/arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi`

-> I have to build the OS again
`./build.sh -board=orangepi-5 `

ahsan-kaukab, 2024-01-31:
debian/rules binary
make KERNELRELEASE=5.10.160 ARCH=arm64 KBUILD_BUILD_VERSION=1 -f ./Makefile
CALL scripts/atomic/check-atomics.sh
DTC arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-ipc-4x-linux.dtb
CALL scripts/checksyscalls.sh
arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:620.27-641.7: ERROR (phandle_references): /i2c@feab0000/ov9281-1@60: Reference to non-existent node or label "clk_cam_24m"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:620.27-641.7: ERROR (phandle_references): /i2c@feab0000/ov9281-1@60: Reference to non-existent node or label "cam0_reg"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:620.27-641.7: ERROR (phandle_references): /i2c@feab0000/ov9281-1@60: Reference to non-existent node or label "cam_dovdd"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:620.27-641.7: ERROR (phandle_references): /i2c@feab0000/ov9281-1@60: Reference to non-existent node or label "cam_dvdd"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:643.25-665.7: ERROR (phandle_references): /i2c@feab0000/ov9281@10: Reference to non-existent node or label "clk_cam_24m"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:643.25-665.7: ERROR (phandle_references): /i2c@feab0000/ov9281@10: Reference to non-existent node or label "cam1_reg"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:643.25-665.7: ERROR (phandle_references): /i2c@feab0000/ov9281@10: Reference to non-existent node or label "cam_dovdd"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:643.25-665.7: ERROR (phandle_references): /i2c@feab0000/ov9281@10: Reference to non-existent node or label "cam_dvdd"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:569.27-590.7: ERROR (phandle_references): /i2c@fec90000/ov9281-3@60: Reference to non-existent node or label "clk_cam_24m"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:569.27-590.7: ERROR (phandle_references): /i2c@fec90000/ov9281-3@60: Reference to non-existent node or label "cam2_reg"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:569.27-590.7: ERROR (phandle_references): /i2c@fec90000/ov9281-3@60: Reference to non-existent node or label "cam_dovdd"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:569.27-590.7: ERROR (phandle_references): /i2c@fec90000/ov9281-3@60: Reference to non-existent node or label "cam_dvdd"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:592.27-613.7: ERROR (phandle_references): /i2c@fec90000/ov9281-4@10: Reference to non-existent node or label "clk_cam_24m"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:592.27-613.7: ERROR (phandle_references): /i2c@fec90000/ov9281-4@10: Reference to non-existent node or label "cam3_reg"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:592.27-613.7: ERROR (phandle_references): /i2c@fec90000/ov9281-4@10: Reference to non-existent node or label "cam_dovdd"

arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-cam-4x.dtsi:592.27-613.7: ERROR (phandle_references): /i2c@fec90000/ov9281-4@10: Reference to non-existent node or label "cam_dvdd"

ERROR: Input tree has errors, aborting (use -f to force output)
make[5]: *** [scripts/Makefile.lib:386: arch/arm64/boot/dts/rockchip/rk3588-nvr-demo-v10-ipc-4x-linux.dtb] Error 2
make[4]: *** [scripts/Makefile.build:516: arch/arm64/boot/dts/rockchip] Error 2
make[3]: *** [Makefile:1468: dtbs] Error 2
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [debian/rules:7: build-arch] Error 2
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
make[1]: *** [scripts/Makefile.package:83: bindeb-pkg] Error 2
make: *** [Makefile:1661: bindeb-pkg] Error 2
Error: in ./scripts/build-kernel.sh on line 44
Error: in ./build.sh on line 157

Sorry I am application level developer so I don't have a good knowledge about DTS and river development

ahsan-kaukab, 2024-02-02:
For my module after updating dts from 
https://github.com/Joshua-Riek/linux-rockchip/blob/6e8316cd73b25e9c0a06750034552ee1b77e852f/arch/arm64/boot/dts/rockchip/rk3588-orangepi-5-plus-camera1.dtsi

The new DTS is like below
`// SPDX-License-Identifier: (GPL-2.0+ OR MIT)
/*
 * Copyright (c) 2021 Rockchip Electronics Co., Ltd.
 *
 */

&csi2_dphy0_hw {
	status = "disabled";
};

&csi2_dphy0 {
	status = "disabled";
	ports {
		#address-cells = <1>;
		#size-cells = <0>;
		port@0 {
			reg = <0>;
			#address-cells = <1>;
			#size-cells = <0>;

			mipi_in_ucam0: endpoint@0 {
				reg = <0>;
				remote-endpoint = <&ov9281_out2>;
				data-lanes = <1>;
			};
		};
		port@1 {
		        reg = <1>;
		        #address-cells = <1>;
		        #size-cells = <0>;
		        csidphy0_out: endpoint@0 {
		                reg = <0>;
		                remote-endpoint = <&mipi2_csi2_input>;
		        };
		};
	};
};

&i2c3 {
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <&i2c3m0_xfer>;

	vm149c_p1: vm149c-p1@c {
		compatible = "silicon touch,vm149c";
		status = "disabled";
		reg = <0x0c>;
		rockchip,camera-module-index = <1>;
		rockchip,camera-module-facing = "back";
	};

	ov9281_1: ov9281-1@10 {
		compatible = "ovti,ov9281";
		status = "okay";
		reg = <0x10>;
		clocks = <&cru CLK_MIPI_CAMARAOUT_M3>;
		clock-names = "xvclk";
		pinctrl-names = "default";
		pinctrl-0 = <&mipim1_camera3_clk>;
		reset-gpios = <&gpio1 RK_PC4 GPIO_ACTIVE_HIGH>;
		pwdn-gpios = <&gpio1 RK_PD5 GPIO_ACTIVE_HIGH>;
		rockchip,camera-module-index = <0>;
		rockchip,camera-module-facing = "back";
		rockchip,camera-module-name = "CMK-CT0116";
		rockchip,camera-module-lens-name = "default";
		lens-focus = <&vm149c_p1>;
		port {
			ov9281_out2: endpoint {
				remote-endpoint = <&mipi_in_ucam0>;
				data-lanes = <1>;
			};
		};
	};

	dw9714_p1: dw9714-p1@c {
		compatible = "dongwoon,dw9714";
		status = "disabled";
		reg = <0x0c>;
		rockchip,camera-module-index = <0>;
		rockchip,vcm-start-current = <10>;
		rockchip,vcm-rated-current = <85>;
		rockchip,vcm-step-mode = <5>;
		rockchip,camera-module-facing = "back";
	};
};

&mipi2_csi2 {
	status = "disabled";
	ports {
		#address-cells = <1>;
		#size-cells = <0>;
		port@0 {
			reg = <0>;
			#address-cells = <1>;
			#size-cells = <0>;
			mipi2_csi2_input: endpoint@1 {
				reg = <1>;
				remote-endpoint = <&csidphy0_out>;
			};
		};
		port@1 {
			reg = <1>;
			#address-cells = <1>;
			#size-cells = <0>;
			mipi2_csi2_output: endpoint@0 {
				reg = <0>;
				remote-endpoint = <&cif_mipi_in2>;
			};
		};
	};
};

&rkcif_mipi_lvds2 {
	status = "disabled";
	port {
		cif_mipi_in2: endpoint {
			remote-endpoint = <&mipi2_csi2_output>;
		};
	};
};

&rkcif_mipi_lvds2_sditf {
	status = "disabled";
	port {
		mipi2_lvds_sditf: endpoint {
			remote-endpoint = <&isp0_vir1>;
		};
	};
};

&rkisp0_vir1 {
	status = "disabled";
	port {
		#address-cells = <1>;
		#size-cells = <0>;
		isp0_vir1: endpoint@0 {
			reg = <0>;
			remote-endpoint = <&mipi2_lvds_sditf>;
		};
	};
};`

ahsan-kaukab, 2024-02-02:
For my module after updating dts from
https://github.com/Joshua-Riek/linux-rockchip/blob/6e8316cd73b25e9c0a06750034552ee1b77e852f/arch/arm64/boot/dts/rockchip/rk3588-orangepi-5-plus-camera1.dtsi

The new DTS is like below::

```
&csi2_dphy0_hw {
	status = "disabled";
};

&csi2_dphy0 {
	status = "disabled";
	ports {
		#address-cells = <1>;
		#size-cells = <0>;
		port@0 {
			reg = <0>;
			#address-cells = <1>;
			#size-cells = <0>;

			mipi_in_ucam0: endpoint@0 {
				reg = <0>;
				remote-endpoint = <&ov9281_out2>;
				data-lanes = <1>;
			};
		};
		port@1 {
		        reg = <1>;
		        #address-cells = <1>;
		        #size-cells = <0>;
		        csidphy0_out: endpoint@0 {
		                reg = <0>;
		                remote-endpoint = <&mipi2_csi2_input>;
		        };
		};
	};
};

&i2c3 {
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <&i2c3m0_xfer>;

	vm149c_p1: vm149c-p1@c {
		compatible = "silicon touch,vm149c";
		status = "disabled";
		reg = <0x0c>;
		rockchip,camera-module-index = <1>;
		rockchip,camera-module-facing = "back";
	};

	ov9281_1: ov9281-1@10 {
		compatible = "ovti,ov9281";
		status = "okay";
		reg = <0x10>;
		clocks = <&cru CLK_MIPI_CAMARAOUT_M3>;
		clock-names = "xvclk";
		pinctrl-names = "default";
		pinctrl-0 = <&mipim1_camera3_clk>;
		reset-gpios = <&gpio1 RK_PC4 GPIO_ACTIVE_HIGH>;
		pwdn-gpios = <&gpio1 RK_PD5 GPIO_ACTIVE_HIGH>;
		rockchip,camera-module-index = <0>;
		rockchip,camera-module-facing = "back";
		rockchip,camera-module-name = "CMK-CT0116";
		rockchip,camera-module-lens-name = "default";
		lens-focus = <&vm149c_p1>;
		port {
			ov9281_out2: endpoint {
				remote-endpoint = <&mipi_in_ucam0>;
				data-lanes = <1>;
			};
		};
	};

	dw9714_p1: dw9714-p1@c {
		compatible = "dongwoon,dw9714";
		status = "disabled";
		reg = <0x0c>;
		rockchip,camera-module-index = <0>;
		rockchip,vcm-start-current = <10>;
		rockchip,vcm-rated-current = <85>;
		rockchip,vcm-step-mode = <5>;
		rockchip,camera-module-facing = "back";
	};
};

&mipi2_csi2 {
	status = "disabled";
	ports {
		#address-cells = <1>;
		#size-cells = <0>;
		port@0 {
			reg = <0>;
			#address-cells = <1>;
			#size-cells = <0>;
			mipi2_csi2_input: endpoint@1 {
				reg = <1>;
				remote-endpoint = <&csidphy0_out>;
			};
		};
		port@1 {
			reg = <1>;
			#address-cells = <1>;
			#size-cells = <0>;
			mipi2_csi2_output: endpoint@0 {
				reg = <0>;
				remote-endpoint = <&cif_mipi_in2>;
			};
		};
	};
};

&rkcif_mipi_lvds2 {
	status = "disabled";
	port {
		cif_mipi_in2: endpoint {
			remote-endpoint = <&mipi2_csi2_output>;
		};
	};
};

&rkcif_mipi_lvds2_sditf {
	status = "disabled";
	port {
		mipi2_lvds_sditf: endpoint {
			remote-endpoint = <&isp0_vir1>;
		};
	};
};

&rkisp0_vir1 {
	status = "disabled";
	port {
		#address-cells = <1>;
		#size-cells = <0>;
		isp0_vir1: endpoint@0 {
			reg = <0>;
			remote-endpoint = <&mipi2_lvds_sditf>;
		};
	};
};
```

everskies, 2024-02-03:
@ahsan-kaukab 

I've managed to get 1 lane OV9281 to work

Had to fix a bug in the driver code, and change it to work in 1 lane mode instead of 2

[ov9281.patch](https://github.com/Joshua-Riek/ubuntu-rockchip/files/14154359/ov9281.patch)

ahsan-kaukab, 2024-02-05:
Hello @everskies , Sorry for disturbing again but after doing necessary thing and making some changes in camera device tree , the new device tree is as follow:
https://github.com/ahsan-kaukab/linux-rockchip-ov2981/blob/orange-pi-5-ov9281/arch/arm64/boot/dts/rockchip/rk3588s-orangepi-5-camera1.dtsi

still ,I did not see any /dev/video0 file in OS also no ov9281 message in dmesg,
some of the relevant kernel messages are as follow 

User
```
[    9.329247] rockchip-csi2-dphy-hw fedc8000.csi2-dphy1-hw: csi2 dphy hw probe successfully!
[    9.330575] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up phy-supply from device tree
[    9.330583] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up phy-supply property in node /syscon@fd5d0000/usb2-phy@0/otg-port failed
[    9.330626] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up vbus-supply from device tree
[    9.330632] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up vbus-supply property in node /syscon@fd5d0000/usb2-phy@0/otg-port failed
[    9.331858] phy phy-fd5d8000.syscon:usb2-phy@8000.1: Looking up phy-supply from device tree
[    9.331866] phy phy-fd5d8000.syscon:usb2-phy@8000.1: Looking up phy-supply property in node /syscon@fd5d8000/usb2-phy@8000/host-port failed
[    9.332973] phy phy-fd5dc000.syscon:usb2-phy@c000.2: Looking up phy-supply from device tree
[    9.332981] phy phy-fd5dc000.syscon:usb2-phy@c000.2: Looking up phy-supply property in node /syscon@fd5dc000/usb2-phy@c000/host-port failed
[    9.334748] phy phy-fee00000.phy.3: Looking up phy-supply from device tree
[    9.334754] phy phy-fee00000.phy.3: Looking up phy-supply property in node /phy@fee00000 failed
[    9.334887] phy phy-fee20000.phy.4: Looking up phy-supply from device tree
[    9.334892] phy phy-fee20000.phy.4: Looking up phy-supply property in node /phy@fee20000 failed
[    9.335366] phy phy-feda0000.phy.5: Looking up phy-supply from device tree
[    9.335371] phy phy-feda0000.phy.5: Looking up phy-supply property in node /phy@feda0000 failed
[    9.335494] phy phy-fedb0000.phy.6: Looking up phy-supply from device tree
[    9.335499] phy phy-fedb0000.phy.6: Looking up phy-supply property in node /phy@fedb0000 failed
[    9.335858] phy phy-fed60000.hdmiphy.7: Looking up phy-supply from device tree
[    9.335863] phy phy-fed60000.hdmiphy.7: Looking up phy-supply property in node /hdmiphy@fed60000 failed
[    9.336567] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy init success
[    9.337431] phy phy-fed80000.phy.8: Looking up phy-supply from device tree
[    9.337437] phy phy-fed80000.phy.8: Looking up phy-supply property in node /phy@fed80000/dp-port failed
[    9.337469] phy phy-fed80000.phy.9: Looking up phy-supply from device tree
[    9.337474] phy phy-fed80000.phy.9: Looking up phy-supply property in node /phy@fed80000/u3-port failed
[    9.339011] rk-pcie fe190000.pcie: invalid prsnt-gpios property in node
[    9.339022] rk-pcie fe190000.pcie: Looking up vpcie3v3-supply from device tree
[    9.339219] pwm-backlight backlight: Looking up power-supply from device tree
[    9.339224] pwm-backlight backlight: Looking up power-supply property in node /backlight failed
[    9.339230] pwm-backlight backlight: supply power not found, using dummy regulator
[    9.339484] pwm-backlight backlight_1: Looking up power-supply from device tree
[    9.339489] pwm-backlight backlight_1: Looking up power-supply property in node /backlight_1 failed
[    9.339493] pwm-backlight backlight_1: supply power not found, using dummy regulator
[    9.339764] iep: Module initialized.
[    9.339800] mpp_service mpp-srv: 1f3f1adec author: Eng.Ahsan 2024-02-05 Remove Error in rk3588s-orangepi-5-camera1.dtsi
[   10.282462] rockchip-mipi-csi2-hw fdd10000.mipi0-csi2-hw: enter mipi csi2 hw probe!
[   10.282533] rockchip-mipi-csi2-hw fdd10000.mipi0-csi2-hw: probe success, v4l2_dev:mipi0-csi2-hw!
[   10.282567] rockchip-mipi-csi2-hw fdd20000.mipi1-csi2-hw: enter mipi csi2 hw probe!
[   10.282623] rockchip-mipi-csi2-hw fdd20000.mipi1-csi2-hw: probe success, v4l2_dev:mipi1-csi2-hw!
[   10.282653] rockchip-mipi-csi2-hw fdd30000.mipi2-csi2-hw: enter mipi csi2 hw probe!
[   10.282708] rockchip-mipi-csi2-hw fdd30000.mipi2-csi2-hw: probe success, v4l2_dev:mipi2-csi2-hw!
[   10.282741] rockchip-mipi-csi2-hw fdd40000.mipi3-csi2-hw: enter mipi csi2 hw probe!
[   10.282793] rockchip-mipi-csi2-hw fdd40000.mipi3-csi2-hw: probe success, v4l2_dev:mipi3-csi2-hw!
[   10.282824] rockchip-mipi-csi2-hw fdd50000.mipi4-csi2-hw: enter mipi csi2 hw probe!
[   10.282877] rockchip-mipi-csi2-hw fdd50000.mipi4-csi2-hw: probe success, v4l2_dev:mipi4-csi2-hw!
[   10.282906] rockchip-mipi-csi2-hw fdd60000.mipi5-csi2-hw: enter mipi csi2 hw probe!
[   10.282959] rockchip-mipi-csi2-hw fdd60000.mipi5-csi2-hw: probe success, v4l2_dev:mipi5-csi2-hw!
```

I have run following commands
```
1. v4l2-ctl --list-devices
2. lsmod | grep ov9281
3. media-ctl -p
```
having output as follow
```
Output 1 command : Cannot open device /dev/video0, exiting.
No output for 2 command
output 3 command : Failed to enumerate /dev/media0 (-2)
```

everskies, 2024-02-06:
@ahsan-kaukab 

I think this stems from using `status = "enable";`

'enable' is not a valid status keyword, should be `okay` or `disabled`

ahsan-kaukab, 2024-02-07:
After updating the dts file and ov9281 driver file
https://github.com/ahsan-kaukab/linux-rockchip-ov2981/blob/orange-pi-5-ov9281/arch/arm64/boot/dts/rockchip/rk3588s-orangepi-5-camera1.dtsi
https://github.com/Joshua-Riek/linux-rockchip/commit/897fa2f41bf283278f9384559903ee607b8fb935

Some of logs using dmesg are as follow:
```
[    9.328926] rockchip-csi2-dphy-hw fedc0000.csi2-dphy0-hw: csi2 dphy hw probe successfully!
[    9.329007] rockchip-csi2-dphy-hw fedc8000.csi2-dphy1-hw: csi2 dphy hw probe successfully!
[    9.330341] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up phy-supply from device tree
[    9.330349] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up phy-supply property in node /syscon@fd5d0000/usb2-phy@0/otg-port failed
[    9.330390] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up vbus-supply from device tree
[    9.330396] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up vbus-supply property in node /syscon@fd5d0000/usb2-phy@0/otg-port failed
[    9.331758] phy phy-fd5d8000.syscon:usb2-phy@8000.1: Looking up phy-supply from device tree
[    9.331765] phy phy-fd5d8000.syscon:usb2-phy@8000.1: Looking up phy-supply property in node /syscon@fd5d8000/usb2-phy@8000/host-port failed
[    9.332867] phy phy-fd5dc000.syscon:usb2-phy@c000.2: Looking up phy-supply from device tree
[    9.332875] phy phy-fd5dc000.syscon:usb2-phy@c000.2: Looking up phy-supply property in node /syscon@fd5dc000/usb2-phy@c000/host-port failed
[    9.334673] phy phy-fee00000.phy.3: Looking up phy-supply from device tree
[    9.334679] phy phy-fee00000.phy.3: Looking up phy-supply property in node /phy@fee00000 failed
[    9.334812] phy phy-fee20000.phy.4: Looking up phy-supply from device tree
[    9.334817] phy phy-fee20000.phy.4: Looking up phy-supply property in node /phy@fee20000 failed
[    9.335301] phy phy-feda0000.phy.5: Looking up phy-supply from device tree
[    9.335307] phy phy-feda0000.phy.5: Looking up phy-supply property in node /phy@feda0000 failed
[    9.335428] phy phy-fedb0000.phy.6: Looking up phy-supply from device tree
[    9.335433] phy phy-fedb0000.phy.6: Looking up phy-supply property in node /phy@fedb0000 failed
[    9.335796] phy phy-fed60000.hdmiphy.7: Looking up phy-supply from device tree
[    9.335802] phy phy-fed60000.hdmiphy.7: Looking up phy-supply property in node /hdmiphy@fed60000 failed
[    9.336511] rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy init success
[    9.337385] phy phy-fed80000.phy.8: Looking up phy-supply from device tree
[    9.337391] phy phy-fed80000.phy.8: Looking up phy-supply property in node /phy@fed80000/dp-port failed
[    9.337423] phy phy-fed80000.phy.9: Looking up phy-supply from device tree
[    9.337428] phy phy-fed80000.phy.9: Looking up phy-supply property in node /phy@fed80000/u3-port failed
[    9.338992] rk-pcie fe190000.pcie: invalid prsnt-gpios property in node
[    9.339003] rk-pcie fe190000.pcie: Looking up vpcie3v3-supply from device tree
[    9.339202] pwm-backlight backlight: Looking up power-supply from device tree
[    9.339208] pwm-backlight backlight: Looking up power-supply property in node /backlight failed
[    9.339213] pwm-backlight backlight: supply power not found, using dummy regulator
[    9.339464] pwm-backlight backlight_1: Looking up power-supply from device tree
[    9.339470] pwm-backlight backlight_1: Looking up power-supply property in node /backlight_1 failed
[    9.339475] pwm-backlight backlight_1: supply power not found, using dummy regulator
[    9.339732] iep: Module initialized.
[    9.339770] mpp_service mpp-srv: 4b899b576 author: Eng.Ahsan 2024-02-05 Removing error in ov9281.c
[    9.339775] rk-pcie fe190000.pcie: IRQ msi not found
[    9.339781] rk-pcie fe190000.pcie: use outband MSI support
[    9.339796] mpp_service mpp-srv: probe start
[    9.339801] rk-pcie fe190000.pcie: Missing *config* reg space
[    9.339832] rk-pcie fe190000.pcie: host bridge /pcie@fe190000 ranges:
[    9.339856] rk-pcie fe190000.pcie:      err 0x00f4000000..0x00f40fffff -> 0x00f4000000
[    9.339872] rk-pcie fe190000.pcie:       IO 0x00f4100000..0x00f41fffff -> 0x00f4100000
[    9.339890] rk-pcie fe190000.pcie:      MEM 0x00f4200000..0x00f4ffffff -> 0x00f4200000
[    9.339903] rk-pcie fe190000.pcie:      MEM 0x0a00000000..0x0a3fffffff -> 0x0a00000000
[    9.339934] rk-pcie fe190000.pcie: Missing *config* reg space
[    9.339976] rk-pcie fe190000.pcie: invalid resource
[    9.341721] mpp_vdpu2 fdb50400.vdpu: Adding to iommu group 1
[    9.341922] mpp_vdpu2 fdb50400.vdpu: probe device
[    9.342033] mpp_vdpu2 fdb50400.vdpu: reset_group->rw_sem_on=0
[    9.342039] mpp_vdpu2 fdb50400.vdpu: reset_group->rw_sem_on=0
[    9.342182] mpp_vdpu2 fdb50400.vdpu: probing finish
```

its showing error in ov9281.c file

```
[    9.906516] i2c /dev entries driver
[    9.910170] rk860-regulator 0-0042: Looking up vin-supply from device tree
[    9.910178] vdd_cpu_big0_s0: supplied by vcc5v0_sys
[    9.910184] vcc5v0_sys: could not add device link regulator.29: -ENOENT
[    9.914282] vdd_cpu_big0_s0: 550 <--> 1050 mV at 800 mV, enabled
[    9.918785] rk860-regulator 0-0043: Looking up vin-supply from device tree
[    9.918791] vdd_cpu_big1_s0: supplied by vcc5v0_sys
[    9.918795] vcc5v0_sys: could not add device link regulator.30: -ENOENT
[... 63 lines cut ...]
[   10.281462] rockchip-mipi-csi2-hw fdd60000.mipi5-csi2-hw: probe success, v4l2_dev:mipi5-csi2-hw!
[   10.282662] rockchip-mipi-csi2 mipi2-csi2: attach to csi2 hw node
[... cut ...]
[   10.458150] ov9281 7-0010: supply dovdd not found, using dummy regulator
[   10.458177] ov9281 7-0010: Looking up dvdd-supply from device tree
[   10.458186] ov9281 7-0010: Looking up dvdd-supply property in node /i2c@fec90000/ov9281-1@10 failed
[   10.458197] ov9281 7-0010: supply dvdd not found, using dummy regulator
[   10.460463] ov9281 7-0010: Unexpected sensor id(000000), ret(-5)
```

I can see that its trying to initialize ov9281 and dev/video0 is appearing in OS but not able to open the video file
[image]

ahsan-kaukab, 2024-02-08:
After doing some more driver level changes 
I can see that ov9281 is detected in OS using dmesg
```
[   10.273524] ov9281 7-0060: driver version: 00.01.05
[   10.273609] ov9281 7-0060: could not get default pinstate
[   10.273633] ov9281 7-0060: could not get sleep pinstate
[   10.273646] ov9281 7-0060: Looking up avdd-supply from device tree
[   10.273657] ov9281 7-0060: Looking up avdd-supply property in node /i2c@fec90000/ov9281-1@60 failed
[   10.273681] ov9281 7-0060: supply avdd not found, using dummy regulator
[   10.273753] ov9281 7-0060: Looking up dovdd-supply from device tree
[   10.273763] ov9281 7-0060: Looking up dovdd-supply property in node /i2c@fec90000/ov9281-1@60 failed
[   10.273775] ov9281 7-0060: supply dovdd not found, using dummy regulator
[   10.273803] ov9281 7-0060: Looking up dvdd-supply from device tree
[   10.273813] ov9281 7-0060: Looking up dvdd-supply property in node /i2c@fec90000/ov9281-1@60 failed
[   10.273824] ov9281 7-0060: supply dvdd not found, using dummy regulator
[   10.276488] ov9281 7-0060: Detected OV009281 sensor
```

The output of command `v4l2-ctl --list-devices` is as follow

[image]

Also media-ctl -p results us following
```
opi@opi-5:/dev$ media-ctl -p
Media controller API version 5.10.160

Media device information
------------------------
driver          rkcif
model           rkcif-mipi-lvds2
serial          
bus info        
hw revision     0x0
driver version  5.10.160

Device topology
- entity 1: stream_cif_mipi_id0 (1 pad, 11 links)
            type Node subtype V4L flags 0
            device node name /dev/video0
	pad0: Sink
		<- "rockchip-mipi-csi2":1 [ENABLED]
		<- "rockchip-mipi-csi2":2 []
		<- "rockchip-mipi-csi2":3 []
		<- "rockchip-mipi-csi2":4 []
		<- "rockchip-mipi-csi2":5 []
		<- "rockchip-mipi-csi2":6 []
		<- "rockchip-mipi-csi2":7 []
		<- "rockchip-mipi-csi2":8 []
		<- "rockchip-mipi-csi2":9 []
		<- "rockchip-mipi-csi2":10 []
		<- "rockchip-mipi-csi2":11 []
```
But when we try to capture image using fswebcam , we get following
[image]

@everskies do you know how to test it ?

everskies, 2024-02-08:
@ahsan-kaukab 

Try 

`sudo gst-launch-1.0 v4l2src num-buffers=1 device=/dev/video0 ! jpegenc ! filesink location=picture.jpg`

closed 2024-02-09 by ahsan-kaukab

GONGZHEMG, 2024-03-03:
> Hello @Joshua-Riek @everskies Thanks for the help
> 
> ### OV-9281 is now working fine for Orange-Pi-5
> **Steps for enabling OV-9281 are as follow: (May It help some one else in future)** (NOTE: For me the module is connected on CSI-1) **_1. Fork the following repos_** https://github.com/ahsan-kaukab/ubuntu-rockchip-ov2981 https://github.com/Joshua-Riek/linux-rockchip
> 
> _**2. Enable camera in defconfig**_ enable ov9281 module in arch/arm64/configs/rockchip_linux_defconfig `CONFIG_VIDEO_OV9281=y` Reference:: [Joshua-Riek/linux-rockchip@791f6d2](https://github.com/Joshua-Riek/linux-rockchip/commit/791f6d25c0318b876ca23688b3939a2b40137b58)
> 
> _**3. Updating DTSI**_ Update dtsi to add support for OV9281 in following file arch/arm64/boot/dts/rockchip/rk3588s-orangepi-5-camera1.dtsi New rk3588s-orangepi-5-camera1.dtsi file is as follow:
> 
> ```
> // SPDX-License-Identifier: (GPL-2.0+ OR MIT)
> /*
>  * Copyright (c) 2021 Rockchip Electronics Co., Ltd.
>  *
>  */
> 
> &csi2_dphy0_hw {
> 	status = "okay";
> };
> 
> &csi2_dphy0 {
> 	status = "okay";
> 	ports {
> 		#address-cells = <1>;
> 		#size-cells = <0>;
> 		port@0 {
> 			reg = <0>;
> 			#address-cells = <1>;
> 			#size-cells = <0>;
> 
> 			mipi_in_ucam0: endpoint@0 {
> 				reg = <0>;
> 				remote-endpoint = <&ov9281_out2>;
> 				data-lanes = <1>;
> 			};
> 		};
> 		port@1 {
> 		        reg = <1>;
> 		        #address-cells = <1>;
> 		        #size-cells = <0>;
> 		        csidphy0_out: endpoint@0 {
> 		                reg = <0>;
> 		                remote-endpoint = <&mipi2_csi2_input>;
> 		        };
> 		};
> 	};
> };
> 
> &i2c7 {
> 	status = "okay";
> 	pinctrl-names = "default";
> 	pinctrl-0 = <&i2c7m0_xfer>;
> 
> 	vm149c_p1: vm149c-p1@c {
> 		compatible = "silicon touch,vm149c";
> 		status = "disabled";
> 		reg = <0x0c>;
> 		rockchip,camera-module-index = <1>;
> 		rockchip,camera-module-facing = "back";
> 	};
> 
> 	ov9281_1: ov9281-1@60 {
> 		compatible = "ovti,ov9281";
> 		status = "okay";
> 		reg = <0x60>;
> 		clocks = <&cru CLK_MIPI_CAMARAOUT_M3>;
> 		clock-names = "xvclk";
> 		pinctrl-names = "default";
> 		pinctrl-0 = <&mipim0_camera3_clk>;
> 		reset-gpios = <&gpio3 RK_PC4 GPIO_ACTIVE_HIGH>;
[... 132 lines cut ...]
> 
>     while True:
>         ret, frame = cap.read()
>         if not ret:
>             print("Error: Couldn't read frame.")
>             break
> 
>         # Display the frame
>         cv2.imshow('Frame', frame)
> 
>         # Exit when 'q' is pressed
>         if cv2.waitKey(1) & 0xFF == ord('q'):
>             break
> 
>     # Release the video capture device and close the OpenCV windows
>     cap.release()
>     cv2.destroyAllWindows()
> 
> if __name__ == "__main__":
>     main()
> ```
> 
> camera should be showing output like this ![image](https://private-user-images.githubusercontent.com/71211526/303691534-8f9fe6ad-7855-42c4-9b6b-135db351673f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk0Nzk2NDYsIm5iZiI6MTcwOTQ3OTM0NiwicGF0aCI6Ii83MTIxMTUyNi8zMDM2OTE1MzQtOGY5ZmU2YWQtNzg1NS00MmM0LTliNmItMTM1ZGIzNTE2NzNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzAzVDE1MjIyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM3ODU1MzY5ZjRkYmMyYmJkMjIwNTc3ZjE1NDM4YjAxZTQzZDdmZDdmYmFmZjNmMDQxNGU0NjJiNWQzMzAzZmUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.WpyD2H6xRXvbdL9_GXQZqZwAJpR3XexkjXkCgjYytec)

ohhhh，I'm also doing a driver on OrangePi5 about OV9281, some time ago I did a kernel compilation on my development board, but I had a problem with the sensor module PCB soldering, and I can't solder the sensor module and test it. Can you provide your sensor for my reference? There is also some problem with whether your image is displayed, and you need 3A or ISP assistant software. I got it before. You can contact me to develop together. My email: 2272599707@qq.com.

GONGZHEMG, 2024-03-03:
Hello@ahsan-kaukab  Thank you for the help！
I'm also doing a driver on OrangePi5 about OV9281, some time ago I did a kernel compilation on my development board, but I had a problem with the sensor module PCB soldering, and I can't solder the sensor module and test it. Can you provide your sensor for my reference? There is also some problem with whether your image is displayed, and you need 3A or ISP assistant software. I got it before. You can contact me to develop together. My email: 2272599707@qq.com.

GONGZHEMG, 2024-03-03:
Hello @ahsan-kaukab thanks for the help.
I'm also doing a driver on OrangePi5 about OV9281, some time ago I did a kernel compilation on my development board, but I had a problem with the sensor module PCB soldering, and I can't solder the sensor module and test it. Can you provide your sensor for my reference? There is also some problem with whether your image is displayed, and you need 3A or ISP assistant software. I got it before. You can contact me to develop together. My email: 2272599707@qq.com.

mdurrani808, 2024-04-15:
Hi @everskies, a question unrelated to the issue here but can you provide more information as to the exact OV9281 hardware you used to interface with the orange pi? It was my understanding that most OV9281 cameras have a different CSI pinout than that of the orange pi.

mdurrani808, 2024-04-15:
@ahsan-kaukab if you have any insights that would be appreciated as well, re: the above.

mdurrani808, 2024-04-15:
> > Hi @everskies, a question unrelated to the issue here but can you provide more information as to the exact OV9281 hardware you used to interface with the orange pi? It was my understanding that most OV9281 cameras have a different CSI pinout than that of the orange pi.
> 
> It was a custom PCB

Ahh I see. Any chance that PCB is open source or is it closed source?
Thanks!

cyclonev, 2024-08-28:
@ahsan-kaukab 
Hello, I can take photos using the OV9281, but when I use the v4l2 -l command, no parameters are displayed. I can take photos and record videos using the Cheese app in Ubuntu, but the images are green. I checked and it seems to be due to the lack of corresponding IQ files. Does your system capture images correctly without ISP adjustments? Do you have any OV9281-related JSON configuration files under /etc/iqfiles/   in your system? If so, could you upload them for me to try out? Thank you very much!

ronjacoby, 2025-06-22:
hi, im trying to connect an adv7282-m to rk3588.its a single lane connected to dphy1 clk0 ,d0 lane.
the adv is working and outputting data, but i cant get data from /dev/video device
im trying to understand the logic of how to select which mipiN_csi2 and csi2_dphyN to use,
here's the current dts

&i2c1 {
	pinctrl-names = "default";
	pinctrl-0 = <&i2c1m2_xfer>;
	status = "okay";

	adv7282@21 {
		compatible = "adi,adv7282-m";
		status = "okay";
		reg = <0x21>;
		powerdown-gpios = <&tca6408_saturn_cvbs PCA_IO0_2 GPIO_ACTIVE_LOW>;
		reset-gpios = <&tca6408_saturn_cvbs PCA_IO0_3 GPIO_ACTIVE_HIGH>;
		rockchip,camera-module-index = <1>;
		rockchip,camera-module-facing = "front";
		rockchip,camera-module-name = "ADV7282";
		rockchip,camera-module-lens-name = "NC";
		port {
			adv7282_sd_2: endpoint {
				remote-endpoint = <&mipi_in_sd_2>;
				data-lanes = <1>;
			};
		};
	};

};

&csi2_dphy3 {
	status = "okay";

	ports {
		#address-cells = <1>;
		#size-cells = <0>;
		port@0 {
			reg = <0>;
			#address-cells = <1>;
			#size-cells = <0>;

			mipi_in_sd_2: endpoint@1 {
				reg = <1>;
				remote-endpoint = <&adv7282_sd_2>;
				data-lanes = <1>;
			};
		};
		port@1 {
			reg = <1>;
			#address-cells = <1>;
			#size-cells = <0>;

			csidphy3_out: endpoint@0 {
				reg = <0>;
				remote-endpoint = <&mipi3_csi2_input>;
			};
		};
	};
};

&csi2_dphy0_hw {
	status = "okay";
};

&csi2_dphy1_hw {
	status = "okay";
};
&mipi3_csi2 {
    status = "okay";

    ports {
        #address-cells = <1>;
        #size-cells = <0>;

        port@0 {
            reg = <0>;
            #address-cells = <1>;
            #size-cells = <0>;

            mipi3_csi2_input: endpoint@1 {
                reg = <1>;
                remote-endpoint = <&csidphy3_out>;
            };
        };

        port@1 {
            reg = <1>;
            #address-cells = <1>;
            #size-cells = <0>;

            mipi3_csi2_output: endpoint@0 {
                reg = <0>;
                remote-endpoint = <&cif_mipi3_in0>;
            };
        };
    };
};

&rkcif {
	status = "okay";
};

&rkcif_mmu {
	status = "okay";
};

&rkcif_mipi_lvds3 {
	status = "okay";

	port {
		cif_mipi3_in0: endpoint {
			remote-endpoint = <&mipi3_csi2_output>;
		};
	};
};