Joshua-Riek/ubuntu-rockchip #581: Patch to solve the problem of green screen when playing videos

Z-Keven, 2024-01-26:
Hello, I'm having trouble solving a problem and need your help. A green screen appeared when I played the video, so I installed the deb package of rk3588's libmali on the board. The deb package is in the attachment. After installation, the green screen will no longer appear when playing videos, but when I restart the board, I cannot enter the desktop system. Can you help me find out what is the reason?

Z-Keven, 2024-01-26:
This is the deb package of libmali. My installation command is:
sudo dpkg -i libmali-valhall-g610-g13p0-x11-gbm_1.9-1_arm64.deb
sudo dpkg -i libmali-valhall-g610-g13p0-x11-gbm-dbgsym_1.9-1_arm64.deb
[Uploading RK3588-libmali.zip…]()

Z-Keven, 2024-01-26:
This is the deb package of libmali. My installation command is: . You can also try
sudo dpkg -i libmali-valhall-g610-g13p0-x11-gbm_1.9-1_arm64.deb
sudo dpkg -i libmali-valhall-g610-g13p0-x11-gbm-dbgsym_1.9-1_arm64.deb
[RK3588-libmali.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/14061631/RK3588-libmali.zip)

Z-Keven, 2024-01-26:
I hope you can install these two deb packages on the RK3888 board before trying to play the video. The green screen problem has been solved, but I can’t access the desktop after restarting.

Joshua-Riek, 2024-01-28:
These blobs will conflict with panfork and are outdated with the latest rkr6 kernel (this is why the desktop does not work). However, a workaround can be to install a chrome extension `h264ify`.

https://github.com/Joshua-Riek/ubuntu-rockchip/issues/270#issuecomment-1862781481

closed 2024-01-28 by Joshua-Riek

Z-Keven, 2024-01-30:
If I want to install the deb package after setting the username and password on the first system startup, where should I add the dpkg -i command in the source code?