Joshua-Riek/ubuntu-rockchip #588: iwlwifi/cfg80211/mac80211

Xeue, 2024-01-28:
I've been stuggling to get an intel 210AX chip working properly and think I've narrowed it down.
I can conenct to an SSID, but it wont roam and once disconnected can't reconnect without disabling the device and re-enabling it again (rfkill etc)
I've had it working just fine on the friendly elec 6.1 kernel build (nanopc T6), so ported the same firmware and iw/iwconfig builds over and it's made no differance.

However, looking at lsmod, I can see that cfg80211 and mac80211 are not listed, but they do show up under modinfo.
The errors I'm seeing all point towards these modules and a few things online talk about how these modules need to be loaded as normal modules and not built directly into the kernel. However, I don't know enough about any of this stuff to really understand what that means, but I'm hoping someone can take a look/point me in the right direction on either how to fix this or how I can get more info/logs to be useful to someone!

Joshua-Riek, 2024-01-28:
Take a look here: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/563#issuecomment-1892690075 it's a known issue on ubuntu.

Joshua-Riek, 2024-01-28:
Try to `sudo modprobe cfg80211` and `sudo modprobe mac80211`?

Joshua-Riek, 2024-01-28:
Also the kernel used by Friendlyelec is from an experimental dev branch, it's a poor choice on their part to release it.

Xeue, 2024-01-28:
That hasn't made any difference in afraid, it Connect's first time just fine, but won't roam and won't re-connect once it's disconnected...

Yeah I'm aware of the questionable kernel choice, I just wanted to cross reference if it's a build thing or a board/hardware thing!

Joshua-Riek, 2024-01-28:
It is likely a kernel thing, I'm very limited with supporting hardware because the current 5.10 kernel is Android. Even when the 6.1 kernel comes out, it will be hacked to bits. I also spent the last week debugging WiFi + Bluetooth drivers for 2 other boards, and I'm burnt out on debugging that stuff right now.

Xeue, 2024-01-28:
I completely understand, thanks for helping! If you have a pointers for things I could look at, I'll see what I can do.
I'll keep investigating and I'll comment if I get anywhere!

Joshua-Riek, 2024-01-28:
My guess is some kernel modules is not enabled for cfg80211 or mac80211. A quick look and i see it enabled, so it could be some hidden config or something else.

https://github.com/Joshua-Riek/linux-rockchip/blob/a85fe0f1699ab4292e518106b7bc9379d5d44c26/arch/arm64/configs/rockchip_linux_defconfig

dexterphillips, 2024-02-03:
I'm having the same issue here! 

Board: **NanoPC T6 16GB**
Running latest Joshua Desktop Build

Using an intel ax210 chip, wifi will connect but not roam to another access point of the same SSID. From the logs, this seems to be mac address authentication trouble.

_I have tried using the friendlyelec focal ubuntu kernel 6.1 build. Wifi roams **correctly** using this build.... But there's other stuff wrong with their image that is way better on josh's image, so I'd love to get this working._

_I have tried using the armbian build, and have the same issues as in the josh build. I've tried recompiling using the armbian builder and loaded different combinations of networking modules without any success._

**What i've found so far:** 
1 - Once disconnected from an access point, it never reconnects to any Access point: not the one it came from, and not another one nearby.
2 - Doing an rfkill block <interface>, followed by an rfkill unblock <interface> makes it "roam"... well, reconnect to the highest strength BSSID.

3 - It appears that the major difference between Josh's build and the Friendly Elec builds is that in the friendly build, it loads cfg80211 and mac80211 as a module, rather than loading it as a built-in kernel module. I wonder if this is somehow related? 

Does anyone have any thoughts on how to proceed with debugging/fixing this? I'd love to see this working on josh's 5.10 jammy build :)