Joshua-Riek/ubuntu-rockchip #589: Orange PI 5B Ubuntu 20.04 PWN not working

redradist, 2024-01-28:
Hi all,

I've faced with the issue that if I try to run PWM program in python first it shows a permission error.
After fixing it with the following `rules` file:
```console
SUBSYSTEM=="pwm*", PROGRAM="/bin/sh -c '\
        chown -R root:gpiouser /sys/class/pwm/pwmchip0 && chmod -R 770 /sys/class/pwm/pwmchip0;\
        chown -R root:gpiouser /sys/class/pwm/pwmchip1 && chmod -R 770 /sys/class/pwm/pwmchip1;\
        chown -R root:gpiouser /sys/devices/platform/*.pwm/pwm/pwmchip* && chmod -R 770 /sys/devices/platform/*.pwm/pwm/pwmchip*\
'"
```
It start showing the following error:
```console
chip_path = /sys/class/pwm/pwmchip0/export
січ 28 18:08:55 orangepi-desktop python3[4060]: OSError: [Errno 19] No such device
січ 28 18:08:55 orangepi-desktop python3[4060]: During handling of the above exception, another exception occurred:
січ 28 18:08:55 orangepi-desktop python3[4060]: Traceback (most recent call last):
січ 28 18:08:55 orangepi-desktop python3[4060]:   File "/home/orangepi/.local/lib/python3.10/site-packages/periphery/pwm.py", line 66, in _open
січ 28 18:08:55 orangepi-desktop python3[4060]:     with open(os.path.join(chip_path, "export"), "w") as f_export:
січ 28 18:08:55 orangepi-desktop python3[4060]: OSError: [Errno 19] No such device
січ 28 18:08:55 orangepi-desktop python3[4060]: During handling of the above exception, another exception occurred:
січ 28 18:08:55 orangepi-desktop python3[4060]: Traceback (most recent call last):
січ 28 18:08:55 orangepi-desktop python3[4060]:   File "/home/orangepi/./fpv_58g_video_capture/cv2_fpv_58g_scanner.py", line 25, in <module>
січ 28 18:08:55 orangepi-desktop python3[4060]:     pwm = PWM(0, 15)
січ 28 18:08:55 orangepi-desktop python3[4060]:   File "/home/orangepi/.local/lib/python3.10/site-packages/periphery/pwm.py", line 39, in __init__
січ 28 18:08:55 orangepi-desktop python3[4060]:     self._open(chip, channel)
січ 28 18:08:55 orangepi-desktop python3[4060]:   File "/home/orangepi/.local/lib/python3.10/site-packages/periphery/pwm.py", line 69, in _open
січ 28 18:08:55 orangepi-desktop python3[4060]:     raise PWMError(e.errno, "Exporting PWM channel: " + e.strerror)
січ 28 18:08:55 orangepi-desktop python3[4060]: periphery.pwm.PWMError: [Errno 19] Exporting PWM channel: No such device
```

But the device `/sys/class/pwm/pwmchip0/export` exists, but I still cannot access it

Joshua-Riek, 2024-01-28:
This project does not support Ubuntu 20.04.

redradist, 2024-01-28:
> This project does not support Ubuntu 20.04.

@Joshua-Riek Oh, sorry, I've forgot that I've installed Ubuntu 22.04, `ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-5b.img.xz`

Joshua-Riek, 2024-01-28:
I mean, looks like a user error. /sys/class/pwm/pwmchip0 and /sys/class/pwm/pwmchip1 could not exist?

redradist, 2024-01-28:
> I mean, looks like a user error. /sys/class/pwm/pwmchip0 and /sys/class/pwm/pwmchip1 could not exist?

@Joshua-Riek They exists:
```console
orangepi@orangepi-desktop:~$ ls -lsa /sys/class/pwm/
total 0
0 drwxr-xr-x  2 root root     0 січ  1  2021 .
0 drwxr-xr-x 78 root root     0 січ  1  2021 ..
0 lrwxrwxrwx  1 root gpiouser 0 січ 28 18:17 pwmchip0 -> ../../devices/platform/fd8b0020.pwm/pwm/pwmchip0
0 lrwxrwxrwx  1 root gpiouser 0 січ 28 18:17 pwmchip1 -> ../../devices/platform/febd0020.pwm/pwm/pwmchip1
orangepi@orangepi-desktop:~$
```

Joshua-Riek, 2024-01-28:
Does /sys/class/pwm/pwmchip0/export exist?

redradist, 2024-01-28:
> Does /sys/class/pwm/pwmchip0/export exist?

Yes:
```console
orangepi@orangepi-desktop:~$ ls -lsa /sys/class/pwm/pwmchip0/export 
0 -rwxrwx--- 1 root gpiouser 4096 січ 28 20:06 /sys/class/pwm/pwmchip0/export
orangepi@orangepi-desktop:~$
```

Joshua-Riek, 2024-01-28:
What python script is being run?
```
chip_path = /sys/class/pwm/pwmchip0/export
січ 28 18:08:55 orangepi-desktop python3[4060]: OSError: [Errno 19] No such device
січ 28 18:08:55 orangepi-desktop python3[4060]: During handling of the above exception, another exception occurred:
січ 28 18:08:55 orangepi-desktop python3[4060]: Traceback (most recent call last):
січ 28 18:08:55 orangepi-desktop python3[4060]:   File "/home/orangepi/.local/lib/python3.10/site-packages/periphery/pwm.py", line 66, in _open
січ 28 18:08:55 orangepi-desktop python3[4060]:     with open(os.path.join(chip_path, "export"), "w") as f_export:
```

redradist, 2024-01-28:
> What python script is being run?
>
> ```
> chip_path = /sys/class/pwm/pwmchip0/export
> січ 28 18:08:55 orangepi-desktop python3[4060]: OSError: [Errno 19] No such device
> січ 28 18:08:55 orangepi-desktop [...]

Here it is:
```python
from periphery import GPIO, PWM

# Open PWM chip 0, channel 10
pwm = PWM(0, 1)

# Set frequency to 1 kHz
pwm.frequency = 1e3
# Set duty cycle to 75%
pwm.duty_cycle = 0.75

pwm.enable()
```

Joshua-Riek, 2024-01-28:
Are you sure that is the correct pwm chip and channel? Have you looked at the device tree overlays?

https://github.com/Joshua-Riek/linux-rockchip/tree/linux-5.10-gen-rkr6/arch/arm64/boot/dts/rockchip/overlay

redradist, 2024-01-28:
> Are you sure that is the correct pwm chip and channel? Have you looked at the device tree overlays?
> 
> https://github.com/Joshua-Riek/linux-rockchip/tree/linux-5.10-gen-rkr6/arch/arm64/boot/dts/rockchip/overlay

But, it crashed on `/sys/class/pwm/pwmchip0/export` according the source files and this file exists ...

Joshua-Riek, 2024-01-28:
It cashed because the device does not exist, I don't have time to dig into the schematics for something I'm quite sure is a user error. The path `/sys/class/pwm/pwmchip0/export` is correct, the issue is the PWM channel you are using is not valid.

redradist, 2024-01-28:
> It cashed because the device does not exist, I don't have time to dig into the schematics for something I'm quite sure is a user error. The path `/sys/class/pwm/pwmchip0/export` is correct, the issue is the PWM channel you are using is not valid.

It is not the channel, because according the source code of library, channel should be in `/sys/class/pwm/pwmchip0/pwm15`, but it will be available after opening the `/sys/class/pwm/pwmchip0/export`
Okay, thanks for you time, I'll figured out the root cause !!

redradist, 2024-01-28:
Also I strange issue is that:
```bash
orangepi@orangepi-desktop:~$ sudo gpio readall
 +------+-----+----------+--------+---+   OPI5   +---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 |      |     |     3.3V |        |   |  1 || 2  |   |        | 5V       |     |      |
 |   47 |   0 |    SDA.5 |    OUT | 1 |  3 || 4  |   |        | 5V       |     |      |
 |   46 |   1 |    SCL.5 |    OUT | 1 |  5 || 6  |   |        | GND      |     |      |
 |   54 |   2 |    PWM15 |     IN | 1 |  7 || 8  | 0 | IN     | RXD.0    | 3   | 131  |
 |      |     |      GND |        |   |  9 || 10 | 0 | IN     | TXD.0    | 4   | 132  |
 |  138 |   5 |  CAN1_RX |     IN | 1 | 11 || 12 | 1 | IN     | CAN2_TX  | 6   | 29   |
 |  139 |   7 |  CAN1_TX |     IN | 1 | 13 || 14 |   |        | GND      |     |      |
 |   28 |   8 |  CAN2_RX |     IN | 1 | 15 || 16 | 1 | IN     | SDA.1    | 9   | 59   |
 |      |     |     3.3V |        |   | 17 || 18 | 1 | IN     | SCL.1    | 10  | 58   |
 |   49 |  11 | SPI4_TXD |     IN | 1 | 19 || 20 |   |        | GND      |     |      |
 |   48 |  12 | SPI4_RXD |     IN | 1 | 21 || 22 | 1 | ALT1   | GPIO2_D4 | 13  | 92   |
 |   50 |  14 | SPI4_CLK |     IN | 1 | 23 || 24 | 1 | IN     | SPI4_CS1 | 15  | 52   |
 |      |     |      GND |        |   | 25 || 26 | 1 | IN     | PWM1     | 16  | 35   |
 +------+-----+----------+--------+---+----++----+---+--------+----------+-----+------+
 | GPIO | wPi |   Name   |  Mode  | V | Physical | V |  Mode  | Name     | wPi | GPIO |
 +------+-----+----------+--------+---+   OPI5   +---+--------+----------+-----+------+
orangepi@orangepi-desktop:~$ sudo gpio mode 2 pwm
the pin you choose doesn't support hardware PWM
you can select wiringPi pin 42 for PWM pin
or you can use it in softPwm mode
```
It is not possible to set pin 2 of Orange Pi 5 for PWM, but according the table it should be supported

Joshua-Riek, 2024-01-28:
The wiringpi lib is outdated and i see they re-work pwm support. I'm building the package now.

https://launchpad.net/~jjriek/+archive/ubuntu/rockchip/+build/27690759

redradist, 2024-01-28:
@Joshua-Riek 

Looks like this issue is not in `/sys/class/pwm/pwmchip0/export` file.
I've added the following print statement after file is open and looks like it opened successfully:
```python
            try:
                print(f"chip_path = {os.path.join(chip_path, 'export')}")
                with open(os.path.join(chip_path, "export"), "w") as f_export:
                    print(f"Try the following channel = {channel}")
                    f_export.write("{:d}\n".format(channel))
            except IOError as e:
                print(f">>> Exporting PWM channel: errno = {e.errno}, strerror = {e.strerror}")
                raise PWMError(e.errno, "Exporting PWM channel: " + e.strerror)
```
and `Try the following channel = ...` is printed ...

After that I decided that you is right and I have wrong channel, I decided to get first successful PWM channel using the following trick ))) :
```python
success_chip_channels = []
for chip in range(100):
    for channel in range(640):
        try:
            pwm = PWM(chip, channel)
            success_chip_channels.append((chip, channel))
        except:
            pass
```
But there are no  `success_chip_channels`, it has size 0 ...

Also here is my `ubuntuEnv` file, maybe I have missed overlays ... ?
```console
orangepi@orangepi-desktop:~$ sudo cat /boot/firmware/ubuntuEnv.txt 
bootargs=root=UUID=0ad21eec-033c-43ed-9f29-413f88df5225 rootfstype=ext4 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 systemd.unified_cgroup_hierarchy=0 quiet splash plymouth.ignore-serial-consoles
fdtfile=rk3588s-orangepi-5b.dtb
overlay_prefix=rk3588
overlays=
```

redradist, 2024-01-29:
> Are you sure that is the correct pwm chip and channel? Have you looked at the device tree overlays?
> 
> https://github.com/Joshua-Riek/linux-rockchip/tree/linux-5.10-gen-rkr6/arch/arm64/boot/dts/rockchip/overlay

@Joshua-Riek Looks I've figured out the issue ...
You in `ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-5b.img.xz` has the following path for overlays `/boot/firmware/dtbs/rockchip/*.dtb`
In this directory I have missed the `rk3588-pwm*-m*.dtb` files !!
After adding at least one to `rk3588-pwm*-m*.dtb` file to `/boot/firmware/dtbs/rockchip/` folder, this dummy python code:
```python
success_chip_channels = []
for chip in range(100):
    for channel in range(640):
        try:
            pwm = PWM(chip, channel)
            success_chip_channels.append((chip, channel))
        except:
            pass
```
found at least on PWM chip = 0 with channel = 0

closed 2024-01-29 by Joshua-Riek

redradist, 2024-01-29:
@Joshua-Riek Could you add this files in future builds of this image ??

Joshua-Riek, 2024-01-29:
The device tree overlays exist already, I do not understand what you are asking.

Joshua-Riek, 2024-01-29:
The files do exist as overlays, you do not understand that they are compiled as .dtbo files.

redradist, 2024-01-29:
> The files do exist as overlays, you do not understand that they are compiled as .dtbo files.

Why them was missed in folder `/boot/firmware/dtbs/rockchip/` ?

Joshua-Riek, 2024-01-29:
They are in /boot/firmware/dtbs/rockchip/ but they do not go by .dts, they are . dtbo.

redradist, 2024-01-29:
> They are in /boot/firmware/dtbs/rockchip/ but they do not go by `*.dts`, they are `*.dtbo`.

Oh, I see, yes they already in `/boot/firmware/dtbs/rockchip/overlay/`:
```bash
orangepi@orangepi-desktop:~$ ls -lsa /boot/firmware/dtbs/rockchip/overlay/ | grep pwm
 4 -rwxr-xr-x 1 root root   343 січ 29 21:11 rk3588-pwm0-m0.dtbo
 4 -rwxr-xr-x 1 root root   309 січ 29 21:11 rk3588-pwm0-m1.dtbo
 4 -rwxr-xr-x 1 root root   309 січ 29 21:11 rk3588-pwm0-m2.dtbo
 4 -rwxr-xr-x 1 root root   343 січ 29 21:11 rk3588-pwm1-m0.dtbo
 4 -rwxr-xr-x 1 root root   309 січ 29 21:11 rk3588-pwm1-m1.dtbo
 4 -rwxr-xr-x 1 root root   309 січ 29 21:11 rk3588-pwm1-m2.dtbo
 4 -rwxr-xr-x 1 root root   311 січ 29 21:11 rk3588-pwm10-m0.dtbo
 4 -rwxr-xr-x 1 root root   311 січ 29 21:11 rk3588-pwm11-m0.dtbo
 4 -rwxr-xr-x 1 root root   543 січ 29 21:11 rk3588-pwm11-m1.dtbo
 4 -rwxr-xr-x 1 root root   311 січ 29 21:11 rk3588-pwm12-m0.dtbo
 4 -rwxr-xr-x 1 root root   311 січ 29 21:11 rk3588-pwm13-m0.dtbo
 4 -rwxr-xr-x 1 root root   311 січ 29 21:11 rk3588-pwm13-m2.dtbo
 4 -rwxr-xr-x 1 root root   345 січ 29 21:11 rk3588-pwm14-m0.dtbo
 4 -rwxr-xr-x 1 root root   345 січ 29 21:11 rk3588-pwm14-m1.dtbo
 4 -rwxr-xr-x 1 root root   311 січ 29 21:11 rk3588-pwm14-m2.dtbo
 4 -rwxr-xr-x 1 root root   543 січ 29 21:11 rk3588-pwm15-m0.dtbo
 4 -rwxr-xr-x 1 root root   311 січ 29 21:11 rk3588-pwm15-m1.dtbo
 4 -rwxr-xr-x 1 root root   311 січ 29 21:11 rk3588-pwm15-m2.dtbo
 4 -rwxr-xr-x 1 root root   587 січ 29 21:11 rk3588-pwm15-m3.dtbo
 4 -rwxr-xr-x 1 root root   541 січ 29 21:11 rk3588-pwm2-m1.dtbo
 4 -rwxr-xr-x 1 root root   309 січ 29 21:11 rk3588-pwm3-m0.dtbo
 4 -rwxr-xr-x 1 root root   541 січ 29 21:11 rk3588-pwm3-m1.dtbo
 4 -rwxr-xr-x 1 root root   309 січ 29 21:11 rk3588-pwm3-m2.dtbo
 4 -rwxr-xr-x 1 root root   309 січ 29 21:11 rk3588-pwm3-m3.dtbo
 4 -rwxr-xr-x 1 root root   541 січ 29 21:11 rk3588-pwm5-m2.dtbo
 4 -rwxr-xr-x 1 root root   541 січ 29 21:11 rk3588-pwm6-m0.dtbo
 4 -rwxr-xr-x 1 root root   541 січ 29 21:11 rk3588-pwm6-m2.dtbo
 4 -rwxr-xr-x 1 root root   541 січ 29 21:11 rk3588-pwm7-m0.dtbo
 4 -rwxr-xr-x 1 root root   541 січ 29 21:11 rk3588-pwm7-m3.dtbo
 4 -rwxr-xr-x 1 root root   541 січ 29 21:11 rk3588-pwm8-m0.dtbo
```