Joshua-Riek/ubuntu-rockchip #592: Orange Pi 3B - built in WiFi not working

lukaszsobala, 2024-01-29:
Using the latest Ubuntu build (#184), the built in WiFi is not recognized.

Might be related to the absence of the kernel module in the Ubuntu kernel (see: https://github.com/MichaIng/DietPi/issues/6659#issue-1922158975)

Joshua-Riek, 2024-01-29:
Try latest build then update apt packages.

lukaszsobala, 2024-01-29:
Thanks, apt upgrade alone helped (not tried before because of no Internet).

Joshua-Riek, 2024-01-29:
I was missing firmware and was applied here:

https://github.com/Joshua-Riek/firmware

closed 2024-01-29 by Joshua-Riek

defencedog, 2024-07-22:
@Joshua-Riek which of these firmware particularly relates to orangepi-3b? Because my WiFi is working however BT is not ... these modules if activated should get both operational or am I not correct?