Joshua-Riek/ubuntu-rockchip #594: USB 5.8GHz Arkmicro Technologies S-Video Capture Device not always detected

redradist, 2024-01-30:
Hi @Joshua-Riek,

After latest `sudo apt update && sudo apt upgrade` command, I've started noticed that my USB 5.8GHz receiver is not always detected by Ubuntu `ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-5b.img.xz`

Here the what I have seen after bunch of reconnection to USB port, after some number of attempts is start recognizing the USB device:
```bash
orangepi@orangepi-desktop:~$ lsusb 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 24ae:2013 Shenzhen Rapoo Technology Co., Ltd. Rapoo 2.4G Wireless Device
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
orangepi@orangepi-desktop:~$ lsusb 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 24ae:2013 Shenzhen Rapoo Technology Co., Ltd. Rapoo 2.4G Wireless Device
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
orangepi@orangepi-desktop:~$ lsusb 
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 010: ID 18ec:5850 Arkmicro Technologies Inc. CVBS / S-Video Capture Device [UVC]
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 24ae:2013 Shenzhen Rapoo Technology Co., Ltd. Rapoo 2.4G Wireless Device
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

First I thought that maybe it is issue with USB cable. I changed different cables, but nothing was working.
Also I've connected USB receiver to my Mac and it was working from the first time (verified number of times)

Looks like it started happens after latest `sudo apt update && sudo apt upgrade` command.
Here is my results of `dmesg | grep -i USB` command, maybe it will give some clue what happens ...

```bash
orangepi@orangepi-desktop:~$ dmesg | grep -i USB
[    8.375267] reg-fixed-voltage vcc5v0-usbdcin: Looking up vin-supply from device tree
[    8.375270] vcc5v0_usbdcin: supplied by vcc12v_dcin
[    8.375288] vcc5v0_usbdcin: 5000 mV, enabled
[    8.375341] reg-fixed-voltage vcc5v0-usbdcin: vcc5v0_usbdcin supplying 5000000uV
[    8.375381] reg-fixed-voltage vcc5v0-usb: Looking up vin-supply from device tree
[    8.375384] vcc5v0_usb: supplied by vcc5v0_usbdcin
[    8.375387] vcc5v0_usbdcin: could not add device link regulator.4: -ENOENT
[    8.375401] vcc5v0_usb: 5000 mV, enabled
[    8.375457] reg-fixed-voltage vcc5v0-usb: vcc5v0_usb supplying 5000000uV
[    8.375622] vbus5v0_typec: supplied by vcc5v0_usb
[    8.436576] usbcore: registered new interface driver usbfs
[    8.436590] usbcore: registered new interface driver hub
[    8.436602] usbcore: registered new device driver usb
[    9.229985] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up phy-supply from device tree
[    9.229992] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up phy-supply property in node /syscon@fd5d0000/usb2-phy@0/otg-port failed
[    9.230032] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up vbus-supply from device tree
[    9.230038] phy phy-fd5d0000.syscon:usb2-phy@0.0: Looking up vbus-supply property in node /syscon@fd5d0000/usb2-phy@0/otg-port failed
[    9.231267] phy phy-fd5d8000.syscon:usb2-phy@8000.1: Looking up phy-supply from device tree
[    9.231274] phy phy-fd5d8000.syscon:usb2-phy@8000.1: Looking up phy-supply property in node /syscon@fd5d8000/usb2-phy@8000/host-port failed
[    9.232331] phy phy-fd5dc000.syscon:usb2-phy@c000.2: Looking up phy-supply from device tree
[    9.232338] phy phy-fd5dc000.syscon:usb2-phy@c000.2: Looking up phy-supply property in node /syscon@fd5dc000/usb2-phy@c000/host-port failed
[    9.680563] usbcore: registered new interface driver rndis_wlan
[    9.680729] usbcore: registered new interface driver asix
[... 120 lines cut ...]
[  597.406970] usb 7-1: device descriptor read/64, error -71
[  597.636978] usb 7-1: device descriptor read/64, error -71
[  597.743766] usb usb7-port1: attempt power cycle
[  598.146988] usb 7-1: new full-speed USB device number 8 using xhci-hcd
[  598.147129] usb 7-1: Device not responding to setup address.
[  598.353694] usb 7-1: Device not responding to setup address.
[  598.560263] usb 7-1: device not accepting address 8, error -71
[  598.683620] usb 7-1: new full-speed USB device number 9 using xhci-hcd
[  598.683708] usb 7-1: Device not responding to setup address.
[  598.890421] usb 7-1: Device not responding to setup address.
[  599.096963] usb 7-1: device not accepting address 9, error -71
[  599.097223] usb usb7-port1: unable to enumerate USB device
[  637.237502] usb 7-1: new high-speed USB device number 10 using xhci-hcd
[  637.559655] usb 7-1: New USB device found, idVendor=18ec, idProduct=5850, bcdDevice= 1.00
[  637.559664] usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  637.559669] usb 7-1: Product: USB2.0 PC CAMERA
[  637.559673] usb 7-1: Manufacturer: ARKMICRO
[  637.614019] uvcvideo: Found UVC 1.00 device USB2.0 PC CAMERA (18ec:5850)
[  637.623883] input: USB2.0 PC CAMERA as /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.11.auto/usb7/7-1/7-1:1.0/input/input11
[  637.645935] usb 7-1: 3:1: cannot get freq at ep 0x82
[  637.681415] usb 7-1: Warning! Unlikely big volume range (=11903), cval->res is probably wrong.
[  637.681424] usb 7-1: [2] FU [Mic Capture Volume] ch = 2, val = 384/12287/1
[  637.838283] usb 7-1: 3:1: cannot get freq at ep 0x82
[  637.902107] usb 7-1: 3:1: cannot get freq at ep 0x82
```

redradist, 2024-01-30:
> This is a camera or a WiFi dongle?

It is the following receiver [Skydroid Mini 5.8G FPV Receiver UVC Video Downlink OTG](https://www.ebay.co.uk/itm/334488609325?hash=item4de113222d:g:kd4AAOSwVWxhC6t7&amdata=enc%3AAQAIAAAA4Jp6Do1Pmg8gogxSM%2BWkOVWsU3r%2FaCsYqb8KG30FHqc2Ee7MOqVY58hsoErKbak2V1%2FtRH3Ip8%2B8KJ9JzXIwaG4dRpx%2FcsfIsti%2BKp%2FVp1tMId3nDLdQ15gNytUMrj%2FsTM2Q65sypnpZ2GjG12DEavv7MNyKmGwBuHxqgPlPfb6r7eA3C%2BlT8MQL1vdaIAeCy5ClgYV1dYrbvHGNqLHSBC9yBbVoQ1Rk1MHEXi7OoI37QhOajPg0x9P4AWeawLPFmDcoPSKAdOJk%2Fm%2Fl8DsPjQH70RzG6ki1gzoPKCQigs4f%7Ctkp%3ABFBMyvLcq6tj)

It worked after fresh installation for long time, then I did few `sudo apt update && sudo apt upgrade` and after that started noticed this problem

Joshua-Riek, 2024-01-30:
There is likely nothing I can do here. Looks like a low level USB quirk / bug,

Joshua-Riek, 2024-01-30:
You can dig through my kernel builds below.

https://launchpad.net/~jjriek/+archive/ubuntu/rockchip

redradist, 2024-01-31:
@Joshua-Riek Looks like I have figured out the issue
[image]

When I inserted any USB cable in vertical USB port along side with USB 5.8GHz Receiver, USB Hub probably due to these broken 2 USB contacts start detecting USB device which do not respond and after reboot did not any more detected receiver with properly functioning ...

closed 2024-01-31 by redradist