Joshua-Riek/ubuntu-rockchip #596: Source code location for setting username and password at first startup

Z-Keven, 2024-01-31:
Hello, I have a question. If I want to install the deb package after setting the username and password on the first system startup, where should I add the dpkg -i command in the source code?

Z-Keven, 2024-02-01:
[libmali.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/14122071/libmali.zip)
I tried installing these two libmali deb packages, which can solve the problem of green screen playing videos, not only for youtube. I have tested and found no other problems, but these two packages can only be installed after setting the user name and password when the system is started for the first time.

DHDAXCW, 2024-02-03:
This will lose hardware acceleration and become llvm. Of course, there is no problem with llvm playing on the browser.

closed 2024-12-16 by github-actions[bot]