Joshua-Riek/ubuntu-rockchip #598: Orange Pi 3B (CLI) image error - Xauthority does not exist

thewonderer73, 2024-02-01:
/usr/bin/xauth:  file /home/ubuntu/.Xauthority does not exist

Just received my new OPi 3B.  Imaged the new SD card and booted.  Prompted to update default password and logged back in.  Error appeared as soon as I logged in.

Power usage is half that of the OPi 5+ (to be expceted).  Drawing 1.7w with no peripherals.  May be lower when I get the EMMC installed...

Joshua-Riek, 2024-02-01:
Where did you see this error?

thewonderer73, 2024-02-01:
I connected via SSL, as soon as I logged in to a session the second time (after updating default password and logging back out

Error appeared straight after I logged in with my creds.

thewonderer73, 2024-02-01:
As I copied the OS to the EMMC using your commands in the wiki for the OPi 5+ and changing the image name to match OPi3B, i managed to capture the error, on first login after default password is updated.

Last login: Thu Feb  1 23:45:20 2024 from 192.168.1.29
/usr/bin/xauth:  file /home/ubuntu/.Xauthority does not exist

Like you said, sounds benign, but wanted to share with you, just in case.

closed 2025-02-17 by github-actions[bot]