Joshua-Riek/ubuntu-rockchip #599: Orange Pi5 Plus onboard PWM-Fan Adjusting

numetaler, 2024-02-01:
Hello Joshua,
Thank you for your great work!

Is it possible to make a overlay for the onboard PWM-Fan settings?
In Armbian i could edit the dtb myself and adjust the temps and values for the speed. On your distro i didn't find out how i could edit the device tree myself.

Im speking of those values:

    fan: pwm-fan {
        compatible = "pwm-fan";
        #cooling-cells = <2>;
        pwms = <&pwm3 0 50000 0>;
        cooling-levels = <0 50 100 150 200 255>;
        rockchip,temp-trips = <
            50000   1
            55000   2
            60000   3
            65000   4
            70000   5
        >;

        status = "okay";
    };

I would like to adjust those to my likings since im not using the original Orange Pi 5 Plus Fan

Aelliari, 2024-02-02:
https://github.com/metamot/opi5plus_fan_fix

Aelliari, 2024-02-02:
> I would like to adjust those to my likings since im not using the original Orange Pi 5 Plus Fan

https://github.com/metamot/opi5plus_fan_fix

There is an instruction on how to work with dtb and at the same time fix pwm operation on the built-in 2pin fan port

numetaler, 2024-02-28:
Hello,

Thank you, this solved my Problem (kind of)

closed 2024-02-28 by numetaler