Joshua-Riek/ubuntu-rockchip #604: NanoPC T6 - LTS Different I/O not working

Xeue, 2024-02-02:
Hi Joshua,
FriendlyElec have brought out a new revision of the NanoPC T6 that has different I/O... They removed the mini PCIe and repalced it with some USBs. They don't seem to work, I assume there's an overlay that needs to be added somwhere?
If this is something you can look at, we can get a board sent to you if needed... We have quite a few spares!

nyanmisaka, 2024-02-02:
https://github.com/friendlyarm/kernel-rockchip/blob/nanopi6-v6.1.y/arch/arm64/boot/dts/rockchip/rk3588-nanopi6-rev07.dts

Xeue, 2024-02-02:
Complete noob here, when you say apply, do you mean by editing the ubuntuEnv.txt and adding it in overlays=

Or do I need to modify something?
Or do I need to compile my own version with something tweaked?

Joshua-Riek, 2024-02-02:
You will need to modify the device tree in the kernel. Since there are some unique hardware changes I think a new device tree is required.

chuichuan, 2024-04-03:
Could you let us know how to? thanks.

Joshua-Riek, 2024-04-03:
I'm building an Armbian image with a patched device tree to fix this. When its done I will give you the link and please let me know if it works. I can merge it after.

Xeue, 2024-04-03:
I will test as soon as I can, have time over the next 3 weeks to mess about with builds

Joshua-Riek, 2024-04-04:
The build has finished and can be found below:

https://github.com/Joshua-Riek/armbian-images/releases/tag/20240404-0112

Xeue, 2024-04-04:
The USBs definitely work now!
I'll do some more testing on all the other things like PCIe, GPIO, PWM etc a bit later to confirm.

chuichuan, 2024-04-06:
USB & PCIE works fine now.
thanks very much.

Joshua-Riek, 2024-04-06:
Cool, i will add the device tree to the kernel.

Xeue, 2024-04-07:
Agreed, all tested all good! Didn't see all the PWM nodes, but I don't care about that anyway... The one it has is all I need.
Any idea when you'll have a bit of time to add this in? I know you're working on a lot of way more important/long term/useful stuff atm!

Joshua-Riek, 2024-04-07:
I added the device tree to the 6.1 kernel, I can add it to 5.10 later tonight and start a new kernel build on Launchpad.

Xeue, 2024-04-07:
No rush from me, so don't let it distract you if you've got more important things to look at!

closed 2024-04-20 by Xeue

Xeue, 2024-06-04:
@Joshua-Riek they don't appear to be selling the non-LTS version anymore, would it be possible to make this overlay active by default in future builds? It may cause less confusion for people...

Joshua-Riek, 2024-06-04:
I thought the device tree was in the kernel. If i had the hardware i could create an image for it.