Joshua-Riek/ubuntu-rockchip #605: Orange Pi 3B: MAC Address Changes on Reboot

MarkhamLee, 2024-02-02:
Bug 

Upon rebooting the Orange Pi 3B the MAC address always changes, as a result the IP address changes for the LAN connection and I'm unable to assign a static IP address via my router. After a recent round of installing updates the MAC address temporarily went back to the original, but on subsequent reboots the address started changing again. 

If it matters, I haven't gotten Wi-Fi working yet. 

Expected Behavior 

The MAC address would remain static 

Impact 
Unable to assign a static IP via LAN 

Details: 

Kernel: 5.10.160 

I've included my last boot log in case that helps 

Feb 02 12:35:34 opi3bnode0 kernel: Booting Linux on physical CPU 0x0000000000 [0x412fd050]
Feb 02 12:35:34 opi3bnode0 kernel: Linux version 5.10.160 (root@fv-az773-303) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4>
Feb 02 12:35:34 opi3bnode0 kernel: Machine model: Rockchip RK3566 OPi 3B
Feb 02 12:35:34 opi3bnode0 kernel: earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
Feb 02 12:35:34 opi3bnode0 kernel: printk: bootconsole [uart8250] enabled
Feb 02 12:35:34 opi3bnode0 kernel: efi: UEFI not found.
Feb 02 12:35:34 opi3bnode0 kernel: OF: fdt: Reserved memory: failed to reserve memory for node 'drm-logo@00000000': base 0x0000000000000>
Feb 02 12:35:34 opi3bnode0 kernel: OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@00000000': base 0x00000000>
Feb 02 12:35:34 opi3bnode0 kernel: Zone ranges:
Feb 02 12:35:34 opi3bnode0 kernel:   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
Feb 02 12:35:34 opi3bnode0 kernel:   DMA32    empty
Feb 02 12:35:34 opi3bnode0 kernel:   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
Feb 02 12:35:34 opi3bnode0 kernel: Movable zone start for each node
Feb 02 12:35:34 opi3bnode0 kernel: Early memory node ranges
Feb 02 12:35:34 opi3bnode0 kernel:   node   0: [mem 0x0000000000200000-0x00000000efffffff]
Feb 02 12:35:34 opi3bnode0 kernel:   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
Feb 02 12:35:34 opi3bnode0 kernel: Initmem setup node 0 [mem 0x0000000000200000-0x00000001ffffffff]
Feb 02 12:35:34 opi3bnode0 kernel: On node 0 totalpages: 2031104
Feb 02 12:35:34 opi3bnode0 kernel:   DMA zone: 15352 pages used for memmap
Feb 02 12:35:34 opi3bnode0 kernel:   DMA zone: 0 pages reserved
Feb 02 12:35:34 opi3bnode0 kernel:   DMA zone: 982528 pages, LIFO batch:63
Feb 02 12:35:34 opi3bnode0 kernel:   Normal zone: 16384 pages used for memmap
Feb 02 12:35:34 opi3bnode0 kernel:   Normal zone: 1048576 pages, LIFO batch:63
Feb 02 12:35:34 opi3bnode0 kernel: cma: Reserved 16 MiB at 0x00000000ef000000
Feb 02 12:35:34 opi3bnode0 kernel: psci: probing for conduit method from DT.
Feb 02 12:35:34 opi3bnode0 kernel: psci: PSCIv1.1 detected in firmware.
Feb 02 12:35:34 opi3bnode0 kernel: psci: Using standard PSCI v0.2 function IDs
Feb 02 12:35:34 opi3bnode0 kernel: psci: MIGRATE_INFO_TYPE not supported.
Feb 02 12:35:34 opi3bnode0 kernel: psci: SMC Calling Convention v1.2
Feb 02 12:35:34 opi3bnode0 kernel: percpu: Embedded 32 pages/cpu s93480 r8192 d29400 u131072
Feb 02 12:35:34 opi3bnode0 kernel: pcpu-alloc: s93480 r8192 d29400 u131072 alloc=32*4096

bderleta, 2024-04-06:
I have the same issue. Serial console provides following information:

```
...
[    6.194620] spi-nor spi4.0: XM25QU128C (16384 Kbytes) read_data x4
[    6.201366] rk_gmac-dwmac fe010000.ethernet: IRQ eth_lpi not found
[    6.201968] rk_gmac-dwmac fe010000.ethernet: no regulator found
[    6.202008] rk_gmac-dwmac fe010000.ethernet: clock input or output? (input).
[    6.202044] rk_gmac-dwmac fe010000.ethernet: TX delay(0x30).
[    6.202074] rk_gmac-dwmac fe010000.ethernet: RX delay(0x10).
[    6.202114] rk_gmac-dwmac fe010000.ethernet: integrated PHY? (no).
[    6.202211] rk_gmac-dwmac fe010000.ethernet: clock input from PHY
[    6.202502] rk_gmac-dwmac fe010000.ethernet: init for RGMII
[    6.202914] rk_gmac-dwmac fe010000.ethernet: User ID: 0x30, Synopsys ID: 0x51
[    6.202957] rk_gmac-dwmac fe010000.ethernet:         DWMAC4/5
[    6.202990] rk_gmac-dwmac fe010000.ethernet: DMA HW capability register supported
[    6.203022] rk_gmac-dwmac fe010000.ethernet: RX Checksum Offload Engine supported
[    6.203052] rk_gmac-dwmac fe010000.ethernet: TX Checksum insertion supported
[    6.203078] rk_gmac-dwmac fe010000.ethernet: Wake-Up On Lan supported
[    6.203205] rk_gmac-dwmac fe010000.ethernet: TSO supported
[    6.203237] rk_gmac-dwmac fe010000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    6.203275] rk_gmac-dwmac fe010000.ethernet: rk_get_eth_addr: rk_vendor_read eth mac address failed (-1)
[    6.203321] rk_gmac-dwmac fe010000.ethernet: rk_get_eth_addr: generate random eth mac address: 36:7e:62:25:ff:33
[    6.203398] rk_gmac-dwmac fe010000.ethernet: rk_get_eth_addr: rk_vendor_write eth mac address failed (-1)
[    6.203441] rk_gmac-dwmac fe010000.ethernet: rk_get_eth_addr: id: 1 rk_vendor_read eth mac address failed (-1)
[    6.203477] rk_gmac-dwmac fe010000.ethernet: rk_get_eth_addr: mac address: 36:7e:62:25:ff:33
[    6.203510] rk_gmac-dwmac fe010000.ethernet: device MAC address 36:7e:62:25:ff:33
[    6.203541] rk_gmac-dwmac fe010000.ethernet: TSO feature enabled
[    6.203570] rk_gmac-dwmac fe010000.ethernet: Using 32 bits DMA width
[    6.304552] rk-pcie 3c0000000.pcie: PCIe Linking... LTSSM is 0x3
...
```

MarkhamLee, 2024-04-23:
I read through [274(https://github.com/Joshua-Riek/ubuntu-rockchip/issues/274) and something going sideways with the boot loader makes sense, as I never had the issue when booting from the SD card, the eMMC or when booting Armbian from the NVME. Question: are you booting from NVME or the SD card? I already had the device booting Armbian from NVME before I switched to this distro, so I just imaged another NVME drive, dropped it in the device booted it up and everything worked fine save Wi-Fi and the MAC address. 

As for hints, I can just share what I'm going to try:

* I'm going to put the Armbian NVME back in and see if the issue exists, to see if it's boot loader HW specific or something related to this OS. 

* IF MAC issue exists with Armbian despite my not having had that issue before than I'll put Ubuntu back in and pursue the bootloader angle, otherwise, I'll try re-installing Ubuntu from scratch. 

I will say that I've had booting issues on this device before, I had it originally booting from eMMC and then it stopped after I had put in a new case, and I had to put the SD card back in, boot it, fix the config but it still wouldn't boot from eMMC, but ran off of the SD card, so I configured it to boot from NVME and that's how I had it running until Josh made an Ubuntu distro for the 3B. 

So, maybe another angle for me will be to image an SD card with Ubuntu and see if MAC address exists there, and if it doesn't, fix the boot loader with from a config where everything works and then re-try it with my NVME drive. 

I may not be able to try any of these things today, but I'll should be able to get to it this week and will share what happens, I just know it might be time consuming so I want to make sure I get some other things done before I dive into fixing this issue.

MarkhamLee, 2024-04-24:
I fixed the issue, it's convoluted so bear with me:

1) I first dropped in an NVME drive with Armbian on it, everything booted up fine
2) I rebooted several times to see if the MAC address changed, it never did 
3) I then tried Armbian config, went to the boot loader screen
[image]
the SATA/USB/NVME option kept returning a "no space on /dev/" error, so I just chose the last option: update on MTD Flash
4) Once that was complete, I booted from an SD card with a fresh copy of Ubuntu 22.04, that was fine, rebooted a few times and the MAC address didn't change 
5) I then flashed an NVME with a fresh copy of Ubuntu and dropped that into the device, things booted up fine, rebooted several times and the MAC address did not change. 

As an added bonus I was able to get Wi-Fi up and running, wasn't able to that prior to this. 

So I think the lesson is: if Wi-Fi doesn't work or you get weird MAC address issues AND you installed one of the first builds, downloading the most recent copy and starting from scratch might be your best bet.

closed 2024-12-01 by MarkhamLee