Joshua-Riek/ubuntu-rockchip #608: Orange Pi 5 Plus v1.32 release randomly restarts

flyzet-prog, 2024-02-03:
Hi, there.I have installed v1.32 Ubuntu 22.04.3. with latest Frigate and Home Assistant Supervised installed on it.  I have random reboots   
and don't know what the root cause of this issue. Would be much appreciated if some one can help to resolve it. Where located the log and witch  one i have too check?

Joshua-Riek, 2024-02-04:
What power supply are you using? Can you get boot logs from U-Boot with serial console?

MarkhamLee, 2024-02-04:
I encountered this when using multi-port chargers that have 20, 40, and higher wattage outputs that never caused issues with my other devices and show up as PD on my USB cable that shows power output. Lower wattage outputs would cause reboots under load (like running Geekbench), higher wattage outputs would work fine for days and then randomly reboot. However, when using the official Orange Pi 5 or a MacBook power supply, I've had no issues. I.e., it not just power output it's how the power is managed

flyzet-prog, 2024-02-04:
> U-Boot

I use Apple USB-C Power Adapter 20W. Sorry for dumb question can you tell me location of lof and command to show up?
Thank you for your reply

flyzet-prog, 2024-02-04:
> What power supply are you using? Can you get boot logs from U-Boot with serial console?

I use Apple USB-C Power Adapter 20W. Sorry for dumb question can you tell me location of lof and command to show up?
Thank you for your reply

flyzet-prog, 2024-02-04:
> I encountered this when using multi-port chargers that have 20, 40, and higher wattage outputs that never caused issues with my other devices and show up as PD on my USB cable that shows power [...]

Hm, i will certainly  try Macbook 30 W charger. Now i use Apple USB-C Power Adapter 20W

Joshua-Riek, 2024-02-04:
The Orange Pi 5 Plus does not support PD chargers... You need to use a STATIC 5v 4a power supply.

Closing issue.

closed 2024-02-04 by Joshua-Riek