Joshua-Riek/ubuntu-rockchip #613: When will Orangep5plusd ubuntu adapt to 6.1 kernel?

Jason123456ma, 2024-02-09:


Joshua-Riek, 2024-02-09:
I have it booting, but panfork will not work with the kernel. This means GPU accel will not work on the GNOME desktop. I can use the Mali blobs but they have other issues...

I could use some help if anyone wants to fix panfork for the 6.1 kernel or backport the upcoming panthor driver. But until this is done I will not upgrade fully to the 6.1 kernel, as many users will complain about the lack of GPU support. For the moment we are stuck on 5.10.

Jason123456ma, 2024-02-09:
> I have it booting, but panfork will not work with the kernel. This means GPU accel will not work on the GNOME desktop. I can use the Mali blobs but they have other issues...
> 
> I could use some help if anyone wants to fix panfork for the 6.1 kernel or backport the upcoming panthor driver. But until this is done I will not upgrade fully to the 6.1 kernel, as many users will complain about the lack of GPU support. For the moment we are stuck on 5.10.
> 
> 我让它启动了，但是panfork不能和内核一起工作。这意味着GPU accel将无法在GNOME桌面上工作。我可以使用马里斑点，但他们有其他问题...
> 
> 如果有人想为6.1内核修复panfork或backport即将到来的panthor驱动程序，我可以使用一些帮助。但在此之前，我不会完全升级到6.1内核，因为许多用户会抱怨缺乏GPU支持。目前，我们被困在5.10。

Thank you for your efforts and look forward to a better version.

simeononsecurity, 2024-02-12:
> I have it booting, but panfork will not work with the kernel. This means GPU accel will not work on the GNOME desktop. I can use the Mali blobs but they have other issues...
>
> I could use some help [...]

What about getting us to at least kernel level 5.19?
We're missing out on many wifi 6 adapters without it.

simeononsecurity, 2024-02-12:
Well that is definitely a downer. The RK3588 is becoming more and more mainstream. It's also significantly more powerful than the chipset in the RPi5. Issues like this are the main thing holding it back IMO.
Thanks for your hard work @Joshua-Riek 
I guess the best thing we can do is petition for mainline linux support?

thewonderer73, 2024-02-14:
This might give us some hope for mainline.

https://gitlab.collabora.com/hardware-enablement/rockchip-3588/notes-for-rockchip-3588/-/blob/main/mainline-status.md

Though this article from a while back was more positive (if only they had a crystal ball)

https://www.cnx-software.com/2021/04/15/rk3566-rk3568-linux-mainline/

There are so many devices that use the RK3588, that it would be in the interest of those SBC makers to ensure it's fully supported in mainline and not a hacked kernel

BTW, it's not clear to me if RK3566 is going mainline too?  Any info on that?

bagong, 2024-02-15:
I am actually curious about Xulong's move to release a HarmonyOS image. Are they going to push it forward, or is it just another - try a bit and then let it rot - project? Rumors on Reddit have it that Huawei has strong relations with Xulong? The intra-chinese download sites seem to be Huawei's?

MakKou96, 2024-02-16:
Good news, Orange has released an Ubuntu image on the 6.1 kernel. Today I hope to test

MakKou96, 2024-02-16:
https://pan.baidu.com/s/1MMyK2cA54zV-swELYAu5yw?pwd=mjbi#list/path=%2Fsharelink1077680202-761095036344852%2FUbuntu&parentPath=%2Fsharelink1077680202-761095036344852

MakKou96, 2024-02-16:
https://disk.yandex.ru/d/lJK8iWL_Ulbt_g - For those who do not have an account in Baidu

JFLim1, 2024-02-16:
> https://disk.yandex.ru/d/lJK8iWL_Ulbt_g - For those who do not have an account in Baidu

Thanks.  Don't mind inquiring, does wifi, GPU and VPU hw acceleration work?  

Had earlier tried @Joshua-Riek kernel-6.1.43 where I couldn't get it to boot on Gnome Wayland or KDE Plasma Wayland a week ago and it boot up Gnome-Xorg but wasn't stable.

Joshua-Riek, 2024-02-16:
The 6.1.43 kernel will only work with opengl-es, so i don't see the point in using it.. 

Panfork on 5.10 is much better for a user experience. I will likely upgrade 5.10 to 5.10.198 to give users something now.

MakKou96, 2024-02-16:
This core is suitable for my purposes. The image from the HDMI camera gives a stable picture, 4k video in Chromium also works flawlessly.

MakKou96, 2024-02-16:
> The 6.1.43 kernel will only work with opengl-es, so i don't see the point in using it..

Where can I download your version of the image?

Joshua-Riek, 2024-02-16:
You can dig through github actions, but I've not uploaded anything because i fear backporting panthor to 6.1 may break currently installed systems with this kernel when i distribute an update.

JFLim1, 2024-02-16:
> The 6.1.43 kernel will only work with opengl-es, so i don't see the point in using it..

Thanks for the info.

Noticed that you are building Ubuntu 24.04 with kernel-6.1.43.  Does the Ubuntu-24.04 kernel-6.1.43 pre-installed Desktop come with Panthor?

Joshua-Riek, 2024-02-16:
I'm currently blocked as the Panthor driver is not yet merged into the mainline Linux kernel. But I would like to release Ubuntu 24.04 with kernel 6.1 and Panthor. This will take some time and not be an easy process.

If people continue to ask about the 6.1 kernel without providing helpful insight I will start to lock threads. I've already spent hours working on the kernel and bug fixing but I do not want to release Linux 6.1 or Ubuntu 24.04 if they are worse than what is currently available.

simeononsecurity, 2024-02-16:
> Good news, Orange has released an Ubuntu image on the 6.1 kernel. Today I hope to test

That is interesting, That kernel is still two years old. At this point they should be targeting 6.6 no?

Joshua-Riek, 2024-02-16:
> > Good news, Orange has released an Ubuntu image on the 6.1 kernel. Today I hope to test
> 
> That is interesting, That kernel is still two years old. At this point they should be targeting 6.6 no?
> 
> Also not everyone needs all features. What limitations are there currently for the server os from getting to 6.1/6.6 ?

This is what I'm talking about.. 

1. Linux 6.1 is the latest SDK from Rockchip that fully supports the hardware and its still technically Android.
2. I still need to work on some WiFi drivers, but from a server perspective it's fine. However if released I will be spammed by users asking about the GPU and this would annoy me to absolutely no end. I would not be nice in dealing with this as i explain the same things over and over again (this is seriously driving me insane, do some of your own research). Panthor will require new firmware and other userspace hacks that will not translate to currently released systems, this could easily break a users system.

Joshua-Riek, 2024-02-19:
I have some good news, I was able to get Panfork working on the 6.1 kernel, so the GPU is working. So now I need to work on WiFi drivers and stability. 

[image]

MakKou96, 2024-02-19:
This is great news. I'm looking forward to your fork.

Joshua-Riek, 2024-02-20:
> > I have some good news, I was able to get Panfork working on the 6.1 kernel, so the GPU is working. So now I need to work on WiFi drivers and stability.
> 
> Yeah, looking forward to wifi being available on kernel 6.1. Just checking, does mpv vpu hw acceleration work with bsp-kernel-6.1? Tested linux-rkbsp-6.1.43 on Arch linux earlier on mpv vpu hw acceleration is not available.

I'm not sure yet, now that panfork is working I've started to work on ethernet / wifi drivers. I will take a look tomorrow when i have time.

JFLim1, 2024-02-20:
> Oh, it is my first time to see glmark2 score on ubuntu. It is not same as mine at all.(nearly 3 times) I'm wondering if the wayland the reason.

Most likely is "Wayland" difference.

On KDE-Plasma-Wayland Session with kernel aarch64 Linux 5.10.160-31-rockchip-joshua

```
[jfl@opi5plus ~]$ glmark2-wayland
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      Panfrost
    GL_RENDERER:    Mali-G610 (Panfrost)
    GL_VERSION:     3.0 Mesa 23.0.0-devel (git-120202c675)
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   800x600 windowed
=======================================================
[build] use-vbo=false: FPS: 2766 FrameTime: 0.362 ms
[build] use-vbo=true: FPS: 3352 FrameTime: 0.298 ms
[texture] texture-filter=nearest: FPS: 3572 FrameTime: 0.280 ms
[texture] texture-filter=linear: FPS: 3584 FrameTime: 0.279 ms
[texture] texture-filter=mipmap: FPS: 3554 FrameTime: 0.281 ms
[shading] shading=gouraud: FPS: 2882 FrameTime: 0.347 ms
[shading] shading=blinn-phong-inf: FPS: 2835 FrameTime: 0.353 ms
[shading] shading=phong: FPS: 2563 FrameTime: 0.390 ms
[shading] shading=cel: FPS: 2537 FrameTime: 0.394 ms
[bump] bump-render=high-poly: FPS: 1687 FrameTime: 0.593 ms
[bump] bump-render=normals: FPS: 3642 FrameTime: 0.275 ms
[bump] bump-render=height: FPS: 3539 FrameTime: 0.283 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 2514 FrameTime: 0.398 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 1399 FrameTime: 0.715 ms
[pulsar] light=false:quads=5:texture=false: FPS: 3441 FrameTime: 0.291 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 626 FrameTime: 1.599 ms
[desktop] effect=shadow:windows=4: FPS: 1902 FrameTime: 0.526 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 578 FrameTime: 1.732 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 559 FrameTime: 1.791 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 650 FrameTime: 1.539 ms
[ideas] speed=duration: FPS: 1489 FrameTime: 0.672 ms
[jellyfish] <default>: FPS: 2201 FrameTime: 0.454 ms
[terrain] <default>: FPS: 112 FrameTime: 8.954 ms
[shadow] <default>: FPS: 1774 FrameTime: 0.564 ms
[refract] <default>: FPS: 233 FrameTime: 4.301 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 3383 FrameTime: 0.296 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 3038 FrameTime: 0.329 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 3362 FrameTime: 0.297 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 3243 FrameTime: 0.308 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 2968 FrameTime: 0.337 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 3231 FrameTime: 0.310 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 3217 FrameTime: 0.311 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 2963 FrameTime: 0.338 ms
=======================================================
                                  glmark2 Score: 2404 
=======================================================
```

JFLim1, 2024-02-20:
> There could be more issues not yet found, but the latest experimental 6.1 builds can be found [here](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/workflows/build.yml).

Hi @Joshua-Riek,

To test the new bsp-kernel 6.1, can I just download the kernel-6.1 alone and installed it on an existing Ubuntu-22.04.3 v1.33?  Will it work.?

How to revert back to kernel-5.10.160 after installing the new kernel-6.1?

Joshua-Riek, 2024-02-20:
> Hi @Joshua-Riek,
>
> To test the new bsp-kernel 6.1, can I just download the kernel-6.1 alone and installed it on an existing Ubuntu-22.04.3 v1.33? Will it work.?
>
> How to revert back to kernel-5.10.160 [...]

I would not do this unless you know exactly what you are doing and do not mind breaking your system. But you can download the kernel-6.1 alone and install it as a deb package. To revert you then need to uninstall the deb package with apt. But again, this can break your system or other functionality since 6.1 is still a work in progress.

nyanmisaka, 2024-02-20:
> * [ ]  YouTube 4k 60fps video playback lag (probobly using the CPU)
> * [ ]  Test 4k 60fps video playback with MPV / Gstreamer

The mpp in amazingfate's ppa may be a bit old for 6.1. See the reason [here](https://github.com/JeffyCN/mirrors/issues/18).

However, there are some bugs in the latest mpp, so Jellyfin's ffmpeg uses the [patched](https://github.com/nyanmisaka/mpp/commits/jellyfin-mpp/) mpp.

wyf9661, 2024-02-21:
> 
> Most likely is "Wayland" difference.
> 
unfortunately not. Could you please help me test glmark on your opi5plus in x11 and wayland with the same kernel? Thanks a lot.
[image]

wyf9661, 2024-02-21:
> For anyone intrested, I have listed a few bugs that I have found with the 6.1 kernel and tasks that need to be done. There could be more issues not yet found, but the latest experimental 6.1 builds can be found [here](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/workflows/build.yml).
> 
> ### Tasks and bugs for Rockchip 6.1 BSP
> * [ ]  AP6275P Bluetooth not working when WiFi firmware loads (Orange Pi 5B)
> * [ ]  Port RTL8821CS WiFi / Bluetooth driver (Indiedroid Nova)
> * [ ]  Port UWE5622 WiiFi / Bluetooth driver (Orange Pi 3B)
> * [ ]  YouTube 4k 60fps video playback lag (probobly using the CPU)
> * [ ]  Test 4k 60fps video playback with MPV / Gstreamer

@Joshua-Riek another bugs I find based on the latest commit of 6.1.
On wayland (KDE), audio module can not work but x11 does.

InfernalWolf666, 2024-02-21:
Joshua Joshua Joshua!!!! I'm rooting for Joshua!!!

MPC-GH, 2024-02-21:
> > > There could be more issues not yet found, but the latest experimental 6.1 builds can be found [here](https://github.com/Joshua-Riek/ubuntu-rockchip/actions/workflows/build.yml).

Just to say I flashed the server build img on a Turing RK1 16GB, on a Turing Pi 2.4 host board with the RK1 in Node 2, using BMC firmware 2.0.5 via the web UI and it booted up for me without any trouble.

NVME, Ethernet and the serial interface from the BMC work, and that's all I need for use as a qemu hosting platform.

**_Thanks Joshua for your work on making this so seamless!_**

U-Boot 2024.01 (Feb 20 2024 - 19:59:09 +0000)

Model: Turing Machines RK1
DRAM:  16 GiB (effective 15.7 GiB)

[snip]

Starting kernel ...

[   10.872283] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[   10.872320] Linux version 6.1.43-rockchip (root@fv-az891-806) (aarch64-linux-gnu-gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1 SMP Tue Feb 20 19:50:31 UTC 2024
[   10.887848] Machine model: Turing Machines RK1

[snip]

[   21.372186] rk-pcie fe180000.pcie: failed to initialize host

Ubuntu 22.04.4 LTS ubuntu ttyS9

ubuntu login:

Joshua-Riek, 2024-02-21:
I pushed the rest of the WiFi / Bluetooth drivers needed. I need to do some additional testing to make sure they work as expected (some firmware paths may be broken).

Build: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/7996476376

wyf9661, 2024-02-22:
> I pushed the rest of the WiFi / Bluetooth drivers needed. I need to do some additional testing to make sure they work as expected (some firmware paths may be broken).
> 
> Build: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/7996476376

FEEDBACK:
with the latest build kernel realse, the audio module works well in wayland.
[image]

Joshua-Riek, 2024-02-22:
I think the next step is to get mpp, ffmpeg and chrome working properly on 6.1. There has been some previous discussion on this topic, but my knowledge is minimal in this area. I will start to do some research, but any help would be very much appreciated.

wyf9661, 2024-02-22:
> I think the next step is to get mpp, ffmpeg and chrome working properly on 6.1. There has been some previous discussion on this topic, but my knowledge is minimal in this area. I will start to do some research, but any help would be very much appreciated.

FYI
https://github.com/7Ji-PKGBUILDs/.report/issues?q=is%3Aissue+is%3Aclosed

nyanmisaka, 2024-02-22:
FYI @Joshua-Riek may want to double check rock5b's USB-C PD timeout issue. There were major changes in `drivers/usb/typec/tcpm/tcpm.c` in 6.1 kernel, and I don't have rock5b on hand to test, so I ignored a hack from amazingfate at that time, which is difficult to apply directly.

- https://github.com/Joshua-Riek/linux-rockchip/commit/0f37382ba6c368786d13430725665c815cc4f951
- https://github.com/Joshua-Riek/linux-rockchip/commit/dc285ba334641a52aefe8a2b516bf6197e196d04

Joshua-Riek, 2024-02-22:
> FYI @Joshua-Riek may want to double check rock5b's USB-C PD timeout issue. There were major changes in `drivers/usb/typec/tcpm/tcpm.c` in 6.1 kernel, and I don't have rock5b on hand to test, so I ignored a hack from amazingfate at that time, which is difficult to apply directly.
> 
>     * [Joshua-Riek/linux-rockchip@0f37382](https://github.com/Joshua-Riek/linux-rockchip/commit/0f37382ba6c368786d13430725665c815cc4f951)
> 
>     * [Joshua-Riek/linux-rockchip@dc285ba](https://github.com/Joshua-Riek/linux-rockchip/commit/dc285ba334641a52aefe8a2b516bf6197e196d04)

I will take a look, my initial testing and development has been on the Rock 5B, but I did not use a USB-C PD power supply. Thank you for reminding me about this.

wyf9661, 2024-02-23:
@Joshua-Riek also another issue about reg and ARGB from rkbsp.
https://github.com/7Ji-PKGBUILDs/.report/issues/17#issuecomment-1913746159
https://github.com/7Ji-PKGBUILDs/.report/issues/17#issuecomment-1914356917
Not sure whether it is already merged the kernel repo.

Joshua-Riek, 2024-02-23:
Thanks for the tips on mpp and ffmpeg @nyanmisaka and @wyf9661, you both are a big help! I was able to get both working last night, now I need to work on updating package builds. 

When I have a stable PPA with package builds, I will see if Amazingfate can update their rockchip-multimedia PPA. Or I'll create a fork and produce packages for the upcoming Ubuntu 24.04 release. 

I have been thinking about releasing the 6.1 kernel only for Ubuntu 24.04 so the current Ubuntu 22.04 builds remain in a stable state (I will still maintain 5.10 and likely bump it to 5.10.198). This will mean that Ubuntu 24.04 will be more "bleeding edge". Does anyone have thoughts against this?

nyanmisaka, 2024-02-23:
> Thanks for the tips on mpp and ffmpeg @nyanmisaka and @wyf9661, you both are a big help! I was able to get both working last night, now I need to work on updating package builds.
>
> When I have a [...]

No objections about this. 24.04 should be bundled with 6.1 and marked as a Beta release until enough people test it and prove it is stable. Forcing all 22.04 users to upgrade from 5.10 to the 6.1 kernel and update existing packages prematurely will cause a predictable mess.

InfernalWolf666, 2024-02-23:
Sorry for the stupid question, but will the VLC player work well now?

I have an orange pi 5 plus: video in the browser works just fine!

Games work well too, through box 86 and 64, steam works

But my 4k films did not work in VLC; the built-in video player froze very badly.

Joshua-Riek, 2024-02-23:
> Sorry for the stupid question, but will the VLC player work well now?
> 
> I have an orange pi 5 plus: video in the browser works just fine!
> 
> Games work well too, through box 86 and 64, steam works
> 
> But my 4k films did not work in VLC; the built-in video player froze very badly.

I don't know, I've been using MPV just fine for 4k video playback. But this is off-topic.

InfernalWolf666, 2024-02-23:
> > Извините за глупый вопрос, а будет ли теперь нормально работать VLC плеер?
> > У меня Orange Pi 5 Plus: видео в браузере работает отлично!
> > Игры тоже работают хорошо, через бокс 86 и 64, стим работает
> > Но мои фильмы 4К не работали в VLC; встроенный видеоплеер очень сильно зависал.
> 
> Не знаю, я прекрасно использую MPV для воспроизведения видео 4K. Но это не по теме.
Sorry, please, you have already corrected it, I was using the old version. Thank you, please forgive me again

wyf9661, 2024-02-23:
> Sorry for the stupid question, but will the VLC player work well now?
> 
> I have an orange pi 5 plus: video in the browser works just fine!
> 
> Games work well too, through box 86 and 64, steam works
> 
> But my 4k films did not work in VLC; the built-in video player froze very badly.

you need to build a specific ffmpeg4.4 for vlc like this:
https://github.com/7Ji-PKGBUILDs/ffmpeg4.4-mpp

Joshua-Riek, 2024-02-23:
Does anyone have chrome working with the VPU on the 6.1 kernel? Compiling the patched mpp and ffmpeg against v4l-utils and libv4l-rkmpp are not working. I wonder if i need to wait for chrome to finish compiling. 

I have a work-in-progress PPA for testing if anyone would like to look: https://launchpad.net/~jjriek/+archive/ubuntu/rockchip-multimedia-wip/+packages?field.name_filter=&field.status_filter=published&field.series_filter=jammy

nyanmisaka, 2024-02-23:
I don't have the environment at the moment, but I think libv4l-rkmpp also needs some updates since the old V4L2 VDA is deprecated. https://github.com/JeffyCN/libv4l-rkmpp/commits/master/

Joshua-Riek, 2024-02-23:
Good find, let me update to the latest version of libv4l-rkmpp and see if anything else is missing.

nyanmisaka, 2024-02-23:
Chromium was updated to R119 too. I think amazingfate would be interested in it.
https://github.com/JeffyCN/meta-rockchip/commit/7f491ad965b9b499c39f77c75940f2e3222f5ea9

stevepringle, 2024-02-24:
Hi Joshua,Mainstream Ubuntu 22.04 is now running with kernel 6.5, so I think upgrading Rockchip Ubuntu 22.04 to kernel 6.1 would be the correct thing to do.Moving to Ubuntu 24.04 and kernel version 6.1 at the same time is possibly going to lead to a lot of stability issues.   Just my opinion of course ;-)Regards,Steve PringleSent from my iPad

nyanmisaka, 2024-02-24:
> Hi Joshua,Mainstream Ubuntu 22.04 is now running with kernel 6.5, so I think upgrading Rockchip Ubuntu 22.04 to kernel 6.1 would be the correct thing to do.Moving to Ubuntu 24.04 and kernel version [...]

The only reason for upstream to adopt 6.5 is HWE, to support a wider range of new Intel&AMD graphics cards. But there is no such scalability on RK3588, and 6.1 already supports most mainstream wireless network cards, which is also the kernel version of Debian 12 bookworm.

Josh may add an option to the build script to allow users to build 22.04 and 6.1 together, but it definitely shouldn't be the default option, unless he wants to risk breaking the existing user experience (~1.3k stars).
[image]

Joshua-Riek, 2024-02-24:
I'm doing development with the 6.1 kernel on Ubuntu 22.04 as a proof of concept to ensure everything is working properly. I will eventually remove the option to build the 6.1 kernel with Ubuntu 22.04, but I need time to package and port things to Ubuntu 24.04.

MPC-GH, 2024-02-24:
Cherry picking https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net.git/commit/?id=a7602e7332b9 might be quite helpful as an ethernet fix for some bridging scenarios when using qemu+kvm.

[ As an aside as I know you're not looking at the moment - the build.sh script with mainline 6.7.0 built and installed well enough on an RK1 for BMC-Serial & Ethernet & NVME.  The fan speed control wasn't operational or tuned to the same degree as the 5.10 or 6.1 kernels though and so fan noise was much higher with 6.7.0 than 6.1.43. ]

I don't have a strong view on how you manage the kernel versions vs ubuntu versions.  From the sideline, 'not more than one per LTS version' so you gain as much headroom as possible / deduplicate bug report handling where possible.

Joshua-Riek, 2024-02-24:
FYI I hope to have a release available by the time Ubuntu 24.04 is officially released by Canonical in a little over a month. I may create an Alpha / Beta release before then to get some feedback.

Joshua-Riek, 2024-02-24:
> Does anyone have chrome working with the VPU on the 6.1 kernel? Compiling the patched mpp and ffmpeg against v4l-utils and libv4l-rkmpp are not working. I wonder if i need to wait for chrome to [...]

I did not have the correct permissions set for the dummy V4L2 devices (I removed the default user from being added to the video group, and /dev/video-* is owned by that group), time to do testing all over again! :sweat_smile:

Joshua-Riek, 2024-02-24:
I think Chrome needs a new patch for V4L2 to work properly, I noticed Jeffy patched his patches for Yocto.

https://github.com/JeffyCN/meta-rockchip/commit/2dd446275c317c0045a5e169f255cfc99b737543

Joshua-Riek, 2024-02-27:
I finally have the 6.1 kernel working with ffmpeg, mpp, and chrome perfectly. I found some regression issues in the build system. It's likely that i will revert the main branch back to [v1.33](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v1.33) so i can figure out the exact problem.

For anyone interested, here are the changes made to Amazingfate's PPA, after some more testing I will see if they can update it for anyone who would like to use the 6.1 kernel on a jammy system.

> https://launchpad.net/~jjriek/+archive/ubuntu/rockchip-multimedia?field.series_filter=jammy

wyf9661, 2024-02-27:
Just curious, I saw a package which kernel is 6.2.21, which branch this pkg come from?

JFLim1, 2024-02-27:
> I finally have the 6.1 kernel working with ffmpeg, mpp, and chrome perfectly.

Is this the Build Ubuntu 22.04 with Rockchip Linux 6.1 #246 having ffmpeg, mpp and chrome working?  Thank you.

thewonderer73, 2024-02-28:
I'm curious of this newer 6.1 BSP kernel will be less hacked than Rockchip 5.10 kernel they released.  Having read comments mentioned numerous times about legacy 5.10 released by Rockchip, I'm hopeful that 6.1 will be better *  ( * Reliable?  Quicker? ).

I'm assuming the 5.10 isn't as efficient ( possibly the RK3566/ RK3588 will have better CPU, RAM and storage throughput).

Joshua-Riek, 2024-02-28:
Please give me time. i'm trying to fix regression issues. But my intent is not to release the kernel with Ubuntu 22.04. I may make it available but it will be at your own risk..

nyanmisaka, 2024-02-28:
> I'm curious of this newer 6.1 BSP kernel will be less hacked than Rockchip 5.10 kernel they released. Having read comments mentioned numerous times about legacy 5.10 released by Rockchip, I'm [...]

FYI linux 6.0 is actually linux 5.20. They decided to bump the major version number. So IMO it won’t make much difference vs 5.10 in perf, except for a wider range of device driver support.

Joshua-Riek, 2024-02-28:
> > I finally have the 6.1 kernel working with ffmpeg, mpp, and chrome perfectly.
> 
> Is this the Build Ubuntu 22.04 with Rockchip Linux 6.1 #246 having ffmpeg, mpp and chrome working?  Thank you.

I have regression in the build system and will likely need to revert the last 30 commits as I don't have time to pinpoint the exact issue. Right now, the only fully working 6.1 builds I have are locally built on my computer.

[85 more replies]

closed 2024-03-27 by Joshua-Riek