Joshua-Riek/ubuntu-rockchip #616: Adding DRM protected media playback support

infinite-dark, 2024-02-11:
Hi there.
Before I get to the point I would like to ask for your leniency as I suggest a new feature.
I'm just a simple user with a GitHub account who tinkers a little with SBCs, and not an expert.

I would like to say - absolutely brilliant work on the OS images. Before discovering this repository, I had only used what Radxa or Orange Pi have provided since I needed the support for hardware accessories. And I must say - so far, your images are far superior to theirs :)

I've been wondering - would it be possible to add DRM-protected media playback support to your images?
In my opinion, it would really be the icing on the cake when it comes to providing the full desktop experience on an SBC.

I know it's a hard topic to handle and I already tried researching it, but the walls are too high for me.
There's DRM playback on Raspberry Pi in Firefox and on some RK3588 boards in openFyde, so my hopes are high.
Unless there's some kind of a licensing issue of course, because, frankly speaking, I know nothing about the law.

Thank you for considering my suggestion.
Best regards.

Joshua-Riek, 2024-02-11:
DRM will work on the RK3588 boards, but only when using chromium. A major hack does this, there are so many hacks used currently that it's a nightmare to maintain and limits the system. For example, I can not release images for Ubuntu 24.04 or upgrade to the latest Rockchip 6.1 kernel as both will break the GPU...

If I'm being honest - I would have low expectations for RK3588 as no improvements will come until the GPU driver is finished, and even then we will need to backport a bunch of code so it will work.

closed 2024-02-15 by Joshua-Riek

infinite-dark, 2024-03-07:
That's a bummer. I'm sorry to hear that. But I completely understand that it is a lot of work.
And I also understand that updating the kernel is way more important than some entertainment functionalities...

But DRM playback is kinda important to me. I don't want to be stuck on Fyde for that because I want to keep some of the hardware and software features (like all the accessory and communication buses support).

Now I'm not specifically asking or insisting on implementing DRM playback support, but...
...if you could maybe give me some sources of information, maybe I could do that on my own?