Joshua-Riek/ubuntu-rockchip #617: RK1/TuringPi: Cannot create Vulkan instance

0x7CFE, 2024-02-13:
Hello, I've just received my RK1 modules. After installing `ubuntu-22.04.3-preinstalled-server-arm64-turing-rk1_v1.32.img` and updating my system I tried to do some checks with `vulkaninfo`.

Unfortunately, what I see is:

```
Cannot create Vulkan instance.  
This problem is often caused by a faulty installation of the Vulkan driver or attempting to use a GPU that does not support Vulkan.  
ERROR at ./vulkaninfo/vulkaninfo.h:649:vkCreateInstance failed with ERROR_INCOMPATIBLE_DRIVER
```

This is sad, especially since it was advertized that RK1 and its Mali SoC support Vulkan 1.2.

I tried to find drivers myself, but according to https://community.arm.com/support-forums/f/graphics-gaming-and-vr-forum/54732/how-to-install-mali-g610-driver-in-linux it seems that it should be provided by the OEM.

Should I do something extra to support Vulkan Compute or I am totally out of luck?

P.S.: I tried installing `mali-g610-firmware` package. However, it makes no difference and `vulkaninfo` still fails with the same message.

Joshua-Riek, 2024-02-13:
The GPU situation is very bad for the RK3588. Vulkan will not work on Linux, only Android.

Joshua-Riek, 2024-02-13:
Some blobs may come in the future, though these blobs are closed-source and filled with bugs such as the X11 and Wayland G610 blobs.

See: https://github.com/JeffyCN/mirrors/issues/21

InfernalWolf666, 2024-03-05:
Hello! Any new news about graphics? I want to play games on orange pi 5 plus. And why does it turn out that Mali 610 graphics work on Android, but not on Linux? I don't care if it's closed source or open source.

bagong, 2024-03-05:
> Hello! Any new news about graphics? I want to play games on orange pi 5 plus. And why does it turn out that Mali 610 graphics work on Android, but not on Linux? I don't care if it's closed source or open source.

You might not care, for others it's very important. Who stops you from using Android?

InfernalWolf666, 2024-03-05:
> Вам может быть все равно, для других это очень важно. Кто вам мешает использовать Android?

i see news this site - https://www.cnx-software.com/2024/03/05/panthor-open-source-driver-arm-mali-g310-mali-g510-mali-g610-and-mali-g710-gpu-linux-6-10/

khanumballz, 2024-05-28:
> > Hello! Any new news about graphics? I want to play games on orange pi 5 plus. And why does it turn out that Mali 610 graphics work on Android, but not on Linux? I don't care if it's closed source or open source.
> 
> You might not care, for others it's very important. Who stops you from using Android?

You can't use Blender on android

0x7CFE, 2024-06-15:
Interesting, would it be possible eventually to use upcoming patch for Rockchip NPU for Vulkan compute?

https://lore.kernel.org/lkml/20240612-6-10-rocket-v1-0-060e48eea250@tomeuvizoso.net/

MarkhamLee, 2024-06-15:
> Interesting, would it be possible eventually to use upcoming patch for Rockchip NPU for Vulkan compute?
> 
> 
> 
> https://www.phoronix.com/news/Rocket-Rockchip-NPU-Driver#google_vignette
> 
> https://lore.kernel.org/lkml/20240612-6-10-rocket-v1-0-060e48eea250@tomeuvizoso.net/
> 
> 

From talking to the author it's for running ML models in TensorFlow Lite on the NPU, he's very much focused on building open source NPU drivers. I.e., Vulkan and the GPU aren't in scope and would have to be part of a different effort.

0x7CFE, 2024-06-15:
Oh, that's unfortunate. Though I can fully understand his decision to focus on a specific subsystem. Better have one thing polished than a bunch of half-baked stuff.

ethanc8, 2024-07-21:
@MarkhamLee @0x7CFE Unlike with Etnaviv, the GPU and NPU are completely unrelated -- the GPU is an ARM Mali G610, while the NPU is a heavily modified NVIDIA NVDLA. Therefore, Tomeu Vizoso's efforts are completely unrelated to any GPU support, which is currently being handled by Panfork and will be handled with Panthor (kernel driver) plus Panfrost (Mesa driver) in the future. That driver is only for OpenGL, but I think we will automatically get any improvements done to the [PanVk](https://gitlab.freedesktop.org/mesa/mesa/-/tree/main/src/panfrost/vulkan) Vulkan driver once we move off of Panfork.

ArtemS2, 2025-05-04:
https://www.reddit.com/r/SBCs/comments/1jz3dmh/rockchip_rk3588_with_panvk_is_officially_vulkan/?rdt=51827

https://gist.github.com/danmons/c00366c223ae404c23b97c51c8a1b0ee

I wonder if it's possible to somehow take panvk for an experimental build?

haoyangw, 2025-06-17:
> I wonder if it's possible to somehow take panvk for an experimental build?

If you're using a Rock 5A/5B/5B+/5C/5D/5T device, you can try one of my Ubuntu images [here](https://github.com/haoyangw/ubuntu-rockchip/actions)(find the latest Nightly build and download the .img.xz for your device and desired Ubuntu version). This is my forked repo of Joshua-Riek's Ubuntu builds, only with the GPU driver switched from panfork+libmali to panthor. PanVk with Vulkan 1.1 is supported by default, you can gain Vulkan 1.2 support if you use a ppa like kisak mesa-fresh. 

If you want to be adventurous, you can also compile Mesa 25.2 from source, and gain Vulkan 1.4 support.