Joshua-Riek/ubuntu-rockchip #618: Missing env.txt for OrangePi-5

ahsan-kaukab, 2024-02-13:
Hello I have to work on CAN interface for OrangePi-5. I have build this ubuntu-22 OS for OrangePi-5 now I want to enable CAN while the OS is running on board
I method I have got it to add `"overlays=can1-m1"` in /boot/env.txt 
but there is no env.txt in boot folder so should I create a new env.txt file and add overlay for can there ?

I have been following OrangePi-5 official manual (The screenshot of reference attached)

[image]

**If I can get any other method to enable CAN interface, that could also work
Actually I may have to enable UART and SPI interface in future so I am looking a generic method at a moment** 

NOTE: This is just a generic question related to this Ubuntu-22 for Rockchip RK-35XX

Joshua-Riek, 2024-02-13:
This is a different distro and does not follow the manual. overlays can be added as documented below:

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays

ahsan-kaukab, 2024-02-15:
Yes thanks , I have added file ubuntuEnv.txt in /boot/firmware/ and its working as expected. was able to enable CAN , SPI interface by adding overlays

closed 2024-02-15 by ahsan-kaukab