Joshua-Riek/ubuntu-rockchip #619: Android by Mr. Joshua Riek ?!

InfernalWolf666, 2024-02-13:
Mr. Joshua Riek, you are great! Your Ubuntu build is by far the best OS for the Orange Pi 5 Plus and more.

Mr. Joshua Riek, could you please make Android 12?

At least any Android 9,10,11,12 - absolutely any, if only it worked correctly!

Because there are too many problems in the android from orange pi.

Problem with HDMI sound.
Interface errors.
Control errors.
Cut out gamepad support from SONY Playstation.

Even if you, with your experience, build it on your knees, it will be 1000 times better than what the manufacturer offers.

Joshua-Riek, 2024-02-15:
Sorry but no, I do not have the time or resources for this, additionally the Android SDK has known backdoors that I do not want to deal with.

closed 2024-02-15 by Joshua-Riek

ferrellsl, 2024-02-15:
The official release of Android for the Orange Pi 5 works great, including video acceleration, so I'm not sure why InfernalWolf666 even asked you to support Android.  I run Android from an SD card and have Ubuntu installed on an NVME card so I can have a dual-boot system.  The only drawback that I encountered with the Android official release is that I couldn't get screen rotation to work even if I tried to force it.  That issue may have been resolved by now because I was tinkering with screen rotation about 6 months ago. Download link for the official Android 12 release is here:  http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-pi-5.html

InfernalWolf666, 2024-02-15:
thanks for the answer .  topic can be closed