Joshua-Riek/ubuntu-rockchip #622: Orange Pi 5 - Build fails due to missing packages for Noble

lucianor, 2024-02-17:
Command: sudo ./build.sh --board=orangepi-5 --project=preinstalled-server --release=noble --kernel=mainline

While building an image with noble, the WiringPi packages are not available and therefore the image packaging stops.

This is due to line 27 of config/boards/orangepi-5.conf not having a switch for Ubuntu versions for installing WiringPi.

After commenting that line, I was able to build it. Keeping it here for reference and for an easy fix for a low severity issue.

Jonathan-Chayna, 2024-03-03:
So now you have Ubuntu noble on Orange Pi 5 with GPU working?

Joshua-Riek, 2024-03-05:
Yeah, I should have work-in-progress builds on GitHub actions soon.

closed 2024-12-03 by lucianor