Joshua-Riek/ubuntu-rockchip #624: build.sh Usage: $0 --board=[orangepi-5] --project=[preinstalled-desktop] --release=[jammy] --kernel=[bsp]

hakotani-o, 2024-02-18:
Hi Joshua-Riek, thank you for your greate work.

I think build.sh --kernel=[rockchip-5.10] , for example.

Thank you.

Joshua-Riek, 2024-02-18:
What is the issue?

hakotani-o, 2024-02-19:
Hi Joshua-Riek.
Sorry for the poor English.
thank you for your reply.

I see Usage and
 run  sudo ./build.sh --board=orangepi-5 --project=preinstalled-desktop --release=jammy --kernel=bsp -l
Error: "bsp" is an unsupported kernel.

about preinstalled-desktop.sh , It's my misunderstanding. Sorry.

Thank you.

ohyeah521, 2024-02-19:
try this command:
`sudo ./build.sh --board=rock-5b --project=preinstalled-desktop --kernel=rockchip-5.10 --release=jammy`

hakotani-o, 2024-02-19:
Hi ohyeah521.
Thank you for speaking with me right away.
I now try it.
Thank you.

closed 2024-12-16 by github-actions[bot]