Joshua-Riek/ubuntu-rockchip #626: Orange Pi 5B - Build fails due to unable to inform the kernel of the change

davidzong1, 2024-02-20:
In the process of building a project within Docker, I encountered some errors. Here is part of the relevant log:

Error: Partition(s) 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128 on /dev/loop3 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
Error: in ../scripts/build-image.sh on line 95
Error: in ./scripts/config-image.sh on line 139
Error: in ./build.sh on line 263

PS：Why does loop device loop3 have so many partitions?

Joshua-Riek, 2024-02-20:
The building in Docker is currently broken. it has problems creating loop devices.

erd-yunmu, 2024-02-26:
Without -P, a loop device with multiple partitions cannot be generated.

Joshua-Riek, 2024-03-09:
Thanks @erd-yunmu!

closed 2024-03-09 by Joshua-Riek