Joshua-Riek/ubuntu-rockchip #628: Compilation failed: ../config/projects/preinstalled-desktop.sh: line 20: grep-dctrl: command not found

Z-Keven, 2024-02-21:
Hello, I failed during compilation. Compilation error message: ../config/projects/preinstalled-desktop.sh: Line 20: grep-dctrl: Command not found.
How to solve this problem

EvilOlaf, 2024-02-21:
confirmed.
To share a bit more log:

```bash
[...]
19 added, 9 removed; done.
Setting up procps (2:3.3.17-6ubuntu2.1) ...
Setting up libcryptsetup12:arm64 (2:2.4.3-1ubuntu1.2) ...
Setting up libpython3-stdlib:arm64 (3.10.6-1~22.04) ...
Setting up python3.10 (3.10.12-1~22.04.3) ...
Setting up python3 (3.10.6-1~22.04) ...
Setting up systemd (249.11-0ubuntu3.12) ...
Setting up netplan.io (0.106.1-7ubuntu0.22.04.2) ...
Setting up python3-gi (3.42.1-0ubuntu1) ...
Setting up systemd-timesyncd (249.11-0ubuntu3.12) ...
Setting up python3-pkg-resources (59.6.0-1.2ubuntu0.22.04.1) ...
Setting up networkd-dispatcher (2.1-2ubuntu0.22.04.2) ...
Setting up systemd-sysv (249.11-0ubuntu3.12) ...
Setting up ubuntu-minimal (1.481.1) ...
Setting up python3-apt (2.4.0ubuntu3) ...
Setting up libnss-systemd:arm64 (249.11-0ubuntu3.12) ...
Setting up libpam-systemd:arm64 (249.11-0ubuntu3.12) ...
Processing triggers for libc-bin (2.35-0ubuntu3.6) ...
Processing triggers for ca-certificates (20230311ubuntu0.22.04.1) ...
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Running hooks in /etc/ca-certificates/update.d...
done.
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following NEW packages will be installed:
  systemd-hwe-hwdb ubuntu-pro-client-l10n
The following packages will be upgraded:
  libudev1 ubuntu-advantage-tools udev
3 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 1843 kB of archives.
After this operation, 1295 kB disk space will be freed.
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 udev arm64 249.11-0ubuntu3.12 [1541 kB]
Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libudev1 arm64 249.11-0ubuntu3.12 [76.0 kB]
Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 ubuntu-advantage-tools arm64 30~22.04 [203 kB]
Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 ubuntu-pro-client-l10n arm64 30~22.04 [20.3 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 systemd-hwe-hwdb all 249.11.5 [3228 B]
Preconfiguring packages ...
Fetched 1843 kB in 0s (3848 kB/s)
(Reading database ... 13133 files and directories currently installed.)
Preparing to unpack .../udev_249.11-0ubuntu3.12_arm64.deb ...
Unpacking udev (249.11-0ubuntu3.12) over (249.11-0ubuntu3) ...
Preparing to unpack .../libudev1_249.11-0ubuntu3.12_arm64.deb ...
Unpacking libudev1:arm64 (249.11-0ubuntu3.12) over (249.11-0ubuntu3) ...
Setting up libudev1:arm64 (249.11-0ubuntu3.12) ...
(Reading database ... 13133 files and directories currently installed.)
Preparing to unpack .../ubuntu-advantage-tools_30~22.04_arm64.deb ...
Unpacking ubuntu-advantage-tools (30~22.04) over (27.7~22.04.1) ...
Selecting previously unselected package ubuntu-pro-client-l10n.
Preparing to unpack .../ubuntu-pro-client-l10n_30~22.04_arm64.deb ...
Unpacking ubuntu-pro-client-l10n (30~22.04) ...
Selecting previously unselected package systemd-hwe-hwdb.
Preparing to unpack .../systemd-hwe-hwdb_249.11.5_all.deb ...
Unpacking systemd-hwe-hwdb (249.11.5) ...
Setting up udev (249.11-0ubuntu3.12) ...
A chroot environment has been detected, udev not started.
Setting up systemd-hwe-hwdb (249.11.5) ...
Setting up ubuntu-advantage-tools (30~22.04) ...
Installing new version of config file /etc/apt/apt.conf.d/20apt-esm-hook.conf ...
Installing new version of config file /etc/ubuntu-advantage/uaclient.conf ...
Installing new version of config file /etc/update-motd.d/91-contract-ua-esm-status ...
Created symlink /etc/systemd/system/multi-user.target.wants/ubuntu-advantage.service → /lib/systemd/system/ubuntu-advantage.service.
Removing obsolete conffile /etc/update-motd.d/88-esm-announce ...
Removing obsolete conffile /etc/ubuntu-advantage/help_data.yaml ...
Setting up ubuntu-pro-client-l10n (30~22.04) ...
Processing triggers for libc-bin (2.35-0ubuntu3.6) ...
Reading package lists...
Building dependency tree...
Reading state information...
Suggested packages:
  debtags
The following NEW packages will be installed:
  dctrl-tools
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 65.2 kB of archives.
After this operation, 266 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 dctrl-tools arm64 2.24-3build2 [65.2 kB]
Fetched 65.2 kB in 0s (242 kB/s)
                                Selecting previously unselected package dctrl-tools.
(Reading database ... 13283 files and directories currently installed.)
Preparing to unpack .../dctrl-tools_2.24-3build2_arm64.deb ...
Unpacking dctrl-tools (2.24-3build2) ...
Setting up dctrl-tools (2.24-3build2) ...
../config/projects/preinstalled-desktop.sh: line 20: grep-dctrl: command not found
../config/projects/preinstalled-desktop.sh: line 20: grep-dctrl: command not found
../config/projects/preinstalled-desktop.sh: line 20: grep-dctrl: command not found
Reading package lists...
E: Unsupported file ./scripts/build-rootfs.sh given on commandline
E: Unsupported file ./scripts/build-rootfs.sh given on commandline
E: Unsupported file ./scripts/build-rootfs.sh given on commandline
Error: in ./scripts/build-rootfs.sh on line 26
Error: in ./build.sh on line 260
```

Z-Keven, 2024-02-21:
> confirmed. I assume just a temporary breakage due to recent changes. To share a bit more log:
> 
> ```shell
> [...]
> 19 added, 9 removed; done.
> Setting up procps (2:3.3.17-6ubuntu2.1) ...
> Setting up libcryptsetup12:arm64 (2:2.4.3-1ubuntu1.2) ...
> Setting up libpython3-stdlib:arm64 (3.10.6-1~22.04) ...
> Setting up python3.10 (3.10.12-1~22.04.3) ...
> Setting up python3 (3.10.6-1~22.04) ...
> Setting up systemd (249.11-0ubuntu3.12) ...
> Setting up netplan.io (0.106.1-7ubuntu0.22.04.2) ...
> Setting up python3-gi (3.42.1-0ubuntu1) ...
> Setting up systemd-timesyncd (249.11-0ubuntu3.12) ...
> Setting up python3-pkg-resources (59.6.0-1.2ubuntu0.22.04.1) ...
> Setting up networkd-dispatcher (2.1-2ubuntu0.22.04.2) ...
> Setting up systemd-sysv (249.11-0ubuntu3.12) ...
> Setting up ubuntu-minimal (1.481.1) ...
> Setting up python3-apt (2.4.0ubuntu3) ...
> Setting up libnss-systemd:arm64 (249.11-0ubuntu3.12) ...
> Setting up libpam-systemd:arm64 (249.11-0ubuntu3.12) ...
> Processing triggers for libc-bin (2.35-0ubuntu3.6) ...
> Processing triggers for ca-certificates (20230311ubuntu0.22.04.1) ...
> Updating certificates in /etc/ssl/certs...
> 0 added, 0 removed; done.
> Running hooks in /etc/ca-certificates/update.d...
> done.
> Reading package lists...
> Building dependency tree...
> Reading state information...
> Calculating upgrade...
> The following NEW packages will be installed:
>   systemd-hwe-hwdb ubuntu-pro-client-l10n
> The following packages will be upgraded:
>   libudev1 ubuntu-advantage-tools udev
> 3 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
> Need to get 1843 kB of archives.
> After this operation, 1295 kB disk space will be freed.
> Get:1 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 udev arm64 249.11-0ubuntu3.12 [1541 kB]
> Get:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 libudev1 arm64 249.11-0ubuntu3.12 [76.0 kB]
> Get:3 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 ubuntu-advantage-tools arm64 30~22.04 [203 kB]
> Get:4 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 ubuntu-pro-client-l10n arm64 30~22.04 [20.3 kB]
> Get:5 http://ports.ubuntu.com/ubuntu-ports jammy-updates/main arm64 systemd-hwe-hwdb all 249.11.5 [3228 B]
> Preconfiguring packages ...
> Fetched 1843 kB in 0s (3848 kB/s)
> (Reading database ... 13133 files and directories currently installed.)
> Preparing to unpack .../udev_249.11-0ubuntu3.12_arm64.deb ...
> Unpacking udev (249.11-0ubuntu3.12) over (249.11-0ubuntu3) ...
> Preparing to unpack .../libudev1_249.11-0ubuntu3.12_arm64.deb ...
> Unpacking libudev1:arm64 (249.11-0ubuntu3.12) over (249.11-0ubuntu3) ...
> Setting up libudev1:arm64 (249.11-0ubuntu3.12) ...
> (Reading database ... 13133 files and directories currently installed.)
> Preparing to unpack .../ubuntu-advantage-tools_30~22.04_arm64.deb ...
> Unpacking ubuntu-advantage-tools (30~22.04) over (27.7~22.04.1) ...
> Selecting previously unselected package ubuntu-pro-client-l10n.
> Preparing to unpack .../ubuntu-pro-client-l10n_30~22.04_arm64.deb ...
> Unpacking ubuntu-pro-client-l10n (30~22.04) ...
> Selecting previously unselected package systemd-hwe-hwdb.
> Preparing to unpack .../systemd-hwe-hwdb_249.11.5_all.deb ...
> Unpacking systemd-hwe-hwdb (249.11.5) ...
> Setting up udev (249.11-0ubuntu3.12) ...
> A chroot environment has been detected, udev not started.
> Setting up systemd-hwe-hwdb (249.11.5) ...
> Setting up ubuntu-advantage-tools (30~22.04) ...
> Installing new version of config file /etc/apt/apt.conf.d/20apt-esm-hook.conf ...
> Installing new version of config file /etc/ubuntu-advantage/uaclient.conf ...
> Installing new version of config file /etc/update-motd.d/91-contract-ua-esm-status ...
> Created symlink /etc/systemd/system/multi-user.target.wants/ubuntu-advantage.service → /lib/systemd/system/ubuntu-advantage.service.
> Removing obsolete conffile /etc/update-motd.d/88-esm-announce ...
> Removing obsolete conffile /etc/ubuntu-advantage/help_data.yaml ...
> Setting up ubuntu-pro-client-l10n (30~22.04) ...
> Processing triggers for libc-bin (2.35-0ubuntu3.6) ...
> Reading package lists...
> Building dependency tree...
> Reading state information...
> Suggested packages:
>   debtags
> The following NEW packages will be installed:
>   dctrl-tools
> 0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
> Need to get 65.2 kB of archives.
> After this operation, 266 kB of additional disk space will be used.
> Get:1 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 dctrl-tools arm64 2.24-3build2 [65.2 kB]
> Fetched 65.2 kB in 0s (242 kB/s)
>                                 Selecting previously unselected package dctrl-tools.
> (Reading database ... 13283 files and directories currently installed.)
> Preparing to unpack .../dctrl-tools_2.24-3build2_arm64.deb ...
> Unpacking dctrl-tools (2.24-3build2) ...
> Setting up dctrl-tools (2.24-3build2) ...
> ../config/projects/preinstalled-desktop.sh: line 20: grep-dctrl: command not found
> ../config/projects/preinstalled-desktop.sh: line 20: grep-dctrl: command not found
> ../config/projects/preinstalled-desktop.sh: line 20: grep-dctrl: command not found
> Reading package lists...
> E: Unsupported file ./scripts/build-rootfs.sh given on commandline
> E: Unsupported file ./scripts/build-rootfs.sh given on commandline
> E: Unsupported file ./scripts/build-rootfs.sh given on commandline
> Error: in ./scripts/build-rootfs.sh on line 26
> Error: in ./build.sh on line 260
> ```

The phenomenon I encountered is consistent.

closed 2024-02-22 by Joshua-Riek

Z-Keven, 2024-02-22:
My issue still persists, and now the error has changed to: ../config/projects/preinstalled-desktop.sh: line 20: /usr/bin/grep-dctrl: No such file or directory

ohyeah521, 2024-02-22:
apt install dctrl-tools

Z-Keven, 2024-02-22:
> apt install dctrl-tools
I have searched under the /usr/bin directory and found grep-dctrl. However, the error persists: ../config/projects/preinstalled-desktop.sh: line 20: /usr/bin/grep-dctrl: No such file or directory.

ubuntu-rockchip/build$ sudo chroot rootfs /bin/bash -c "find -name "grep-dctrl""
./usr/bin/grep-dctrl

Z-Keven, 2024-02-22:
> apt install dctrl-tools

I have searched under the /usr/bin directory and found grep-dctrl. However, the error persists: ../config/projects/preinstalled-desktop.sh: line 20: /usr/bin/grep-dctrl: No such file or directory.

```
ubuntu-rockchip/build/rootfs$ sudo find -name "grep-dctrl"
./usr/bin/grep-dctrl
```

Z-Keven, 2024-02-22:
```
The following NEW packages will be installed:
  dctrl-tools
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 65.2 kB of archives.
After this operation, 266 kB of additional disk space will be used.
Get:1 http://ports.ubuntu.com/ubuntu-ports jammy/main arm64 dctrl-tools arm64 2.24-3build2 [65.2 kB]
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = "zh_CN:en",
        LC_ALL = (unset),
        LC_PAPER = "zh_CN.UTF-8",
        LC_NUMERIC = "zh_CN.UTF-8",
        LC_IDENTIFICATION = "zh_CN.UTF-8",
        LC_MEASUREMENT = "zh_CN.UTF-8",
        LC_NAME = "zh_CN.UTF-8",
        LC_TELEPHONE = "zh_CN.UTF-8",
        LC_ADDRESS = "zh_CN.UTF-8",
        LC_MONETARY = "zh_CN.UTF-8",
        LC_TIME = "zh_CN.UTF-8",
        LANG = "C.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("C.UTF-8").
locale: Cannot set LC_ALL to default locale: No such file or directory
Fetched 65.2 kB in 1s (69.0 kB/s)
E: Can not write log (Is /dev/pts mounted?) - posix_openpt (19: No such device)
Selecting previously unselected package dctrl-tools.
(Reading database ... 13283 files and directories currently installed.)
Preparing to unpack .../dctrl-tools_2.24-3build2_arm64.deb ...
Unpacking dctrl-tools (2.24-3build2) ...
Setting up dctrl-tools (2.24-3build2) ...
../config/projects/preinstalled-desktop.sh: line 20: /usr/bin/grep-dctrl: No such file or directory.
../config/projects/preinstalled-desktop.sh: line 20: /usr/bin/grep-dctrl: No such file or directory.
../config/projects/preinstalled-desktop.sh: line 20: /usr/bin/grep-dctrl: No such file or directory.
Reading package lists...
E: Unsupported file ./scripts/build-rootfs.sh given on commandline
E: Unsupported file ./scripts/build-rootfs.sh given on commandline
E: Unsupported file ./scripts/build-rootfs.sh given on commandline
Error: in ./scripts/build-rootfs.sh on line 26
Error: in ./build.sh on line 284

```

Z-Keven, 2024-02-23:
> I ran on ubuntu22.04 system and did not encounter the problem you mentioned. Please make sure your system version and network are correct.

ohyeah521，Thank you for your response. Building the old version had no issues, but now I'm stuck at grep-dctrl while building the new version. I haven't found out what the problem is yet. I'll continue to investigate.

Joshua-Riek, 2024-02-23:
I can look more into this when I'm finished with the new 6.1 kernel, I'm working on stability and trying to get mpp, ffmpeg, and chrome working properly right now.

Z-Keven, 2024-02-26:
> I can look more into this when I'm finished with the new 6.1 kernel, I'm working on stability and trying to get mpp, ffmpeg, and chrome working properly right now.

This is great, thank you for your work.

Z-Keven, 2024-02-27:
I haven't made any modifications. The build succeeded today without encountering the issue again. It should be due to problems with my environment.

PR #629 (2024-02-22): Fix building desktop
For some reason after installing `dctrl-tools` I assume the user environment is not updated and therefore the binary is not found. This has been worked around by adding the full path to `grep-dctrl`. Not ideal, the root cause should still be investigated.

~~Test build is still WIP, however it passed the point where this thing was installed and used already. Will update soon~~
Build successful

Fixes #628