Joshua-Riek/ubuntu-rockchip #633: Orange Pi 5b + Ubuntu Server

schmrdty, 2024-02-22:
Hello, I have been attempting for a couple of days to install and update ubuntu server. First the Orange Pi5 image (from Orange Pi directly)  which kept failing to fully update due to an issue with EXT4? Discovered this repo and tried the latest OPi5b image and it is now telling me that "of_dma_request_slave_channel: dma-names property of node '/serial@febc0000' missing or empty"
I am totally new and unsure how to address this matter. Doing some checking around to see what I can find out about the issue.

Joshua-Riek, 2024-02-22:
The kernel is a major hack, errors are shown all the time but will not be properly fixed. If you have a specific problem please explain it.

schmrdty, 2024-02-22:
This error is causing the device to not load Ubuntu or do anything past hitting that error and sitting there.

schmrdty, 2024-02-22:
> The kernel is a major hack
Are you saying I should use a different one?

Joshua-Riek, 2024-02-22:
You're not providing enough information. Do you have serial console logs??? Simply saying, "It does not work" with benign dmesg info gives me no information.

schmrdty, 2024-02-22:
[image]

Joshua-Riek, 2024-02-22:
Lol, I'm closing this. The login screen is shown, I do not see an issue.

closed 2024-02-22 by Joshua-Riek

schmrdty, 2024-02-22:
I can't get past that when I try using the most recent release for Ubuntu server.

schmrdty, 2024-02-22:
So I guess I'll just try to figure out what's wrong

schmrdty, 2024-02-22:
Nowhere does it say "login" except the error message
When I try to type nothing happens with any keyboard

Joshua-Riek, 2024-02-22:
If the system is frozen, then your power supply is bad. I know it's not something wrong with my OS image, I test this extensively. I'm sorry, but if you do know what a login terminal looks like, I can't help.

I'm just an open source dev that is tired and trying not to burn out. I do not have the energy or time to continue dealing with silly problems that are user error.