Joshua-Riek/ubuntu-rockchip #636: Preseed ubiquity

automationStati0n, 2024-02-25:
Hi Joshua or contributors,

Love the work you're doing, wouldn't be able to do much without the time you put in!

I'm using the preinstalled desktop image due to driver shenanigans. 

I noticed that the image is using ubiquity.
https://github.com/genesi/ubiquity/tree/master

My question is, is there a way to disable the startup wizard? I can't seem to figure out the preseed. Wondering if you can give a quick pointer, I want a default user like `orangepi` and password without actually inputting this.

If not, don't worry about it

Joshua-Riek, 2024-02-25:
You would need to remove these packages `oem-config-gtk ubiquity-frontend-gtk ubiquity-slideshow-ubuntu` in the build process, then create your own image with your user defined.

In the early releases I did not have a install wizard and did exactly as you describe, maybe look at the source code available at the release tag: https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v1.8

automationStati0n, 2024-02-25:
Awesome, thanks for the reply!

I'm looking into it further. I'll admit im being lazy and doing a chroot on your prebuilt image, guessing it should still work as I can `su - orangepi` from oem on first boot .

My worry is that ubiquity is doing important things in the background that I need (mainly concerned about proper graphics you have configured). If you're saying all I have to do is remove those packages then it sounds like I won't miss anything important.

Also, side note - first time ever opening an issue anywhere.. noticed I missed the label.

Thanks again

Joshua-Riek, 2024-02-25:
ubiquity only setup the user account, it does not do any meaningful configuration.

automationStati0n, 2024-03-02:
Just reporting back what I managed to get working so I don't leave anyone who wants this hanging later. I wrote this script (may not 100% correct) but gets the job done.

```bash
#!/bin/bash

# This script assumes you are on x86 making changes to aarch64 fs (qemu)
IMG_PATH="~/Downloads/custom_ubuntu/ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-5.img"
MOUNT_POINT="/mnt/ubuntu-img"

mkdir -p $MOUNT_POINT/{dev,proc,sys,boot}
losetup -P /dev/loop0 $IMG_PATH
mount /dev/loop0p2 $MOUNT_POINT
for dir in dev proc sys; do
    mount --bind /$dir $MOUNT_POINT/$dir
done

apt-get install qemu-user-static -y
cp /usr/bin/qemu-aarch64-static $MOUNT_POINT/usr/bin

chroot /mnt/ubuntu-img /usr/bin/qemu-aarch64-static /bin/bash <<EOF
touch /etc/sudoers.d/orangepi
echo "orangepi ALL=(ALL:ALL) ALL" | tee /etc/sudoers.d/orangepi
useradd -p $(openssl passwd -1 orangepi) orangepi
chmod 0440 /etc/sudoers.d/orangepi
echo orangepi:orangepi | chpasswd

systemctl disable oem-config.service
systemctl disable oem-config.target

# Check for additional services that may need to be disabled
systemctl list-unit-files | grep oem-config

#Remove startup wizard
rm -rf /var/lib/oem-config
apt-get remove -y oem-config-gtk ubiquity-frontend-gtk ubiquity-slideshow-ubuntu

#Remove startup popup for gnome-initial-setup
mkdir -p /etc/skel/.config
printf yes | tee /etc/skel/.config/gnome-initial-setup-done >/dev/null

EOF
./unmount.sh
```

You will also need this unmount script:
```bash
#!/bin/bash

# Function to unmount a mount point
unmount_point() {
    if mountpoint -q "$1"; then
        echo "Unmounting $1"
        sudo umount "$1"
    fi
}

# Unmount all bind mounts within the chroot environment
unmount_point /mnt/ubuntu-img/dev/pts
unmount_point /mnt/ubuntu-img/dev
unmount_point /mnt/ubuntu-img/proc
unmount_point /mnt/ubuntu-img/sys

# Unmount the primary mount point
unmount_point /mnt/ubuntu-img

# Detach all associated loop devices
for loop_device in $(losetup -l | awk '{if(NR>1)print $1}'); do
    # Find all mounts associated with this loop device and unmount them
    for assoc_mount in $(findmnt -nlo TARGET -S "$loop_device"); do
        unmount_point "$assoc_mount"
    done

    # Detach the loop device
    echo "Detaching $loop_device"
    sudo losetup -d "$loop_device"
done

echo "All loop devices have been detached and mount points unmounted."
```

automationStati0n, 2024-03-02:
I noticed now that the GPU does not appear to be rendering. Video playback is slow and all cores are at 100% during playback. There must be something else I'm missing.

Joshua-Riek, 2024-03-03:
I know the gpu works out of the box, but x11 is sometimes buggy. If you run glmark2 what score do you get?

automationStati0n, 2024-03-03:
Score is 1374

Joshua-Riek, 2024-03-03:
Gpu is working then, you are likely using x11

automationStati0n, 2024-03-04:
I confirmed my session is wayland. The issue went away on it's own. Not sure if it was due to running apt upgrade and restarting,

Im going to assume it's that but I'm going to test it on a re-flash/reinstall

automationStati0n, 2024-03-08:
yup, something was up with my install. Re-running it from your image was fine. Closing this

closed 2024-03-08 by automationStati0n