Joshua-Riek/ubuntu-rockchip #639: orangepi-3b Playback Has No Sound

yoqy, 2024-02-26:
Whether it is the system burned with Release v1.33 or a system compiled from the source code, when playing videos, the video plays smoothly but there is no sound.

Even when testing with paplay, there is no sound through both the headphone jack and HDMI.

Hardware Version: OPi 3B V1.1.1

Release v1.33 Image: ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-3b.img.xz

git commit 20114e3ed06b8578c76aeb16cf825bac6650ad83
Date:   Wed Feb 21 19:21:23 2024 -0500
Build command: sudo ./build.sh --board=orangepi-3b --release=jammy --project=preinstalled-desktop --kernel=rockchip-5.10
Build Image: images/ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-3b.img.xz

paplay Playback Command:
paplay --device=alsa_output.platform-rk809-sound.stereo-fallback 10s_48k_s16le_stereo_ys.wav
paplay --device=alsa_output.platform-rk809-sound.stereo-fallback 10s_48k_s16le_stereo_ys.wav

yoqy, 2024-02-26:
The aplay program is used to play audio files, but there is also no sound. Could someone please help check what the problem might be?
aplay -D hw:0,0 10s_48k_s16le_stereo_ys.wav
aplay -D hw:1,0 10s_48k_s16le_stereo_ys.wav

yoqy, 2024-02-29:
I found that the possible reason for the lack of sound during HDMI playback might be this:linux-rockchip/drivers/gpu/drm/bridge/synopsys/dw-hdmi-i2s-audio.cdw_hdmi_i2s_prepare()

defencedog, 2024-09-14:
@yoqy Any fix you have found?
Big Thanks to @dust-7 @amazingfate for `v2.1` Previous month I have worked a lot on this board as we use in it chemical engineering experiments. There is also an audio issue with 3.5mm jack & microphone not detected. Though these hardware works in kernel6.6.4 as [mentioned here](https://github.com/defencedog/orangepi3b_v2.1/tree/main/MicroLinux_Armbian_Orangepi3b_jammy_edge_6.6.4)

Per my experience if the output of `aplay -l` is following audio  from 3.5mm jack **works**
```
**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDMI], device 0: fe400000.i2s-i2s-hifi i2s-hifi-0 [fe400000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: RK809 [Analog RK809], device 0: fe410000.i2s-rk817-hifi rk817-hifi-0 [fe410000.i2s-rk817-hifi rk817-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

If this is the output you get 3.5mm **do not** work
```
card 0: rockchiprk809 [rockchip-rk809], device 0: dailink-multicodecs rk817-hifi-0 [dailink-multicodecs rk817-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: rockchiphdmi [rockchip,hdmi], device 0: fe400000.i2s-i2s-hifi i2s-hifi-0 [fe400000.i2s-i2s-hifi i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

closed 2024-12-16 by github-actions[bot]