Joshua-Riek/ubuntu-rockchip #643: Radxa Rock 5A: touch not working on Radxa Display 8 HD

infinite-dark, 2024-02-29:
Just like in the title. I activated the overlay and the display itself works completely fine.
There seems to be no touch functionality though.

Is there perhaps a configuration step I might be missing?

Joshua-Riek, 2024-02-29:
The touchscreen driver may be missing, do you happen to have a link to the exact display being used? I need to look up the driver, it may be a simple fix.

infinite-dark, 2024-02-29:
> The touchscreen driver may be missing, do you happen to have a link to the exact display being used? I need to look up the driver, it may be a simple fix.

Sure! Here's the product page: https://radxa.com/products/accessories/display-8hd
And the docs: https://docs.radxa.com/en/accessories/lcd-8-hd/lcd-8-hd-product

Joshua-Riek, 2024-02-29:
Hmm, i see the driver TOUCHSCREEN_GT9XX is enabled and the commits match radxa's kernel tree. The only commit missing is https://github.com/radxa/kernel/commit/010291e9a3da403dbb0435de62f06aebaf4930d0, but id imagine it should still work. 

Can you open a terminal and post the output of `dmesg | grep -i goodix` and `dmesg | grep -i gt9xx`?

infinite-dark, 2024-02-29:
> dmesg | grep -i goodix
[   10.104169] Error: Driver 'Goodix-TS' is already registered, aborting...

> dmesg | grep -i gt9xx
Returns nothing.

That's all the output I got.

infinite-dark, 2024-03-13:
Just a small, quick comment. The same thing happens on the 5B.

kalyanikasam, 2024-03-22:
We are facing also same issue. Touch functionality is not working but the display is working as normal display. Is there any way that we can make 8HD Touchscreen work with this image. The touch functionality is working with radxa official images but not this image. Please help

Joshua-Riek, 2024-03-22:
I'm currently fully focused on the 6.1 kernel and ubuntu 24.04, I do not currently have time to address this problem. But id compare GT9XX drivers between radxa's kernel and my own.

Joshua-Riek, 2024-03-22:
Yeah, it's just me. I work a full-time job and do this in my free time, but this project takes up most of my free time as you can imagine lol. Anyway I don't mind, it helps me grow as a developer and find new opportunities.

kalyanikasam, 2024-03-22:
Thank you Joshua for your response, i understand that you are quite busy and highly appreciate your contribution. JR Ubuntu image resolved lots of issues which i was facing with official radxa images. I was researching on my own to understand the issue, as i am new to this environment seeking help.

kalyanikasam, 2024-03-27:
I found a reading from goodix driver maintainers which says some newer ACPI/x86 devices are not listed as GPIO resources and there is only one Gpioint resource and _Ps0 is responsible for power management and the corresponding IRQ pin is set as output mode so there should be a reset sequence from ACPI which can put the pin in input mode. Detailed explanation can be found here https://www.spinics.net/lists/linux-input/msg87797.html, hope this works.

MyVizDrake, 2024-05-13:
Issue exists as well on 5A and Radxa Display 10 FHD.  I have requested assistance from the Radxa Discord as the only way I got the display working is via the Arch DTBO from a 5.10 radxa BSP kernel.  If I used the included DTBO from 24.04 the system boots into maintenance mode. Considering Joshua has other priorities I won't create an issue / bug here until I rule out other things.

handledexception, 2024-06-02:
Do I have to do anything specific to get the Radxa touch displays to even power on with the Ubuntu images? I am using the latest 24.04 image. I've tried it with the older 22.04 images as well to no avail. The Radxa forum post about this says to "add `dtoverlay=rock-5b-radxa-display-10fhd` to config.txt". But Joshua's image doesn't seem to have a /boot/config.txt. I'm kind of lost and a newbie to this image.

Joshua-Riek, 2024-06-02:
I just ported the drivers, the 8HD display works but i need someone to test the 10HD panel out. 

Beta images with display support: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9332933194
How to apply a device tree overlay: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9332933194

handledexception, 2024-06-02:
Thank you, Josh! I will test out the 10HD panel post haste!

Joshua-Riek, 2024-06-02:
If it works I can merge the code into the kernel and push an update.

handledexception, 2024-06-02:
But how do I apply a device tree overlay for "rock-5b-radxa-display-10hd" specifically? I did read that wiki page for instructions on applying a Device Tree Overlay in your images but it seems like the syntax is different, and the example you gave is looking for .dtbo files. Not sure how the Radxa-official instructions map on here.

Joshua-Riek, 2024-06-02:
I dont follow the official radxa instructions since this is a diffrent os. 

1. Open the file `/etc/default/u-boot` 
2. Add the text `U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rock-5b-radxa-display-10hd.dtbo"`
3. Enter the command `sudo u-boot-update`

You can then confirm the overlay by looking  in `/boot/extlinux/extlinux.conf`, for example on my system i have:
```
ubuntu@ubuntu-desktop:~$ cat /boot/extlinux/extlinux.conf
## /boot/extlinux/extlinux.conf
##
## IMPORTANT WARNING
##
## The configuration of this file is generated automatically.
## Do not edit this file manually, use: u-boot-update

default l0
menu title U-Boot menu
prompt 1
timeout 20

label l0
	menu label Ubuntu 22.04.4 LTS 5.10.0-1005-rockchip
	linux /boot/vmlinuz-5.10.0-1005-rockchip
	initrd /boot/initrd.img-5.10.0-1005-rockchip
	fdtdir /lib/firmware/5.10.0-1005-rockchip/device-tree/
	fdtoverlays  /lib/firmware/5.10.0-1005-rockchip/device-tree/rockchip/overlay/rock-5b-radxa-display-8hd.dtbo
	append root=UUID=e417fb23-0018-4867-bb99-26651def195d rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 irqchip.gicv3_pseudo_nmi=0 cgroup_enable=memory quiet splash plymouth.ignore-serial-consoles

label l0r
	menu label Ubuntu 22.04.4 LTS 5.10.0-1005-rockchip (rescue target)
	linux /boot/vmlinuz-5.10.0-1005-rockchip
	initrd /boot/initrd.img-5.10.0-1005-rockchip
	fdtdir /lib/firmware/5.10.0-1005-rockchip/device-tree/
	append root=UUID=e417fb23-0018-4867-bb99-26651def195d rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 irqchip.gicv3_pseudo_nmi=0 cgroup_enable=memory splash plymouth.ignore-serial-consoles single
```

handledexception, 2024-06-02:
Thank you so much for the precise instructions. I am downloading the beta image now and will let everyone know how it goes with my Display 10 HD.

handledexception, 2024-06-02:
So the Display 10 HD does power on now after booting the beta image that Joshua mentioned above. I have power and backlight now, as well as some colored lines across the screen. However I'm not getting a normal image. I need to check my ribbon cables and see if its a physical connection issue.

Also, I no longer get video out from the HDMI like I was before. Not sure if that is expected or not.

Joshua-Riek, 2024-06-02:
Which Ubuntu version? Ubuntu 24.04 and 22.04 use different display drivers for the 10 HD, also does touch screen work?

handledexception, 2024-06-02:
> Which Ubuntu version? Ubuntu 24.04 and 22.04 use different display drivers for the 10 HD, also does touch screen work?

I was using 24.04. I can try 22.04. I don't know if touch worked because all I could see was glitchy colored lines on the screen, no Ubuntu desktop image. It didn't seem like the display responded to touch at all, but its hard to say for sure without a desktop image on screen.

Joshua-Riek, 2024-06-02:
Hmm can you try 24.04?

handledexception, 2024-06-02:
> Hmm can you try 24.04?

Sorry, I meant to say I tried 24.04. I will try 22.04 Desktop next.

Joshua-Riek, 2024-06-02:
Ughhh this is going to be a lot of back and fourth then, are you 100% sure its the 10HD panel? 

From my understanding the 10HD panel was able to display a picture before I made any driver changes. The only problem with it was the touch screen.

handledexception, 2024-06-02:
So I changed the device overlay to `rock-5b-radxa-display-10hd.dtbo` on Ubuntu 24.04 and now I have a good image on the display. I didn't realize that there were separate models called "10 HD" and "10 FHD". My display is the 10 FHD model.

It does not appear that touch is working, though.

Also it seems that my USB keyboard and mouse no longer work either. I tried every USB port but it seems that they are not getting power.

Joshua-Riek, 2024-06-02:
Can you please put the output of `dmesg` into a text file and attach it to github?

Joshua-Riek, 2024-06-02:
Also the touchscreen driver is named `focaltech-ts` would you be able to try and open a terminal and enter `sudo modprobe focaltech-ts`? Maybe manually loading the driver will work.

handledexception, 2024-06-02:
[Uploading dmesg-display-10fhd.txt…]()

Joshua-Riek, 2024-06-02:
I cant download the log-file, it looks broken, can you upload it to pastebin?

handledexception, 2024-06-02:
> I cant download the log-file, it looks broken, can you upload it to pastebin?

Sorry, I had accidentally submitted my comment before it had finished uploading. It is attached to my previous comment properly now.

Joshua-Riek, 2024-06-02:
Hmm i dont see the focaltech driver loading at all, there are no kernel messages about it either. To figure out why this is i will likely need the hardware and do some manual debugging in the driver.

However i do think the Display 10 FHD will work as it uses the same gt9xx touch driver as the Display 8 HD.

handledexception, 2024-06-02:
I really appreciate all of your work, Joshua! I've added a comment to the [newer issue about the Display 10 FHD](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/678) that I just found pointing them toward this thread. Sorry for co-opting this issue for my Display 10 FHD problem.

handledexception, 2024-06-02:
Can you point me to the driver sources for the focaltech-ts module?

Joshua-Riek, 2024-06-02:
I've been cherry-picking from Radxa's kernel source as they have modifications to the driver for their specific hardware. So it's hard to point you to a single source for the driver.. 

Radxa 6.1 kernel: https://github.com/radxa/kernel/tree/linux-6.1-stan-rkr1
Radxa 5.10 kernel: https://github.com/radxa/kernel/tree/linux-5.10-gen-rkr3.4

handledexception, 2024-06-02:
I left a message on the Radxa forum asking if touch on the Display 10 FHD has even been incorporated into their kernels yet. https://forum.radxa.com/t/mipi-display-support/12239/60?u=lilendian

infinite-dark, 2024-06-02:
Hi. I've just noticed a lot of commotion in this thread so I've come here again.
Joshua, thanks for your continued work as always, it is much appreciated.

Firstly, I'd like to ask @handledexception to stop posting comments about his problem in my thread.
I am getting way too much spam about an unrelated configuration (5B + 10FHD).
Please open your own issue.

Secondly, I proceeded to test your images that you posted above @Joshua-Riek with my 5A and 8HD. I tested 24.04 so far and 8HD doesn't work at all. It does get power and the OS sees the "built-in display", but there's no image (black screen).

I will now proceed to test 22.04.

infinite-dark, 2024-06-02:
Alright. Update. Of course it turned out my ribbon cable was damaged...
There is image on both 22.04 and 24.04. Sorry for causing confusion.

However touch still doesn't work...

Joshua-Riek, 2024-06-04:
Both panels will now work as of the following kernel tags:
https://github.com/Joshua-Riek/linux-rockchip/releases/tag/Ubuntu-rockchip-5.10.0-1006.6
https://github.com/Joshua-Riek/linux-rockchip/releases/tag/Ubuntu-rockchip-6.1.0-1015.15

closed 2024-06-04 by Joshua-Riek

kalyanikasam, 2024-06-05:
Hi Joshua, could you please through some light on what the issue was, a brief idea about the issue, why was touch not working. Do we have to install the new release from the above link or can we update the existing one.

MyVizDrake, 2024-06-05:
@kalyanikasam there were several smaller issues IIRC but the big one is that the goodix driver needed to be compiled into the kernel and not loaded as a module, IIRC

kalyanikasam, 2024-06-05:
gt9xx is a tristate module, it depends on how we are going to load it as far as i know. correct me if i am wrong.
@infinite-dark sorry for flooding your thread, but since the issue is the same rock5B + 8in Touchscreen , i did not create new issue

infinite-dark, 2024-06-05:
> Both panels will now work as of the following kernel tags: https://github.com/Joshua-Riek/linux-rockchip/releases/tag/Ubuntu-rockchip-5.10.0-1006.6 https://github.com/Joshua-Riek/linux-rockchip/releases/tag/Ubuntu-rockchip-6.1.0-1015.15

Alright! Thanks! Much appreciated! Will the changes be present in a future update of your images?
Because, honestly, I'm not sure how to compile the kernel for myself.

@kalyanikasam No worries. I was very curious about what the actual issue was myself.
I wish I knew more about Linux to handle these things myself... :/

paulh-aja, 2024-06-09:
I tested the fix with my Radxa Display 8HD and both display and touch input look to be working perfectly. I'll try it out with my 10 FHD soon. Thank you, Josh!

disruptivepatternmaterial, 2025-01-14:
sorry to re-alive this old thread - but where are these overlays? I do not have them anywhere on my setup which is based on the October (latest) build for rock5b. The display works with the radxa debian build, so I know it is working. I followed the instructions above (adding the line to the correct file) but when I try to check it, the overlay is not added to the rebuilt config in /boot/extlinux/extlinux.conf . So I went on the hunt for the file named   /lib/firmware/6.11-rockchip/device-tree/rockchip/overlay/rock-5b-radxa-display-8hd.dtbo. There is no folder called overlay under rockchip at all, in fact.