Joshua-Riek/ubuntu-rockchip #644: Gnome/desktop crashing on both Xorg and Wayland on Orange PI 5

gehee, 2024-02-29:
I installed `[ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-5.img.xz](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.33/ubuntu-22.04.3-preinstalled-desktop-arm64-orangepi-5.img.xz)` and ran `apt upgrade`.

I can't seem to have a stable gnome session now, it crashes and returns to the login screen after a while.

The problem happens right after login when I select the default gnome session.
If I select the gnome session with Xorg, it does not crash when I run terminal but crashes when I run heavier UIs like chromium or code. it does recover the same way and goes back to the login screen after a while.

Could it be memory related?

Joshua-Riek, 2024-02-29:
You likely do not have a proper power supply are you using a 5v 4a power supply? Or you have RAM instability due to SPI + DDR blob mismatch. 

Please run this command to update the SPI and ensure you are using the proper blobs.
```bash
sudo u-boot-install-mtd /dev/mtdblock0
```

gehee, 2024-02-29:
Looks like the power supply was the issue, thank you!
Glad it was that simple of a fix.