Joshua-Riek/ubuntu-rockchip #646: Wiki for orangepi5 is out of date. SATA setting is not work.

SESProMax, 2024-03-02:
The way of flashing image to SATA SSD is not work anymore, please update.
After adding overlay=sata, system still can not find ssd.

Daavidaviid, 2024-03-16:
For me this: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Orange-Pi-5-Plus
Work perfeectly

SESProMax, 2024-03-17:
plus doesn't have sata option. Pi5 has. But it can’t find sata sad, after editing the env file.

For me this: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Orange-Pi-5-Plus worked perfectly

Daavidaviid, 2024-03-17:
Well I don't know what you're trying to achieve but for me it worked with my orange pi 5. I did it today

SESProMax, 2024-03-17:
I have followed wiki, added 'overlays=sata' to /boot/firmware/ubuntuEnv.txt, then reboot. This is no sata ssd in disk list. Did I do something wrong or miss some details.

closed 2024-07-21 by SESProMax