Joshua-Riek/ubuntu-rockchip #647: Request to enable the nbd module / option to use a customised kernel config

MPC-GH, 2024-03-02:
The current kernel builds don't include the nbd module and it is needed when playing around with qemu-nbd.

Could CONFIG_BLK_DEV_NBD be considered to be set in the default builds please?

I guess my deeper question - any pointers on using a customised kernel config when self-building?

I was poking around the build.sh / build-kernel.sh scripts and didn't follow past the call out into the dpkg-buildpackage set of tools for a way to tweak it easily.

Ultimately my user case is modifying the qcow2 sparse overlay img file when creating new VMs as part of a tweaked version of https://github.com/earlruby/create-vm/tree/main

<code>qemu-nbd -c /dev/nbd0 "${VM_IMAGE_DIR}/images/${HOSTNAME}.img"
sleep 1
mount -o rw /dev/nbd0p1 /tmp/diet

[Fiddle about with dietpi.txt and the pre and post autoconfig files]

umount /tmp/diet
qemu-nbd -d /dev/nbd0</code>

Xeue, 2024-03-02:
When I made kernel modifications to this build I downloaded it as in the wiki page: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Kernel
I ran the first command: make CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 rockchip_linux_defconfig
Then I ran: make menuconfig
Used the menu to modify the modules as needed
Then ran the final command: make KBUILD_IMAGE="arch/arm64/boot/Image" CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 -j "$(nproc)" bindeb-pkg

This outputs a pair of deb packages you can then install as the new kernel and headers. I'd recomend also uninstalling the old kernel with apt after doing this

MPC-GH, 2024-03-02:
Thanks Xeue. I did just about this in the end. Built it once, fiddled the kernel settings, then adjusted build-kernel to not resync.

Your way sounds much more sensible!

closed 2024-12-16 by github-actions[bot]