Joshua-Riek/ubuntu-rockchip #649: hdmirx crashing after obs/4L2 tester. Cheese working fine

stallion25, 2024-03-04:
How to reproduce

1. Start cheese hdmirx is working as webcam. Close open , no problem.
2. Start obs , try to configurehdmirx or start v4L2 tester , black screen.
3. Close obs/tester
4. open cheese , not working.
5. reboot then cheese working again

Joshua-Riek, 2024-03-04:
Cheese should not work properly, there are some source code modifications required. I saw some patches from Rockchip for Cheese, if I have time I can take a look. But currently, this is a very low priority.

stallion25, 2024-03-04:
1.33 out of the box cheese is working only first time. After you close it, it will start with "There was an error playing video from the webcam"

stallion25, 2024-03-06:
Hello,

Found additional informations about patching kernel for reset line in hdmirx

https://patchwork.kernel.org/project/linux-media/patch/20240305123648.8847-3-shreeya.patel@collabora.com/

Hdmirx is in most non S RK3588 SBC's, it would be fine working it out of the box, right?

closed 2025-02-17 by github-actions[bot]