Joshua-Riek/ubuntu-rockchip #650: Intermittent wifi connectivity issues with Orange PI 5 Plus

dsburns, 2024-03-04:
I'm observing an intermittent bug with the wifi module shown here - https://a.co/d/3ICOnOR.

It appears as though the device has two interfaces (wlan0 and wlan1, each with a unique mac address) and systemd will randomly select one of the wlan* device at boot to enumerate as _wlP2p33s0_.  Both devices are able to receive a DHCP address, but wlan1 cannot be ping'd or receive any inbound connections.  The wlan0 interface works as expected.

In the lshw output below -  the wlan0 interface was enumerated as _wlP2p33s0_ at boot.
<details><summary>lshw -c network</summary>
<code>*-network
       description: Wireless interface
       product: Realtek Semiconductor Co., Ltd.
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0002:21:00.0
       logical name: wlP2p33s0
       version: 00
       serial: 40:9c:a7:0d:86:79
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=rtl8852be driverversion=v1.15.10.0.5-0-gfa2af07cf.20220 firmware=N/A ip=192.168.1.99 latency=0 link=yes multicast=yes wireles                                                                                    s=IEEE 802.11AX
       resources: irq:281 ioport:300000(size=256) memory:f2200000-f22fffff
  *-network
       description: Ethernet interface
       product: RTL8125 2.5GbE Controller
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0003:31:00.0
       logical name: enP3p49s0
       version: 05
       serial: c0:74:2b:fe:e8:b7
       capacity: 1Gbit/s
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress msix vpd bus_master cap_list ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=r8125 driverversion=9.009.00-NAPI-RSS latency=0 link=no multicast=yes port=twisted pair
       resources: irq:207 ioport:100000(size=256) memory:f3200000-f320ffff memory:f3210000-f3213fff
  *-network
       description: Ethernet interface
       product: RTL8125 2.5GbE Controller
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0004:41:00.0
       logical name: enP4p65s0
       version: 05
       serial: c0:74:2b:fe:e8:b6
       capacity: 1Gbit/s
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress msix vpd bus_master cap_list ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=r8125 driverversion=9.009.00-NAPI-RSS latency=0 link=no multicast=yes port=twisted pair
       resources: irq:164 ioport:1000(size=256) memory:f4200000-f420ffff memory:f4210000-f4213fff
  *-network DISABLED
       description: Wireless interface
       physical id: 17
       bus info: pci@0002:21:00.0
       logical name: wlan1
       serial: 42:9c:a7:0d:86:79
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8852be driverversion=v1.15.10.0.5-0-gfa2af07cf.20220 firmware=N/A link=no multicast=yes wireless=unassociated
</code>
</details>

Additionally, the working device hangs off a PCI node.
<details><summary>lshw</summary>
<code>*-pci:1
          description: PCI bridge
          product: Rockchip Electronics Co., Ltd
          vendor: Rockchip Electronics Co., Ltd
          physical id: 101
          bus info: pci@0002:20:00.0
          version: 01
[... 154 lines cut ...]
Mar 04 20:04:14 ubuntu systemd-udevd[492]: Using default interface naming scheme 'v249'.
Mar 04 20:04:14 ubuntu kernel: rtl8852be 0002:21:00.0 wlP2p33s0: renamed from wlan0
Mar 04 20:04:14 ubuntu systemd[1]: rtl8852be-reload.service: Deactivated successfully.
Mar 04 20:04:14 ubuntu systemd[1]: Finished Reload RTL8852BE.
Mar 04 20:04:14 ubuntu systemd[1]: Reached target Multi-User System.
Mar 04 20:04:14 ubuntu systemd[1]: Reached target Graphical Interface.
Mar 04 20:04:14 ubuntu systemd[1]: Starting Record Runlevel Change in UTMP...
Mar 04 20:04:14 ubuntu systemd-udevd[498]: Using default interface naming scheme 'v249'.
Mar 04 20:04:14 ubuntu systemd[1]: systemd-update-utmp-runlevel.service: Deactivated successfully.
Mar 04 20:04:14 ubuntu systemd[1]: Finished Record Runlevel Change in UTMP.
Mar 04 20:04:14 ubuntu systemd[1]: Startup finished in 3.636s (kernel) + 2.023s (userspace) = 5.660s.
Mar 04 20:04:14 ubuntu kernel: ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
Mar 04 20:04:14 ubuntu systemd-networkd[480]: wlan0: Interface name change detected, renamed to wlP2p33s0.
Mar 04 20:04:14 ubuntu systemd-udevd[498]: wlan1: Could not set AlternativeName= or apply AlternativeNamesPolicy=, ignoring: File exists
Mar 04 20:04:14 ubuntu systemd-udevd[498]: wlP2p33s0: Failed to rename network interface 7 from 'wlan1' to 'wlP2p33s0': File exists
Mar 04 20:04:14 ubuntu systemd-udevd[498]: wlan1: Failed to process device, ignoring: File exists
Mar 04 20:04:15 ubuntu systemd[1]: dmesg.service: Deactivated successfully.
Mar 04 20:04:15 ubuntu systemd-networkd[480]: wlP2p33s0: Link UP
Mar 04 20:04:16 ubuntu ModemManager[675]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/fe180000.pcie/pci0003:30/0003:30:00.0/0003:31:00.0': not supported by any plugin
Mar 04 20:04:16 ubuntu ModemManager[675]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/fe190000.pcie/pci0004:40/0004:40:00.0/0004:41:00.0': not supported by any plugin
Mar 04 20:04:17 ubuntu ModemManager[675]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/fe170000.pcie/pci0002:20/0002:20:00.0/0002:21:00.0': not supported by any plugin
Mar 04 20:04:19 ubuntu wpa_supplicant[680]: wlP2p33s0: Trying to associate with c8:7f:54:d6:d9:b8 (SSID='SunnyvaleTP' freq=2417 MHz)
Mar 04 20:04:19 ubuntu wpa_supplicant[680]: wlP2p33s0: CTRL-EVENT-REGDOM-CHANGE init=DRIVER type=WORLD
</code>
</details>

Joshua-Riek, 2024-03-05:
The entire kernel is technically Android and one major hack after another. There are very few developers will spend time to properly fix broken or crappy drivers on this messy kernel. But given that WiFi and Bluetooth work, i do not want to spend time on this.

dsburns, 2024-03-05:
Thanks for reviewing this so quickly.  I would argue that if I can only connect to my board if it enumerates the correct wifi interface at boot, then Wifi is broken.  Perhaps there's a udev work around to make the wifi behavior consistent across reboots..

JFLim1, 2024-03-05:
Hi @dsburns,

Are you having WiFi issue with kernel-5.10.160 or kernel-6.1.43?  I have the same WiFi module as you RTL8852BE-CG for Opi5-Plus.

In my case the WiFi is stable on kernel-5.10.160 only occassionally WiFi is not connected and all I have to do is 

sudo modprobe -r 88552be
sudo modprobe -i 88552be

and WiFi will connected and stable.

I have WiFi stability or NO WiFi with kernel-6.1.43.  WiFi essentially does not function on kernel-6.1.43 with RTL8852BE-CG as of now.

Joshua-Riek, 2024-03-05:
I've not released the 6.1 kernel yet, it should not be used for comparison.

dsburns, 2024-03-05:
I'm using the 5.10 kernel with the Ubuntu server image.  I managed to create an ugly workaround that ensures network connectivity via WiFi at boot, but it doesn't address the driver issue.   The suggestion from @JFLim1 doesn't work for my use case since I'm running headless and the system needs to maintain network connectivity across reboots without manual intervention.

JFLim1, 2024-03-06:
> I've not released the 6.1 kernel yet, it should not be used for comparison.

It wasn't meant to be used for comparison between 5.10.160 and 6.1.43.  Wasn't sure which kernel he was using.

For the record, you have done fantastic works on bsp-kernel-6.1.43 and making 3D GPU and vpu hw acceleration available.

arun-511, 2024-03-06:
WiFi modules, specially realtek ones are a mess on rockchip kernel. If you are on a server image, you should configure `/etc/netplan/*.yaml` to connect a specific interface to WIFI specifically that wlP* one and give it dhcp4&6. I have Intel AX21,  8265NGW and 7265NGW and they as appear with wlP* (on mainline-6.7 & rockchip-6.1 builds) with no trace of wlan* interfaces at all.

dsburns, 2024-03-06:
Understood.  But the problem is the naming isn't consistent.  Depending on the timing at boot, you can see the same wlP* device may have been renamed from wlan0 or wlan1.  Look at logs during boot and you can see the errors from the device driver.

Joshua-Riek, 2024-03-07:
This problem will be fixed in Ubuntu 24.04 with the 6.1 Linux kernel. It's still heavily in development but I'm re-visiting the wireless drivers right now.

[image]

KlausBuchegger, 2024-05-04:
I faced the same issue on my Orange Pi 5 Plus (but I'm running armbian). As @dsburns mentioned, configuring the wifi based on the interface name is not reliable, as it randomly changes during boot.
But I noticed, that both interfaces have individual, fixed MAC addresses.
On armbian, I'm using NetworkManager to configure the wifi, and it allows to limit the connection based on the interface MAC.

Instead of specifying the `interface-name` in the `[connection]` section you can specify `mac-address` in the `[wifi]` section.

So far, this seems to be reliable for me. Maybe it also works for you, until it will be fixed properly

closed 2025-02-17 by github-actions[bot]