Joshua-Riek/ubuntu-rockchip #651: kernel only build

wyf9661, 2024-03-05:
Hi, joshua, I'dont know if it is necessary to create this issue, but it does make some affect of my repo which forks from you.
After the latest update, (with ubuntu release involed), Release version must be set so I can not set the `build.sh` with the `kernel-only` command, as when I chose this option, I do not care what release rootfs is, just build kernel only. Or maybe I can just select one of the release ?
[image]

wyf9661, 2024-03-05:
Meanwhile, If `kernel-only` seleced, is `--project` necessary as well?

Joshua-Riek, 2024-03-05:
Yeah, I have not written the parameters well. 

I've been focused on porting hardware accelerated packages for Ubuntu 24.04 and the new 6.1 kernel. I just need something that builds right now :)