Joshua-Riek/ubuntu-rockchip #654: Video playback on Orange Pi 5 Plus

DavidMolTor, 2024-03-06:
I have been having issues with video playback using VLC on my Orange Pi 5 Plus. I installed the v1.31 release and everything worked perfecty. I updated the system every time the prompt appeared and I kept all packages up to date.

Yesterday, after an update, I can no longer play video neither with VLC or any of the preinstalled video players. I don't know if there is an issue with the driver or something like that. I haven't tweaked anything system wise, I just mainly use VLC and Moonlight streaming client, nothing else.

Joshua-Riek, 2024-03-06:
VLC should not work properly with hw accel, you need to use MPV. But you say you cant play video at all which does not make sense, please provide more information.

DavidMolTor, 2024-03-06:
My bad, I was using the videos application instead of MPV. VLC seems to not be using the hardware acceleration since the CPU usage goes through the roof. I will try playing with the VLC setting, but for now I will stick to MPV.

Using MPV I can playback H264 content without issues, but H265 content does no play at all like always. Is there any way to get H265 content to play on the Orange Pi 5 Plus properly with MPV or any other application?

[image]

[image]

Joshua-Riek, 2024-03-06:
If I recall correctly, VLC will need some modifications to the source code to use hardware acceleration. But I do not have time to dig into VLC right now. I thought H265 worked properly, but I'm so busy right now and not sure when i will be able to look further into this.

DavidMolTor, 2024-03-06:
No problem, thanks for the information. I will dig more into getting VLC working. I can survive with MPV and H264 videos. Thanks for the clarifications!

DavidMolTor, 2024-03-06:
This is the error I get when trying to play an H265 file with MPV:

` (+) Video --vid=1 'Encoded by WEM for TAoE using Shinguru V2.1' (hevc 1440x1072 23.976fps)
 (+) Audio --aid=1 --alang=eng (*) '' (dts 2ch 48000Hz)
     Subs  --sid=1 --slang=eng (hdmv_pgs_subtitle)
File tags:
 Title: 
Using hardware decoding (rkmpp).
AO: [pulse] 48000Hz stereo 2ch s32
VO: [gpu] 1440x1072 drm_prime[nv15]
[vo/gpu] Initializing texture for hardware decoding failed.
AV: 00:00:30 / 00:23:50 (2%) A-V:  0.001 Dropped: 43
`

[image]

nyanmisaka, 2024-03-06:
@Joshua-Riek This is the concern that i described before. Amazingfate updated his ppa, now the issue comes to existing users (if they `apt-get upgrade`).

@DavidMolTor You can fix it by adding a one line config: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/613#issuecomment-1974390736

JFLim1, 2024-03-06:
Hi @Joshua-Riek & @nyanmisaka,

After upgrading to the latest multimedia packages mpv have not issue streaming 10Bit HEVC youtube videos.

```
jfl@jfl-opi5plus:~$ mpv https://www.youtube.com/watch?v=7PIji8OubXU
 (+) Video --vid=1 (*) (vp9 1920x1080 60.000fps)
 (+) Audio --aid=1 --alang=eng (*) (opus 2ch 48000Hz)
File tags:
 Uploader: 8K Earth
 Channel_URL: https://www.youtube.com/channel/UChB3UnDddahXU7FKZXmpzMA
Using hardware decoding (rkmpp).
rga_api version 1.10.0_[8]
AO: [pulse] 48000Hz stereo 2ch float
VO: [gpu] 1920x1080 drm_prime[p010]
AV: 00:00:30 / 00:12:16 (4%) A-V:  0.000 Dropped: 27 Cache: 34s/31MB

Exiting... (Quit)
jfl@jfl-opi5plus:~$ 
jfl@jfl-opi5plus:~$ 
jfl@jfl-opi5plus:~$ mpv https://www.youtube.com/watch?v=k7dy1B6bOeM
 (+) Video --vid=1 (*) (vp9 1920x1080 59.940fps)
 (+) Audio --aid=1 --alang=eng (*) (eac3 6ch 48000Hz)
File tags:
 Uploader: 4K Video ULTRA HD
 Channel_URL: https://www.youtube.com/channel/UCRauo6y9tx0FB0OLLtqLA-g
Using hardware decoding (rkmpp).
rga_api version 1.10.0_[8]
AO: [pulse] 48000Hz 5.1(side) 6ch float
VO: [gpu] 1920x1080 drm_prime[p010]
AV: 00:00:28 / 00:11:11 (4%) A-V:  0.000 Dropped: 9 Cache: 27s/25MB

Exiting... (Quit)
```

DavidMolTor, 2024-03-08:
> @Joshua-Riek This is the concern that i described before. Amazingfate updated his ppa, now the issue comes to existing users (if they `apt-get upgrade`).
> 
> @DavidMolTor You can fix it by adding an one line config: [#613 (comment)](https://github.com/Joshua-Riek/ubuntu-rockchip/issues/613#issuecomment-1974390736)

The fix for MPV you referenced worked perfectly. I will work on getting VLC to work as it used to, but for now MPV is able to play H264 and H265 smoothly. Thanks!

closed 2024-03-09 by Joshua-Riek