Joshua-Riek/ubuntu-rockchip #655: Random reboot

rjyo, 2024-03-07:
My Pi 5 Plus is rebooting every few hours. How to find out the reasons?

```
$ last reboot

reboot   system boot  5.10.160-rockchi Thu Mar  7 09:16   still running
reboot   system boot  5.10.160-rockchi Wed Mar  6 20:18   still running
reboot   system boot  5.10.160-rockchi Wed Mar  6 14:34   still running
reboot   system boot  5.10.160-rockchi Wed Mar  6 08:03   still running
reboot   system boot  5.10.160-rockchi Tue Mar  5 22:47   still running
reboot   system boot  5.10.160-rockchi Mon Mar  4 23:42   still running
reboot   system boot  5.10.160-rockchi Mon Mar  4 09:23   still running
reboot   system boot  5.10.160-rockchi Mon Mar  4 09:17   still running
reboot   system boot  5.10.160-rockchi Sun Mar  3 13:01   still running
reboot   system boot  5.10.160-rockchi Sun Mar  3 12:02   still running
reboot   system boot  5.10.160-rockchi Sat Mar  2 16:38   still running
```

flyzet-prog, 2024-03-09:
what power supply did you use?

rjyo, 2024-03-10:
Thanks so much for the hint! I think that's just my problem. I have it on a 2-ports USBC PD charger, which redistribute power whenever a new device connected. The reboot log of my Pi5 Plus turned out to be also the charger utilization log...

closed 2024-03-10 by rjyo

strawgate, 2024-04-02:
I ran into a similar issue where it would reboot right when I logged in. Was also an underpowered power supply.