Joshua-Riek/ubuntu-rockchip #658: Issues with RTL8821CU Chipset

ByteArrayException, 2024-03-09:
Hi Joshua,
first of all thanks for your great work here with the Ubuntu integration of different kernels and releases on Rockchip based systems, but I have a problem with WiFi sticks on Ubuntu 22.04 and the 6.1 kernel based with the RTL8821CU chipset (I know almost everything has been fixed since kernel 6.2), it randomly dies and the dmesg log says:
"[ 4450.912183] rtw_8821cu 3-1.3:1.0: timed out to flush queue 1"
and
"
[ 4515.963899] rtw_8821cu 3-1.1:1.0: failed to download firmware
[ 4515.968271] rtw_8821cu 3-1.1:1.0: leave idle state failed
[ 4515.971775] rtw_8821cu 3-1.1:1.0: failed to send h2c command
[ 4515.987397] rtw_8821cu 3-1.1:1.0: failed to leave ips state
[ 4515.987425] rtw_8821cu 3-1.1:1.0: failed to leave idle state
"

Joshua-Riek, 2024-03-09:
I may be able to backport the driver from Linux 6.2. But can you put dmesg output into a text file and upload it? I know with Ubuntu 24.04 they now use compressed firmware, it may be an issue for some WiFi / Bluetooth modules.

Joshua-Riek, 2024-03-09:
Hmm i do not see RTL8821CU in 6.2

Joshua-Riek, 2024-03-09:
Could you try adding `rtw88_pci.disable_aspm=true rtw88_core.lps_deep_mode=0` to the kernel cmdline.
https://github.com/sebanc/brunch/issues/855

Joshua-Riek, 2024-03-09:
But do note, i have no plans on porting 6.1 to ubuntu 22.04.

ByteArrayException, 2024-03-09:
The dmesg output is here: [dmesg_output.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/14547742/dmesg_output.txt)
I think the RTL8821CU Driver is Part of the RTW88.
However, as I read in this GitHub post, many WLAN driver problems are said to have been solved as of kernel version 6.2: https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md
Do you mean that you shouldn't run kernel 6.1 with 22.04.4 anyway? I didn't know that. Should I rather go for 24.03/24.04 (noble)?

ByteArrayException, 2024-03-09:
> Could you try adding `rtw88_pci.disable_aspm=true rtw88_core.lps_deep_mode=0` to the kernel cmdline. [sebanc/brunch#855](https://github.com/sebanc/brunch/issues/855)

I have tested this, but now no WLAN connects at all.

ByteArrayException, 2024-03-09:
I can build a mainline kernel as a test and see if it works. This would at least give me a clue as to whether everything really works much better from 6.2+ as described in the post.

Joshua-Riek, 2024-03-09:
You can run 6.1 technically on Ubuntu 22.04, but I will not support it in any way. This is due to following the Ubuntu philosophy of introducing a new kernel version with a new release, users will complain that something is broken on the already-known working Ubuntu 22.04 image. Ubuntu 24.04 is not out yet and is under heavy development, use it at your own risk.

Forgive me if this sounds rash, but since I'm a single person I can only do so much.

Joshua-Riek, 2024-03-09:
But I will still try and backport rtw88 from linux 6.2

mizzuritv, 2024-03-09:
Yes, all good, I am grateful anyway or I think the whole fan community of this repo that you do so much, if you see all these repos with linux-rockchip and then the repo here with ubuntu-rockchip, you are always trying very hard 24/7 to develop it further. But for which release is rockchip-6.1 initially intended? 22.04 even if it is not supported?
Thank you in any case for trying to integrate the drivers from 6.2.

ByteArrayException, 2024-03-09:
Yes, all good, I am grateful anyway or I think the whole fan community of this repo that you do so much, if you see all these repos with linux-rockchip and then the repo here with ubuntu-rockchip, you are always trying very hard 24/7 to develop it further. But for which release is rockchip-6.1 initially intended? 22.04 even if it is not supported?
Thank you in any case for trying to integrate the drivers from 6.2.

Joshua-Riek, 2024-03-09:
The release for rockchip 6.1 will be for Ubuntu 24.04. 

My initial testing was with Ubuntu 22.04 because at the time deboostrap was broken so I was not able to build Ubuntu 24.04 images for development.

ByteArrayException, 2024-03-09:
Another thing I noticed is that nmcli says that sometimes rtl8821cu is loaded, but sometimes also rtw_8821cu, where I think the former works better. Is it correct that both drivers are integrated?

ByteArrayException, 2024-03-09:
> The release for rockchip 6.1 will be for Ubuntu 24.04.
> 
> My initial testing was with Ubuntu 22.04 because at the time deboostrap was broken so I was not able to build Ubuntu 24.04 images for development.

Ah, okay, I see.

ByteArrayException, 2024-03-10:
Hi, on another note, I found the GitHub post referencing this issue: https://github.com/morrownr/8821cu-20210916/issues/115#issuecomment-1739343920
It mentions that it should work with the repo https://github.com/lwfinger/rtw88. Maybe you can integrate it (if it doesn't work with kernel 6.2). Thanks for your effort!

morrownr, 2024-03-11:
Regarding the above links: lwfinger was involved in the 8821cu project to improve the support in rtw88. There were five main patches that are in wireless-next and are headed into kernel 6.9 so what you might want to do is wait until 6.9 is released for testing and take a look at the 8821cu support in rtw88. I see that some, if not all, of the problems you were reporting have been fixed.

closed 2025-02-17 by github-actions[bot]