Joshua-Riek/ubuntu-rockchip #660: Orangepi 5 Plus sound issue on 1.33 update(5.10 kernel)

flyzet-prog, 2024-03-13:
Hi everyone. I can't use sound card, no sound when plugin headphone jack or with HDMI(when i plugin jack it does recognize headphone or headset). If i play some video on youtube or MPV player it's just stuck and i see loading logo, like waiting for audio, but it's should play even if no audio devices connected. This issue from the box, on clean system. Today i tried to install on SD card new [kernel 6.1](https://github.com/Joshua-Riek/ubuntu-rockchip/commit/087fba498c2cff23f972330026f8300e3ddac487). And sound works properly without any problem, HDMI, jack, even bluetooth audio. Maybe some one can help to resolve this or i  can try to reinstall some drivers,libraries, etc. Much appreciated for any reply.
[image]
[image]

**This is Screenshots from kernel 6.1**

[image]
[image]
[image]
[image]
[image]
[image]

Joshua-Riek, 2024-03-13:
Have you installed the 6.1 kernel on Ubuntu 22.04 and that is a problem? Sound should be working with 6.1 on Ubuntu 24.04, I've not test Ubuntu 22.04 or plan to support it with the 6.1 kernel.

flyzet-prog, 2024-03-13:
> Have you installed the 6.1 kernel on Ubuntu 22.04 and that is a problem? Sound should be working with 6.1 on Ubuntu 24.04, I've not test Ubuntu 22.04 or plan to support it with the 6.1 kernel.

I'v installed 6.1 on SD card and booted from it and sound wokrs properly. But with Ubuntu 24.04 kernel 5.1 i have issue, installed like main system on NVME. and sound dosen't work it's spinning circle while playing video on youtube. When i plug mini jack and chose headphones,  the same issue. Can you please suggest me some module to reload or reinstall it which related to sound card? Thank you for your time.

Joshua-Riek, 2024-03-13:
I'm confused on what system your running that has an issue.. Please tell me what ubuntu version and kernel that has a problem.

flyzet-prog, 2024-03-13:
My kernel is 5.10.160-rockchip and Ubuntu 22.04.4 LTS

flyzet-prog, 2024-03-13:
> My kernel is 5.10.160-rockchip and Ubuntu 22.04.4 LTS

My kernel is 5.10.160-rockchip and Ubuntu 22.04.4 LTS

Joshua-Riek, 2024-03-13:
Sound works perfectly on that system, you likely broken some audio config. Please try and re-install the OS.

flyzet-prog, 2024-03-13:
> Sound works perfectly on that system, you likely broken some audio config. Please try and re-install the OS.
OK. I will. Any way thank you for your reply

closed 2024-03-13 by flyzet-prog