Joshua-Riek/ubuntu-rockchip #664: feat: enable cgroups v2

jgresham, 2024-03-15:
Is there a reason that cgroups is set to v1? Could we enable v2?

I think setting `systemd.unified_cgroup_hierarchy=1` could be a starting point.
https://github.com/Joshua-Riek/ubuntu-rockchip/blob/ebda769fe41b240283c52f7f0a6ef446d6ccc852/scripts/build-image.sh#L202

jgresham, 2024-03-25:
@Joshua-Riek is this change a good idea?

Joshua-Riek, 2024-03-26:
Yeah, cgroups v1 has been removed by default for the future releases.

closed 2024-03-26 by jgresham

RabHanz, 2024-05-13:
Is it possible to enable cgroup v1 as the Home Assistant Supervised requires that to function. In Ubuntu 22 I managed to make it work by modifying /boot/firmware/ubuntuEnv.txt, but I can't seem to find this file in the latest Ubuntu 24.

Joshua-Riek, 2024-05-13:
@RabHanz please view the below documentation on how to change the linux boot parameters.

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Ubuntu-24.04-LTS#changing-the-linux-boot-parameters

s1gtrap, 2024-09-04:
@shoop 
> FWIW, I'm succesfully running k3s on a 3-node Turing RK1 cluster based on the 22.04 image with the cgroup v1 parameters removed:
> 
> ```
> ubuntu@tpinode1:~$ cat /proc/cmdline
> root=UUID=b24ca5e3-dba8-47ea-a1a1-569c7bcbbae3 rootfstype=ext4 rootwait rw console=ttyS9,115200 console=ttyS2,1500000 swapaccount=1
> ```
> 
> I don't know why cgroup v1 was chosen as default, it seems to work with v2.

Hi there, I'm trying to get rootless k3s working, so I tried to reproduce your setup, and after removing the cgroup entries such that
```
$ cat /proc/cmdline
root=UUID=768f99c6-c04b-451a-8219-fd2491356a70 rootwait rw console=ttyS9,115200 console=tty1
```
cgroup v2 is seemingly still working as
```bash
$ cat /sys/fs/cgroup/cgroup.controllers
cpuset cpu io memory hugetlb pids rdma
```

Unfortunately, cgroup v1 is seemingly also still enabled, as
```bash
$ grep cgroup /proc/filesystems
nodev	cgroup
nodev	cgroup2
```
and my k3s service fails to get running with
```
Sep 04 19:05:22 ubuntu k3s[1461]: time="2024-09-04T19:05:22Z" level=info msg="Wrote kubeconfig /home/rootless/.kube/k3s.yaml"
Sep 04 19:05:22 ubuntu k3s[1461]: time="2024-09-04T19:05:22Z" level=info msg="Run: k3s kubectl"
Sep 04 19:05:22 ubuntu k3s[1461]: time="2024-09-04T19:05:22Z" level=fatal msg="failed to find cpuset cgroup (v2)"
```
which I suspect is stemming from the fact that k3s in rootless mode [doesn't support hybrid cgroup v1/v2.](https://docs.k3s.io/advanced#known-issues-with-rootless-mode)

I'm not sure if this is directly related to this issue, but would love to hear if you have had any luck disabling cgroup v1 entirely?

The man pages for cgroup are also a bit confusing, as `systemd.unified_cgroup_hierarchy` has been deprecated since  systemd 252?
```
Kernel command-line arguments systemd.unified_cgroup_hierarchy and systemd.legacy_systemd_cgroup_controller were deprecated. Please switch to the unified cgroup hierarchy.
```
I would love to hear if anybody else has some input, as I was thinking about maybe submitting a new issue or a Q&A discussion regarding this.

shoop, 2024-09-05:
hi @s1gtrap,

I did not try to not run k3s in rootless mode, so I did not encounter this problem. Possibly I mistook the appearance of the cgroups v1 parameters to mean that cgroups v2 was disabled -- hence my comment.

I do not know of a way to disable cgroups v1 so I think a new discussion would be warranted.