Joshua-Riek/ubuntu-rockchip #669: Banana Pi BPI-M7 support

abravetti, 2024-03-19:
Hello, would it be possible to support Banana Pi BPI-M7?

https://wiki.banana-pi.org/Banana_Pi_BPI-M7

Thanks!

armsom-team, 2024-04-10:
It is indeed the same board, you just need to use the armsom-sige7 firmware, https://docs.armsom.org/zh/armsom-sige7

Joshua-Riek, 2024-04-10:
I would make another image with the Banana Pi M7 name, but I currently facing the 2GB artficat limit on GitHub for the upcoming Ubuntu 24.04 desktop images. There is a high chance I will need to buy a server to host downloads as the projected file size is 2.5GB. I have the Snap Store, Firefox, and Thunderbird removed to stay under 2GB, but there is only 100MB of space before reaching the 2GB limit again.

closed 2024-04-11 by Joshua-Riek