Joshua-Riek/ubuntu-rockchip #672: main/overlay/boot/firmware/meta-data instance_id key incorrect

ralph-mcardell, 2024-03-21:
In the boot partition meta-data file used by Cloud Init the key 'instance_id' is incorrect and should be 'instance-id', with a hyphen rather than an underscore.

This was discovered while experimenting with modifying an installation on a Turing Pi RK1 module and wondering why:

1) Initially the instance id was 'nocloud' and not 'cloud-image' in the data written by Cloud Init to /var/lib/cloud
2) Modifying the value from 'cloud-image' to something else failed to cause Cloud Init to re-perform the first boot actions.

Additional:
Not sure it is required but seems the 'ds' key value for NoCloud data provider is currently documented as 'nocloud' rather than 'local'.

Joshua-Riek, 2024-03-21:
Good catch, thank you for finding this!

Joshua-Riek, 2024-03-21:
Could you point me to the documentation where you see 'NoCloud' being used instead of 'local'? I may be looking at the wrong parameter.

https://cloudinit.readthedocs.io/en/latest/reference/datasources/configdrive.html#ds-mode

ralph-mcardell, 2024-03-21:
Ah! Sorry I seem to have confused my self between ds and dsmode. 
Please ignore that part of my issue. I think local for dsmode is correct.

Again my apologies for the confusion.

closed 2024-04-15 by Joshua-Riek