Joshua-Riek/ubuntu-rockchip #673: Swap file automatically generates on boot

SimplyCorbett, 2024-03-21:
Hi,

I'm using the ubuntu-22.04.3-preinstalled-server-arm64-orangepi-5-plus image and there is a 1GB swap file in root that automatically generates every reboot. This is on a SD card so I definitely don't want a swap file to be generated on reboot. I have to manually disable swap by logging into SSH after rebooting. Deleting the file doesn't prevent it from regenerating on reboot.

This file doesn't exist in /etc/stab. Is this documented anywhere? I can't find any details about this on google. What's the best method to permanently disable this?

Or does the x-systemd.growfs automatically generate it?

icache, 2024-03-22:
try:

sudo systemctl stop swapfile.swap 
sudo systemctl stop mkswap.service 

sudo systemctl mask mkswap.service 
sudo systemctl mask swapfile.swap

sudo rm -rf /swapfile

SimplyCorbett, 2024-03-22:
Yes, I had to mask it. Disabling the services didn't work.

Thank you so much :)

SimplyCorbett, 2024-03-22:
Wouldn't that also disable swap file generation on other specified devices? For example, I might want to put a swap file on the nvme drive.

The problem is the 1GB swap file still generates regardless on the root partition of the SD card, I've tested that.

s1n7ax, 2024-09-30:
I can confirm. After disabling the swap using `sudo systemctl disable swapfile.swap`, `swapon --show` shows nothing but the swap file is generated regardless. Not a major issue but still it takes 1GB of storage right?

MarcAndreWyss, 2024-12-01:
I have not tested it, but my assumption is that this is just happening for the desktop version and not the server version. Maybe someone has time to test my assumption. If I have some time I will probably test it myself.

During my try to fix #1130 @Joshua-Riek pointed me to the project / package [ubuntu-rockchip-settings](https://github.com/Joshua-Riek/ubuntu-rockchip-settings). If my interpretation of this project / package is correct, then the swapfile is generated, because of the settings here: https://github.com/Joshua-Riek/ubuntu-rockchip-settings/tree/noble/data/desktop (mkswap.service and swapfile.swap). Those settings to not exist in the server version here: https://github.com/Joshua-Riek/ubuntu-rockchip-settings/tree/noble/data/server

Therefore my interpretation (which may be wrong) is that these settings were done by intention for the desktop version and not by accident. Therefore my assumption would be that this is not a bug in the sense of an unexpected behaviour, because it looks like an architectual decision.

However I agree that a swap file does not make much sense on a board with a lot of RAM. But if one would remove the settings then people using boards with few RAM would have to always activate swap consciously. I have no idea what @Joshua-Riek tought's were. But when thinking of it, this might be a topic for the wiki here: https://github.com/Joshua-Riek/ubuntu-rockchip/wiki

closed 2025-02-17 by github-actions[bot]