Joshua-Riek/ubuntu-rockchip #674: AX210 Bluetooth does not work on a warm reboot

Joshua-Riek, 2024-03-24:
@nyanmisaka found that the USB Bluetooth for AX210 only works on a cold boot. When entering the command `sudo restart`, the Bluetooth USB disappears. This has been observed on the 5.10 and 6.1 Rockchp kernels and multiple RK3588 devices.

ref: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/613#issuecomment-2016803424
ref: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/613#issuecomment-2016889215
ref: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/613#issuecomment-2016895230

Joshua-Riek, 2024-03-24:
Kernel logs where AX210 works after a cold boot: [dmesg.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/14736446/working_dmesg.txt)

Kernel logs after a warm reboot where AX210 Bluetooth does not work: [dmesg.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/14736444/broken_dmesg.txt)

nyanmisaka, 2024-03-25:
The difference that can be noticed is that the following is added to the working dmesg log.
rfkill enables usb bluetooth (0x8087 intel bt).
```
[   48.488906] [BT_RFKILL]: rfkill_rk_set_power: set bt wake_host high!
[   48.541501] [BT_RFKILL]: rfkill_rk_set_power: set bt wake_host input!
[   48.541529] [BT_RFKILL]: ENABLE UART_RTS
[   48.648168] [BT_RFKILL]: DISABLE UART_RTS
[   48.648213] [BT_RFKILL]: bt turn on power
[   48.648231] [BT_RFKILL]: Request irq for bt wakeup host
[   48.648259] [BT_RFKILL]: ** disable irq
[   48.988155] usb 1-1.1: new full-speed USB device number 3 using xhci-hcd
[   49.102853] usb 1-1.1: New USB device found, idVendor=8087, idProduct=0032, bcdDevice= 0.00
[   49.102903] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   49.167625] usbcore: registered new interface driver btusb
[   49.169354] Bluetooth: hci0: Device revision is 0
[   49.169366] Bluetooth: hci0: Secure boot is enabled
[   49.169371] Bluetooth: hci0: OTP lock is enabled
[   49.169375] Bluetooth: hci0: API lock is enabled
[   49.169380] Bluetooth: hci0: Debug lock is disabled
[   49.169384] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[   49.169390] Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
[   49.188450] Bluetooth: hci0: Found device firmware: intel/ibt-0041-0041.sfi
[   49.188613] Bluetooth: hci0: Boot Address: 0x100800
[   49.188622] Bluetooth: hci0: Firmware Version: 60-48.23
[   49.276405] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   49.276420] Bluetooth: BNEP filters: protocol multicast
[   49.276434] Bluetooth: BNEP socket layer initialized
[   50.451375] Bluetooth: hci0: Waiting for firmware download to complete
[   50.452315] Bluetooth: hci0: Firmware loaded in 1234226 usecs
[   50.452378] Bluetooth: hci0: Waiting for device to boot
[   50.477315] Bluetooth: hci0: Device booted in 24370 usecs
[   50.477320] Bluetooth: hci0: Malformed MSFT vendor event: 0x02
[   50.478805] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-0041-0041.ddc
[   50.480325] Bluetooth: hci0: Applying Intel DDC parameters completed
[   50.483318] Bluetooth: hci0: Firmware timestamp 2023.48 buildtype 1 build 75324
[   50.559492] Bluetooth: MGMT ver 1.22
[   50.565943] Bluetooth: RFCOMM TTY layer initialized
[   50.565956] Bluetooth: RFCOMM socket layer initialized
[   50.565964] Bluetooth: RFCOMM ver 1.11
[   53.202978] [BT_RFKILL]: bt shut off power
```

nyanmisaka, 2024-04-02:
This issue is not specific to rockchip kernel, but to `ubuntu-rockchip`.
I built armbian from scratch and found that it was not affected.

Joshua-Riek, 2024-04-02:
Interesting. I will need to do some digging.

closed 2025-02-17 by github-actions[bot]