Joshua-Riek/ubuntu-rockchip #678: Rock 5A Cannot Enable Radxa Display 10fhd Display

MyVizDrake, 2024-03-27:
Hi! 

I had this display working via an Arch Installation at one point using the Radxa 5.10BSP kernel but would prefer to try your build!

System Info

[image]

Contents of /boot/firmware/ubuntuEnv.txt
> bootargs=root=UUID=9b8c0d17-0823-45cd-834d-bf08749b52b5 rootfstype=ext4 rootwait rw console=ttyS2,1500000 console=tty1 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1 systemd.unified_cgroup_hierarchy=0 quiet splash plymouth.ignore-serial-consoles
>fdtfile=rk3588s-rock-5a.dtb
>overlay_prefix=rk3588
>overlays=rock-5a-radxa-display-10fhd

I tried it with the full path in the overlays statement (e.g. /boot/firmware/overlays/rock-5a-radxa-display-10fhd
I tried it with the .dtbo extension
I tried it with the .dtbo extension and the full path

I have tried it commenting out the overlay_prefix line.

The external display does not power up nor display anything.

Any ideas where to look? What to try? What did I do wrong?

appreciate the help!

handledexception, 2024-06-02:
@MyVizDrake Joshua said that he's created a beta build that where the display portion of this device works. See the discussion in this issue: https://github.com/Joshua-Riek/ubuntu-rockchip/issues/643

These comments from Joshua specifically detail what you need to do as of right now to test the Display 10 FHD. Just make sure you specific `U_BOOT_FDT_OVERLAYS="device-tree/rockchip/overlay/rock-5b-radxa-display-10fhd.dtbo`, rather than `10hd` that Joshua mentioned in his second comment below.

https://github.com/Joshua-Riek/ubuntu-rockchip/issues/643#issuecomment-2143686226
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/643#issuecomment-2143687451

MyVizDrake, 2024-06-02:
@handledexception Yeah we're in[ communication on the Radxa Discord](https://discordapp.com/channels/855634073376260096/1246637381336039524) The touch driver for teh 10FHD should be goodix_ts or goodix_gt9x (?) according to the latest radxa overlay DTS.  I couldn't get touch to work on either the test 22.04 or test 24.04 image and am now trying the Radxa Jammy KDE image

MyVizDrake, 2024-06-02:
<sigh> using the radxa B18 Jammy KDE image ... rsetup doesn't even offer that overlay, yet it is in /usr/lib/linux-image-5.10<blah blah>/rockchip/overlays/  Copied it to root and installed it via rsetup and it is bricked and boots from the eMMC (arch). 

reimaged and added '/usr/lib/linux-image-5.10<blah blah>/rockchip/overlays/' to to the /etc/default/u-boot U_BOOT_FDT_DIR variable and 'rock-5a-radxa-display-10fhd.dtbo' to the U_BOOT_FDT_OVERLAYS variables and then ran 'sudo 'u-boot-update' and rebooted.  fails to boot and proceeds to boot from the Arch install on eMMC again so at this point I have no idea.

UPDATE: apparently i have corrupted this SD card with all of the re-image steps I have been doing over the last month. :-(

Grabbing a new SD card ...

Joshua-Riek, 2024-06-02:
OK i got it working finally on Linux 5.10 and 6.1 as shown below:
```
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 6.1.0-1014-rockchip #14 SMP Sun Jun  2 16:44:21 EDT 2024 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@ubuntu:~$ dmesg | grep -i good
[   15.893025] goodix_ts_probe() start
[   15.893042] Goodix-TS 6-0014: Looking up tp-supply from device tree
[   16.047254] input: goodix-ts as /devices/platform/fec80000.i2c/i2c-6/6-0014/input/input1
```

Apparently the driver does not being a kernel module so it needs to be built in. Since the 10FHD panel uses the same goodix driver i expect it to also work now.

MyVizDrake, 2024-06-02:
@Joshua-Riek awesome!  can't wait to see it working!  Thanks for all of your efforts and your patience!

Joshua-Riek, 2024-06-02:
Below is the beta build **hopefully** will work, it still needs a few hours to finish.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9341238827

MyVizDrake, 2024-06-02:
Thanks again !!!! if I install your current 2.1.0 image on my eMMC today, will the fix work via an apt update / upgrade or will it require a reinstall?  I have my eMMC pulled and if it will be an apt update / upgrade fix, then I will flash the latest image to it in preparation and then test for you via SD card once the test builds complete.   If it will require a reinstall then I wait until I can test the test-build and then it gets rolled into an updated release.

Joshua-Riek, 2024-06-02:
You will be able to get the changes from a simple apt update. The purpose of building images on GitHub is for testing reasons. I try to make sure changes work before pushing them out in an update as I do not want to break anyone's system.

Joshua-Riek, 2024-06-03:
I double checked the beta build and the Display 8HD works on both Ubuntu 22.04 and 24.04.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/9341238827

MyVizDrake, 2024-06-03:
flashing now ....

MyVizDrake, 2024-06-03:
I can confirm that touch now works on a 5a with 24.04 / 6.1 !!!!!!!! need me to test 22.04 as well?

Joshua-Riek, 2024-06-03:
If 24.04 works, then im sure 22.04 will work, thanks for your patience!

closed 2024-06-03 by Joshua-Riek

Joshua-Riek, 2024-06-03:
I created tags for the kernel changes, the kernel update should be out tomorrow. 
- https://github.com/Joshua-Riek/linux-rockchip/releases/tag/Ubuntu-rockchip-5.10.0-1006.6
- https://github.com/Joshua-Riek/linux-rockchip/releases/tag/Ubuntu-rockchip-6.1.0-1015.15