Joshua-Riek/ubuntu-rockchip #680: Overlays for the Radxa zero 3 cameras

Qengineering, 2024-03-30:
First of all, I deeply respect all the work you have done for Rockchip Ubuntu. It is a massive accomplishment!
Your OS is much better and faster than the Debian of Rockchip.

One request. If you can find the time, could you be so kind as to look at the overlays for the Radxa Zero 3 camera?
As where the RK3568 NPU overlay works without a glitch on the Radxa Zero 3, the native Radxa Zero 3 overlays for the RPI cameras fail. Undoubtedly missing some additional drivers, not yet found on your OS.

Joshua-Riek, 2024-03-30:
I'm currently working on development for Ubuntu 24.04 as the beta is estimated to be out next week and I would like to publish a beta release for users to test and provide feedback. I have a lot that needs to be done to prepare so I'm not sure when I will be able to take a look at the camera drivers and overlays.

DangerD1024, 2024-07-01:
any update with that?

Joshua-Riek, 2024-07-01:
I added some rk3566 overlays. Idk I'd any of the cameras work.

Joshua-Riek, 2024-07-01:
Tbh, you're welcome to send a pull request, I have a lot of people ask for things. I can't possibly recall or do every request.

DangerD1024, 2024-07-01:
> I added some rk3566 overlays. Idk I'd any of the cameras work.

can you send me link to list of them?

DangerD1024, 2024-07-01:
cant find any link to wiki with overlays...

Joshua-Riek, 2024-07-01:
Check the kernel source

DangerD1024, 2024-07-01:
> Check the kernel source

https://github.com/Joshua-Riek/ubuntu-rockchip/tree/main/overlay
?

DangerD1024, 2024-07-01:
> Check the kernel source

https://github.com/Joshua-Riek/linux-rockchip/
?
where?)

DangerD1024, 2024-07-01:
Or its already in APT repo?

EvgenSpinin, 2024-09-05:
did you manage to connect the OV5647 camera to Radha Zero3? 
I have the same problem too

ubuntu@ubuntu:~$ dmesg | grep -i csi
[   12.614792] SCSI subsystem initialized
[   14.853540] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 242)
[   14.855436] rockchip-csi2-dphy-hw fe870000.csi2-dphy-hw: csi2 dphy hw probe successfully!
[   14.856624] rockchip-csi2-dphy csi2-dphy0: csi2 dphy0 probe successfully!
[   16.621086] rockchip-mipi-csi2-hw fdfb0000.mipi-csi2-hw: enter mipi csi2 hw probe!
[   16.628895] rockchip-mipi-csi2-hw fdfb0000.mipi-csi2-hw: probe success, v4l2_dev:mipi-csi2-hw!
[   16.701046] rkisp rkisp-vir0: Entity type for entity rkisp-csi-subdev was not initialized!

DangerD1024, 2024-11-29:
> did you manage to connect the OV5647 camera to Radha Zero3? I have the same problem too
>
> ubuntu@ubuntu:~$ dmesg | grep -i csi [ 12.614792] SCSI subsystem initialized [ 14.853540] Block layer SCSI [...]

No, I moved to raspberry since here you have a lot of problems and lack of support...

Joshua-Riek, 2024-11-30:
> No, I moved to raspberry since here you have a lot of problems and lack of support...

Excuse me, do you think I get paid for any of this? People like you are one of the many reasons I don't care anymore.

DangerD1024, 2024-11-30:
> Excuse me, do you think I get paid for any of this? People like you are one of the many reasons I don't care anymore.

sorry, I'm not taking about you, I'm overall about rockchip, on raspberry it's easy to make everything, on Rockchip it's pain every time... Last my fight was with rockchip NPU, but I failed...

closed 2025-02-17 by github-actions[bot]