Joshua-Riek/ubuntu-rockchip #687: [OrangePI 5+][WebGL] browsers freeze

eshcheglov, 2024-04-04:
Hello,

**Problem**: Opening complex WebGL pages causes some browsers to completely freeze.

**Conditions**: Orange Pi 5+, Kernel 5.10.160-rockchip, "desktop" build.

**Steps to reproduce**:
* Install Firefox from the .deb package (http://ports.ubuntu.com/pool/main/f/firefox/firefox_124.0.2+build1-0ubuntu0.20.04.1_arm64.deb).
 * Visit a page with a WebGL demo (https://webglsamples.org/aquarium/aquarium.html or https://browserbench.org/MotionMark1.3/).

This also occurs in the COG browser, which uses WebKit WPE.

I could not reproduce the problem in Chromium or Epiphany.

Joshua-Riek, 2024-04-04:
The only browser that supports GPU and VPU acceleration is Chrome. These other browsers will likely need patches for it to work properly.

eshcheglov, 2024-04-05:
Got it. 

Could you advise how exactly you adapted Chromium? I'm currently testing CEF (Chromium Embedded Framework) and have encountered a problem with freezes, even though CEF uses the same engine as Chrome

Thank you!

closed 2024-04-15 by Joshua-Riek