Joshua-Riek/ubuntu-rockchip #697: Few Issue Orange Pi 5 Plus

Beamaccount, 2024-04-11:
Hi, i've installed your last realase 1.33 but i've few issue:

- During boot i see the continous log rk3x-i2c fec800000 ..... and after 20s boot normally
- i can't launch command sudo u-boot-install-mtd /dev/mtdblock0 . it say me /usr/bin/u-boot-install-mtd: block device '/dev/mtd0' not found
- Ethernet Port not work
- Front USB 3.0 not work

i think something not go well, can you help me to fix?

thanks

Joshua-Riek, 2024-04-11:
Can you provide boot logs? I know the ethernet and usb ports all work correctly. Do you know what power supply are you using? Make sure its 5V 4A.

Beamaccount, 2024-04-11:
i've try different power supply without success. I've also try to boot the old image (original from OrangePi in another SD and boot without eroors or problems).

Joshua-Riek, 2024-04-11:
This is user error I just checked the latest v1.33 release.

Beamaccount, 2024-04-11:
Download and Flash again solved. My laziness to reset the entire environment made me waste more time debugging.... Experience for the next one