Joshua-Riek/ubuntu-rockchip #699: [Question] Supported WIFI dongles?

Krobys, 2024-04-12:
Hello. Could you please advise how I can determine which WiFi dongle I need to buy to ensure compatibility with my Orange Pi 5 and the ubuntu-rockchip build, for instance, for kernels 5.10.160 or 6.1 beta that you are currently working on? Is there a hardcoded list of supported chips, or how should I choose a WiFi USB dongle that will be supported?

Sorry if the question is stupid or not directly related to your build, but rather to the kernel, but I would be grateful if you could tell me the answer.

Sonixier, 2024-04-12:
I am using Radxa A8 wireless module, it works perfectly out of the box

Joshua-Riek, 2024-04-15:
I cant really provide a list, it would take so much time.

closed 2024-04-15 by Joshua-Riek