Joshua-Riek/ubuntu-rockchip #700: Samba shared folders not working

CmansonM, 2024-04-12:
Is samba supported? I've been creating network shared folders, adding users, etc  with no success, I'm using the desktop version

Joshua-Riek, 2024-04-15:
is your user apart of the sambashare group?

ewaldc, 2024-04-26:
I am using Samba on Joshua's distribution since the very beginning (as a NAS server).  The only "problems" I have encountered are with syncing files (it can take days for some files to sync), performance (due to ethernet performance issue) and some issues with masks for users belong to the same unix group.  Have you set "_server min protocol_" to SMB2_10 ?  Is your user and samba password added (using: _smbpasswd -a <user>_) ?

closed 2025-02-17 by github-actions[bot]