Joshua-Riek/ubuntu-rockchip #702: OrangePi5 WiFi not working Ubuntu 24.04 beta (Server)

redkwind, 2024-04-13:
On a Freshly installed Server image i cant get WiFi to work.
I tried a few commands to see if i could get more info.

`ubuntu@ubuntu:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host noprefixroute
       valid_lft forever preferred_lft forever
2: end1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 0e:fe:7e:7a:02:6a brd ff:ff:ff:ff:ff:ff
    inet 192.168.31.73/24 metric 100 brd 192.168.31.255 scope global dynamic end1
       valid_lft 43053sec preferred_lft 43053sec
    inet6 fe80::cfe:7eff:fe7a:26a/64 scope link
       valid_lft forever preferred_lft forever`

`ubuntu@ubuntu:~$ sudo lshw -class network
  *-network UNCLAIMED
       description: Network controller
       product: BCM43752 802.11ax Dual Band Wireless LAN Controller
       vendor: Broadcom Inc. and subsidiaries
       physical id: 0
       bus info: pci@0004:41:00.0
       version: 02
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress cap_list
       configuration: latency=0
       resources: memory:f4200000-f420ffff memory:f4400000-f47fffff
  *-network
       description: Ethernet interface
       physical id: 14
       logical name: end1
       serial: 0e:fe:7e:7a:02:6a
       size: 100Mbit/s
       capacity: 1Gbit/s
       capabilities: ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=st_gmac driverversion=6.1.0-1006-rockchip duplex=full ip=192.168.31.73 link=yes multicast=yes port=twisted pair speed=100Mbit/s`

`ubuntu@ubuntu:~$ ls /sys/class/net
end1  lo`

Joshua-Riek, 2024-04-15:
Fixed with https://github.com/Joshua-Riek/firmware/commit/068165836b5362020cebbdd59d08deb47e4c4754. Wait a few hours then update the system with apt.

closed 2024-04-15 by Joshua-Riek

user2745, 2024-10-03:
I had a similar issue