Joshua-Riek/ubuntu-rockchip #703: OrangePi 5 Plus Ubuntu 24.04 LTS Beta MPV has bad performance

HZFrodo, 2024-04-13:
With Ubuntu 24.04 LTS Beta, 1080P HLS stream drops frames.

Maybe this patch was missed when Ubuntu 24.04 LTS Beta released?

https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/638#discussioncomment-9087653

[image]

Thank you for your great works.

nyanmisaka, 2024-04-13:
Yes. Some video containers such as m3u8/mpegts require that patch to trigger hardware acceleration. This is a quirk in MPV.

Joshua-Riek, 2024-04-13:
This patch https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/638#discussioncomment-9087653, was not yet included. I can try and include it later today, could you provide a link to the video file you tested so I can easily reproduce the issue? I have a lot of bugs to work on.

HZFrodo, 2024-04-13:
> This patch [#638 (reply in thread)](https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/638#discussioncomment-9087653), was not yet included. I can try and include it later today, could you provide a link to the video file you tested so I can easily reproduce the issue? I have a lot of bugs to work on.

@Joshua-Riek 
I just cannot wait the fixed MPV package since I am so excited with your release ;)
And I reverse-engineered the MPV ELF binary to force check the AV_CODEC_HW_CONFIG_METHOD_INTERNAL flag as https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/638#discussioncomment-9087653 says, MPV works well now, no any performance issue.
You can apply the patch and release a new MPV ;)

Joshua-Riek, 2024-04-13:
Build is running now on Launchpad, it should be downloadable from apt in a few hours.

https://launchpad.net/~jjriek/+archive/ubuntu/rockchip-multimedia/+build/28054068

HZFrodo, 2024-04-13:
> Build is running now on Launchpad, it should be downloadable from apt in a few hours.
> 
> https://launchpad.net/~jjriek/+archive/ubuntu/rockchip-multimedia/+build/28054068

New MPV installed via apt, tested some video files and some live streams, all good, we can close this issue now.

closed 2024-04-13 by HZFrodo