Joshua-Riek/ubuntu-rockchip #705: Orange Pi 5 Plus kernel crash

TheSainEyereg, 2024-04-13:
For the last 2 days I've been experiencing a kernel crash at around 2-3am (UTC +3) with the same error in journal. There is also a constant flood with the error "RK SPI transfer timed out" but I don't care much about it, unlike the crash of the whole system

[Pastebin with kernel log](https://pastebin.com/raw/yUQjHDyc)

[image]

Joshua-Riek, 2024-04-13:
Is this with kernel 5.10? It looks like something is spamming spi transfers and there is a memory leak.

TheSainEyereg, 2024-04-13:
> Is this with kernel 5.10?

Yep, `5.10.160-rockchip`

TheSainEyereg, 2024-04-19:
It's happened 3 times already, maybe it's about some userspaace programs?
Since I have an NVMe SSD in my Orange Pi I have moved various disk intensive tasks such as MongoDB, SurrealDB, NodeJS API's and website storage (nginx) from my Raspberry Pi. I also keep a simple 1-person Minecraft server just for me, with 8GB total and 2GB of initial memory pool.
With all these tasks, the system idle load is less than 30% and more than 10GB of RAM is still free so I don't think the problem is a severe lack of memory

flyzet-prog, 2024-05-06:
Unfortunately, i have the same problem on freshly installed system(Kernel 5.1).Only installed Home assistant and frigate docker image. Did you managed to resolve it?

Joshua-Riek, 2024-05-06:
This actually looks like a power problem. Are you both using a 5V 4A power supply?

flyzet-prog, 2024-05-06:
> This actually looks like a power problem. Are you both using a 5V 4A power supply?

Seems like not power issue. Because, i had that issue before, after your suggestions regarding proper  power supply, successfully resolved.

closed 2025-02-17 by github-actions[bot]