Joshua-Riek/ubuntu-rockchip #707: Opi5-Plus WiFi Connection Failed - Activation Of Network Connection Failed - WiFi Module OPi RTL8852BE-CG

JFLim1, 2024-04-13:
Hi @Joshua-Riek,

Image:  v2.0.0-beta/ubuntu-24.04-beta-preinstalled-desktop-arm64-orangepi-5-plus.img.xz
Device:  Orange Pi 5 Plus
WiFi Modules:  RL8852BE-CG (from Xulong)

WiFi module is detected but failed to connect with message:

> Connection failed
> Activation of network connection failed
> 

The system will continue to try to connect but connection failed.

```
[  502.366541] wlP2p33s0: authenticate with 34:e8:94:75:06:73
[  502.547394] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 1/3)
[  502.574746] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 2/3)
[  502.602005] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 3/3)
[  502.633026] wlP2p33s0: authentication with 34:e8:94:75:06:73 timed out
[  506.394444] wlP2p33s0: authenticate with 34:e8:94:75:06:73
[  506.568820] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 1/3)
[  506.598662] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 2/3)
[  506.629714] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 3/3)
[  506.660795] wlP2p33s0: authentication with 34:e8:94:75:06:73 timed out
[  510.827909] wlP2p33s0: authenticate with 34:e8:94:75:06:73
[  511.000644] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 1/3)
[  511.030797] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 2/3)
[  511.057974] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 3/3)
[  511.126142] wlP2p33s0: authentication with 34:e8:94:75:06:73 timed out
[  515.807763] wlP2p33s0: authenticate with 34:e8:94:75:06:73
[  515.977561] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 1/3)
[  516.005286] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 2/3)
[  516.038524] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 3/3)
[  516.074101] wlP2p33s0: authentication with 34:e8:94:75:06:73 timed out
```
Reported this before in #613 and https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/638#top

During first boot, the initial set up did not complete.  Had to power cycle to restart and similar situation after reboot and set up was incomplete but  this time the set up system automatically reboot and allow me to complete the system set up.

Both Chromium and MPV works as expected so far.  Both Chromium and MPV can stream Youtube with vpu hw acceleration.

Joshua-Riek, 2024-04-13:
The Orange Pi RTL8852BE is so strange, I have a generic one and another from Radxa yet they work fine. 

Can you please try the following commands:
```
sudo modprobe -r rtw89_8852be
sudo modprobe -i rtw89_8852be
```

JFLim1, 2024-04-13:
> Can you please try the following commands:
> 
> ```
> sudo modprobe -r rtw89_8852be
> sudo modprobe -i rtw89_8852be
> ```

Ok.  Performed the above, still the same.

```
jfl@localhost:~$ sudo modprobe -r rtw89_8852be
[sudo] password for jfl: 
jfl@localhost:~$ sudo modprobe -i rtw89_8852be
```
> Connection failed
> Activation of network connection failed

Joshua-Riek, 2024-04-13:
Hmm i dont have access to my Orange Pi 5 Plus right now, can you put the output of `ip a` and `dmesg` into a file and upload it to github? Or use pastebin.

JFLim1, 2024-04-13:
> Hmm i dont have access to my Orange Pi 5 Plus right now, can you put the output of `ip a` and `dmesg` into a file and upload it to github? Or use pastebin.

Attached herewith the "ip a" and "dmesg" output.

[Opi5-Plus RTL8852BE-CG 'ip a' and dmesg output.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/14967552/Opi5-Plus.RTL8852BE-CG.ip.a.and.dmesg.output.txt)

Joshua-Riek, 2024-04-13:
Hmm so the WiFi device is found as `wlP2p33s0`, but it looks like there is an authentication error.
```
[  502.366541] wlP2p33s0: authenticate with 34:e8:94:75:06:73
[  502.547394] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 1/3)
[  502.574746] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 2/3)
[  502.602005] wlP2p33s0: send auth to 34:e8:94:75:06:73 (try 3/3)
[  502.633026] wlP2p33s0: authentication with 34:e8:94:75:06:73 timed out
```

May I ask how did you connect to your WiFi network? Through the gnome settings, or a command line?

JFLim1, 2024-04-13:
> May I ask how did you connect to your WiFi network? Through the gnome settings, or a command line?

Not sure I understand your question correctly.  Am on Gnome Desktop, it basically autodetect the WiFi module. Enter the WiFi "password" through Gnome Settings.

Is there another method to enter the WiFi password?

Joshua-Riek, 2024-04-13:
You can connect to wifi using a command line and editing some files. But using the desktop settings menu should work just fine.

JFLim1, 2024-04-23:
Hi @Joshua-Riek,

No luck in WiFi connection with kernel-6.1.0.1008.8.

Joshua-Riek, 2024-04-23:
I'm starting to think this may be an issue with your network. I've tried multiple opi5+ with this wifi card and they all work.

SimplyCorbett, 2024-04-24:
No issues with an intel wifi card on the orange pi 5 plus.

SimplyCorbett, 2024-04-24:
I’m actually losing network connection after running a speed test so it looks like the orange pi 5 plus does have issues with the WiFi card.

Joshua-Riek, 2024-04-30:
Should be fixed with the noble release.

closed 2024-04-30 by Joshua-Riek

JFLim1, 2024-05-01:
Hi @Joshua-Riek,

Thanks for the newly released Ubuntu-Noble.  LibreOffice now can launch from Panel where earlier on Ubuntu-Noble-Beta, LibreOffice could not launch by clicking on the Icon on the Panel.

RTL8852BE-CG WiFi sort of work now.  On boot up it takes quite a number of tries before WiFi get connected.  The reason I said sort of work, is because the link speed seems to be much lower than on kernel-5.10.160.  On kernel-5.10.160 do not have issue streaming 4K/60 youtube either on Chromium or MPV without bufferring.

On kernel-6.1.0-1012.12 WiFi streaming 4K/60 Youtube on Chromium or MPV typically encounters frequent buffering.  Streaming 1080/60 is ok so far.

Found this on Armbian Forum:

> For `vendor` (6.1) kernel branch, the rtw89 driver needs to be updated, to add support for RTL8852BE, used in Orange Pi R6 WiFi module: https://github.com/orangepi-xunlong/linux-orangepi/commit/96b2d2827a3f1e4719ebdfcac0b2108e8cbae932

Most likely it is already patched on your released Ubuntu-Noble Desktop image.

Thanks for the great works and support.

Joshua-Riek, 2024-05-01:
I actually backported rtw89 from Linux 6.6 because it supports loading compressed firmware files. 

The RTL8852BE from Orange Pi is very bad, generic RTL8852BE modules work great and honestly, I don't want to dig into the Wi-Fi drivers to improve it, would take so much time.

JFLim1, 2024-05-01:
I thought WiFI module from the same SBC manufacturer would have better driver support.  Hopefully, in future release/update RTL8852BE-CG will get better drivers.